aircrack-ng/rtl8812au #467: deauth not working :/

azamet90, 2019-11-12:
sudo aireplay-ng -9 wlan0
00:28:02  Trying broadcast probe requests...
00:28:04  No Answer...
00:28:04  Found 5 APs

00:28:04  Trying directed probe requests...
00:28:04  D0:6F:82:56:EF:0C - channel: 11 - 'Univision_1320'
00:28:10   0/30:   0%

00:28:10  2E:0E:3D:A6:43:EF - channel: 11 - 'AndroidAP43EF'
00:28:16   0/30:   0%

00:28:16  34:CC:28:00:FA:EC - channel: 11 - 'G-Mobile 8FEF'
00:28:22   0/30:   0%

00:28:22  00:27:22:C8:6A:C0 - channel: 11 - 'KEWIKO_WiFi'
00:28:28   0/30:   0%

00:28:28  00:27:22:8C:48:72 - channel: 11 - 'KEWIKO_WiFi'
00:28:34   0/30:   0%

azamet90, 2019-11-12:
fixed the problem with uninstalling it and installing from kali  repo

closed 2019-11-12 by azamet90

nemesisalt, 2019-11-13:
This issue should stay open.. I'm experiencing this on multiple machines, even on the latest 5.6.4.2 branch.

kimocoder, 2019-11-13:
Frame injection isn't working atm. That's why

kimocoder, 2019-11-14:
Update and it should be working again.

[image]

azamet90, 2019-11-14:
> Update and it should be working again.
> 
> [image]

Also after mod probe it is working you mean?

kimocoder, 2019-11-14:
For me it does, after reverting a commit. I just pushed an update that fixes injection.

azamet90, 2019-11-14:
> I just pushed an update that fixes injection.

my english is bad sorry for asking again: you  did not install any extra package just modprobe you did right?

kimocoder, 2019-11-14:
I haven't added more iw outout yet, but I added txpower control yesterday..

"iw <adapter> info" will give it to you