aircrack-ng/rtl8812au #468: Multiple problems with AWUS036ACH

BedoM, 2019-11-13:
Hello i pushase a new card for my personal pentest with aircrack but i have AWUS036H and do not have a problem.

Problem 1 : Injection doest work ..
Problem 2 : Fixed channel .. Why ?

`[mathieu@Y520 ~]$ iwconfig wlp0s20f0u2
wlp0s20f0u2  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

[mathieu@Y520 ~]$ sudo aireplay-ng --test wlp0s20f0u2
20:24:59  Trying broadcast probe requests...
20:25:01  No Answer...
20:25:01  Found 0 APs

airodump work buf not the --channel (without -b abg)

sudo airodump-ng wlp0s20f0u2 --channel 11

 CH 11 ][ Elapsed: 18 s ][ 2019-11-13 20:27 ][ fixed channel wlp0s20f0u2: 1   
`

Why is fixed channel ?

Dmesg:

Thank you 

[for-github.txt](https://github.com/aircrack-ng/rtl8812au/files/3843085/for-github.txt)

kimocoder, 2019-11-13:
Need some more output... give me "airmon-ng", "iwconfig" and "modinfo 88XXau" thanks

BedoM, 2019-11-13:
For swith to mode monitor I use this : 
```

wlp0s20f0u1  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
sudo ifconfig  wlp0s20f0u1 down
sudo iwconfig wlp0s20f0u1 mode monitor
sudo ifconfig  wlp0s20f0u1 up

iwconfig wlp0s20f0u1 
wlp0s20f0u1  IEEE 802.11b  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency:2.457 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

because:

```
sudo airmon-ng start wlp0s20f0u1
PHY     Interface       Driver          Chipset

phy5    wlp0s20f0u1     88XXau          Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
 (monitor mode enabled)
Error setting channel: command failed: Device or resource busy (-16)
Error -16 likely means your card was set back to station mode by something.
Removing non-monitor wlp0s20f0u1 interface...
 (monitor mode disabled)
```

------------------------------
```
sudo modinfo 88XXau

filename:       /lib/modules/5.3.9-300.fc31.x86_64/extra/88XXau.ko.xz
version:        v5.3.4_28226.20180614_COEX20171103-6c6d
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     91D5C4958AB1B9E0E6EC5C5
alias:          usb:v0846p9054d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p809Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p809Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0106d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA833d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA834d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1853d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1852d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1817d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p331Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0120d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0122d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p011Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v3823p6249d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4007d*dc*dsc*dp*ic*isc*ip*in*
[... 133 lines cut ...]
parm:           rtw_OffEfuseMask:default open Efuse Mask value:0 (uint)
parm:           rtw_FileMaskEfuse:default drv Mask Efuse value:0 (uint)
parm:           rtw_rxgain_offset_2g:default RF Gain 2G Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gl:default RF Gain 5GL Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gh:uint
parm:           rtw_rxgain_offset_5gm:default RF Gain 5GM Offset value:0 (uint)
parm:           rtw_pll_ref_clk_sel:force pll_ref_clk_sel, 0xF:use autoload value (uint)
parm:           rtw_tx_pwr_by_rate:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_target_tx_pwr_2g_a:2.4G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_b:2.4G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_c:2.4G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_d:2.4G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_a:5G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_b:5G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_c:5G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_d:5G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_phy_file_path:The path of phy parameter (charp)
parm:           rtw_load_phy_file:PHY File Bit Map (int)
parm:           rtw_decrypt_phy_file:Enable Decrypt PHY File (int)
parm:           rtw_en_napi:int
parm:           rtw_en_gro:int
parm:           rtw_iqk_fw_offload:int
parm:           rtw_ch_switch_offload:int

```

kimocoder, 2019-11-13:
Yeah right, your on a left behind, older branch.. there have been several releases after v5.3.4. v5.6.4.1 is the newest, I suggest you update to that one.

BedoM, 2019-11-13:
Thank you y have unistall and install the latest 

but y have an error for the channel, the option of airodump (-c --channel) do'sent work all ap is print.

kimocoder, 2019-11-13:
@BedoM Nope.. your wrong. But i wrote the wrong branch, **v4.6.4.2** is the newest, not v4.6.4.1

[image]

kimocoder, 2019-11-14:
Also, "Error -16 likely means your card was set back to station mode by something." indicates that you may have interference, try "airmon-ng check kill" before setting the adapter to monitor mode.

closed 2019-11-14 by kimocoder