aircrack-ng/rtl8812au #469: Injection not working, aireplay says it's sending but nothing in wireshark..

nemesisalt, 2019-11-13:
As the title says, i'm using the latest branch and i'm trying to inject packets in the 802.11a band, I even created a custom script in python to inject but up to no avail, it can receive perfectly but can't transmit.

airmon-ng and iwconfig show nothing weird.

I'm modprobing the default configuration, only `modprobe 88XXau`, If you need anything else please tell me, Cheers.

OS: Arch Linux
Kernel 5.3.8
Arch: 64 Bit

nemesisalt, 2019-11-13:
5.2.20 kinda works while 5.6.4.2 is extremely buggy

Error0x00000001, 2019-11-14:
Confirmation.
In Kali
5.2.0-kali2-amd64, 5.3.0-kali2-amd64 on 5.6.2 injections do not work. Alfa AWUS036ACH adapter

kimocoder, 2019-11-14:
@nemesisalt @Error0x00000001 

Do a git pull. Been working on the issue and got results.
branch v5.6.4.2 is updated.

kinda just started on that branch, works fine for me, but still have stuff to do with it.

[image]

kimocoder, 2019-11-14:
Injection have been confirmed fixed. Thanks for reporting

closed 2019-11-14 by kimocoder