aircrack-ng/rtl8812au #470: V5.6.4.2 doesn't work with Kismet

ianburton20, 2019-11-14:
V5.6.4.2 much advanced of late - but it still can't work with Kismet which needs the driver to report packets seen at the 802.11 link layer.

kimocoder, 2019-11-14:
@ianburton20 Just pushed a frame injection fix. Download, compile/build/install reboot and voilà

[image]

ianburton20, 2019-11-14:
Thanks. Did a dkms build - but no difference. Still reports no mac80211 controls.

kimocoder, 2019-11-14:
mac80211 controls? You know this is a cfg80211/nl80211 driver right?

ianburton20, 2019-11-15:
Thanks again.  Researched cfg82011/nl80211 but don't fully understand it. Just know that V5.6..4.2 does not work with Kismet - and I guess that many would want it to.

kimocoder, 2019-11-15:
Yeah, it does not support mac80211 (which is newer I/O handlers etc).

closed 2019-11-15 by kimocoder