aircrack-ng/rtl8812au #472: Alfa AWUS036ACH not working in Kali Linux VirtualBox v5.6.4.2

JaumeS4, 2019-11-14:
Hi, I'm trying to install the awus036ACH on kali linux (virtualbox), with drivers 5.6.4.2.
First I did an apt update, upgrade and dist-upgrade, then installed the rtl8812au driver wihout problems, but kali doesn't detect the card.
lsusb show me that the card it's connected.
I know there is some problems with this card and virtualbox, but at least I should recognize the card, right?
Any suggestions? Thanks.

kimocoder, 2019-11-14:
Yes, the README.md clearly states that, unfortunately :100:

JaumeS4, 2019-11-14:
Just to inform, version 5.2.20 is working for me. Monitor mode and injection, but with very low range (just finding my own network, but just for testing purpose it's fine).

kimocoder, 2019-11-14:
Yes I know, that's why I'm on latest, great range on 8812au :+1:

kimocoder, 2019-11-14:
Known issue. Nothing we can do about it, at least for now. Closing

closed 2019-11-14 by kimocoder