aircrack-ng/rtl8812au #473: No AP mode in 5GHz

Kawashima-Azumi, 2019-11-15:
Worked fine previously, but no longer after yesterday's update. Not sure which commit, at work now and no time to debug.

Hostapd says channel (either 5.2 or 5.8 GHz) no in list.
2.4GHz AP Ok.

Pretty sure the adapter allows for both 5.2 and 5.8GHz, and did iw reg set correctly.

BytEvil, 2019-11-15:
Same issue. Can`t start hostapd on RTL8814AU after upgrading drivers.
Hostapd in debug mode output:
```
nl80211: Mode IEEE 802.11g: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467[NO_IR] 2472[NO_IR] 2484[NO_IR]
nl80211: Mode IEEE 802.11a: 5180[NO_IR] 5200[NO_IR] 5220[NO_IR] 5240[NO_IR] 5260[NO_IR][RADAR] 5280[NO_IR][RADAR] 5300[NO_IR][RADAR] 5320[NO_IR][RADAR] 5500[NO_IR][RADAR] 5520[NO_IR][RADAR] 5540[NO_IR][RADAR]
nl80211: Mode IEEE 802.11b: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467[NO_IR] 2472[NO_IR] 2484[NO_IR]
Disable HT capability [DSSS_CCK-40] on 5 GHz band
hw vht capab: 0x3c054b2, conf vht capab: 0x3c001b0
HT40: control channel: 36  secondary channel: 40
HT40 secondary channel 40 not allowed
wlp0s20u3u4: interface state COUNTRY_UPDATE->DISABLED
wlp0s20u3u4: AP-DISABLED
```
Hostapd worked with 4.3.21 drivers without problems with the same hostapd.conf.

Kawashima-Azumi, 2019-11-17:
Enabled 5GHz by bringing back the CONFIG_DISABLE_REGD_C option. Surely I know this can only be a temporary workaround, but really helps with WiFi speed.

BytEvil, 2019-11-18:
> Enabled 5GHz by bringing back the CONFIG_DISABLE_REGD_C option. Surely I know this can only be a temporary workaround, but really helps with WiFi speed.

not worked for me, no changes after adding `CONFIG_DISABLE_REGD_C = n` into MAKEFILE and recompiling & installing. Same problem:
`nl80211: Mode IEEE 802.11a: 5180[NO_IR] 5200[NO_IR] 5220[NO_IR] 5240[NO_IR] 5260[NO_IR][RADAR] 5280[NO_IR][RADAR] 5300[NO_IR][RADAR] 5320[NO_IR][RADAR] 5500[NO_IR][RADAR] 5520[NO_IR][RADAR] 5540[NO_IR][RADAR]`

logan001, 2019-11-20:
how/where you set that?

kimocoder, 2019-11-20:
CONFIG_DISABLE_REGD_C needs to be added several places including Makefile in order to work. 
File "wifi_regd.c" but I don't have time for that at this point

kimocoder, 2019-11-25:
Issue confirmed. I'll do a writeup on this issue later this week/weekend.

vespian, 2020-01-07:
> CONFIG_DISABLE_REGD_C needs to be added several places including Makefile in order to work. File "wifi_regd.c" but I don't have time for that at this point

@kimocoder Could you please have a look at https://github.com/aircrack-ng/rtl8812au/pull/531 and https://github.com/aircrack-ng/rtl8812au/issues/530 ?

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)