aircrack-ng/rtl8812au #476: all problems about this driver i see:

azamet90, 2019-11-15:
first of all sorry for my bad english let me list you all the problems:

1. if i use bettercap with my wifi adaptor it says "could not change tx power" or somethings like that
2. if i use airgeddon it cannot change the wifi adapter to monitor mode
3. if i scan wifi networks some of then the essid cannot be detected but can detected with my internal wifi which is intel
4. i cannot change the txpower with set txpower command too... it looks like it is changed because of not giving any output but if i check the tx power is same...
5. internal wifi is weak but alfa wifi is stronger right? i cannot get beacons same much like my internal wifi...

nemesisalt, 2019-11-15:
I also cannot seem to inject packets at the 5GHz 802.11a range, but the driver for pentesting purposes atm works really well for 2.4GHz

kimocoder, 2019-11-15:
1. TX power control was added yesterday. "iwconfig" is deprecated, so you have to use "iw"
    Many tools have not migrated from "iwconfig" to "iw" and from "ifconfig" to "ip" yet. That's the issue.

2. Airgeddon also needs to migrate from "iwconfig" to "iw" and so on..

3. Scan should be working again after some  work the past days.

4. See picture i added.. it works just fine, with "iw" not "iwconfig" !

[image]

[image]

kimocoder, 2019-11-15:
As for beacons, I have a patch for bad FCS, but need some more time with it

azamet90, 2019-11-16:
> As for beacons, I have a patch for bad FCS, but need some more time with it

thank you for your hard work and answering.
but last thing you still did not answered: some AP ESSID cannot detected from alfa wifi adapter but can be detected from my intel wifi but why? i think this is a problem too or not?

OscarAkaElvis, 2019-11-16:
on airgeddon it is planned to remove ifconfig and iwconfig but didn't start yet to investigate how to do it... I'll need some help to do it... for example:

Instead of `iwconfig wlan0` to see the in which mode a card is, which command should be used?
Instead of `ifconfig wlan0 down && iwconfig wlan0 mode monitor && ifconfig wlan0 up` to put a card in monitor mode, which commands need to launch? if somebody already know the answers it will save some time to me.

azamet90, 2019-11-16:
> on airgeddon it is planned to remove ifconfig and iwconfig but didn't start yet to investigate how to do it... I'll need some help to do it... for example:
>
> Instead of `iwconfig wlan0` to see the in [...]

sudo ip link set IFACE down
sudo iw IFACE set monitor control
sudo ip link set IFACE up
this is maybe the answer to your question on aircrack
sudo ip link set wlan0 down
sudo iw wlan0 set monitor control
sudo ip link set wlan0 up

OscarAkaElvis, 2019-11-17:
I've doing some tests and I see instead of simple iwconfig and parse the output to check the mode of a card, `iw wlan0 info` can be launched, so this part is ok.

Regarding the mode change, thanks but it seems using this method I can set monitor mode but to set managed again is not working. I mean `iw wlan0 set managed control` command is failing. Ok to use these commands to set monitor mode... but, what about to set managed?

OscarAkaElvis, 2019-11-17:
Ok I've found the solution... for managed mode, Instead of `iw wlan0 set monitor control` it should be `iw wlan0 set type managed`. I think I have all I need... next airgeddon version v10.01 will have removed iwconfig and ifconfig commands. Thanks.

ZerBea, 2019-11-26:
Unfortunately iw uses libnl (NETLINK) to set monitor mode which isn't a good choice:
Netlink supports multicast communications between the kernel and multiple user-space processes, while ioctl s are strictly one-to-one. Netlink messages can be lost for various reasons (e.g. out of memory), while ioctl s are generally more reliable due to their immediate-processing nature.
Read more here:
https://www.quora.com/What-are-the-differences-between-netlink-sockets-and-ioctl-calls

Just add debug option to verify this:
sudo ip link set  wlan0 down
sudo iw --debug dev wlan0 set type monitor

you'll see the BEGIN NETLINK MESSAGE and END NETLINK MESSAGE
Read more here:
http://beautifulwork.org/debug-using-iw-command-netlink-debugging/