aircrack-ng/rtl8812au #478: No secrets provided

pshem, 2019-11-15:
Since updating to Ubuntu 19.10(kernel 5.3.0-23), the distro packaged driver crashes(`rtl8812-dkms`), so I'm trying alternatives. This driver gets the furthest, but fails to connect to networks requiring authenticaton with a KDE notifcation
```
Wireless interface (wlx24050fae5062)
No secrets were provided
```
I know the password is correct because it works fine when my built in WiFi card doesn't go on strike.

I'm using an dongle that in `lsusb` shows up as `Bus 001 Devce 006: 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 801.11a/g/n/ac 2T2R DB WLAN Adapter`. Upon `insmoding` the kernel module built with `make`, these lines can be seen in dmesg:
```
usbcore: registered new interface driver 88XXau
88XXau 1-3:1.0 wlx24050fae5062: renamed from wlan0
[phydm_la_set_buff_mode] Warning!
IPv6 ADDRCONF(NETDEV_CHANGE): wlx24050fae5062 link becomes ready
```

The last line reappears with every new connection attempt. I'm trying to connect to eduroam, a WPA2-Enterprise WiFi network used by most EU universities. I'll try to see if the error appears when connecting to WPA2-PSK network at home if I can find an Ethernet cable or my built in WiFi card wakes up.

Let me know if  can provide any additional debugging information.

kimocoder, 2019-12-16:
Confirmed. This is a new one, kernels are updated to fast these days, 'causes lots of work to maintain drivers like this. I'll dive into it

```
[1253182.877161] usb 2-1.2: new high-speed USB device number 28 using ehci-pci
[1253182.991006] usb 2-1.2: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[1253182.991010] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1253182.991011] usb 2-1.2: Product: 802.11n NIC
[1253182.991013] usb 2-1.2: Manufacturer: Realtek
[1253182.991014] usb 2-1.2: SerialNumber: 123456
[1253183.154744] 88XXau: loading out-of-tree module taints kernel.
[1253183.879160] usb 2-1.2: 88XXau 00:c0:ca:a4:7b:06 hw_info[d7]
[1253183.884466] usbcore: registered new interface driver 88XXau
[1253184.956054] [phydm_la_set_buff_mode] Warning!
```

closed 2020-01-11 by kimocoder

pshem, 2020-01-15:
Does closing this issue mean it's fixed? Should I test if the new release allows me to connect?

kimocoder, 2020-01-15:
I testes it, but I see I didn't ask which branch you where using?

pshem, 2020-01-15:
I downloaded a ZIP and moved it via USB(no internet on the laptop in need). The ZIP was of the v5.6.4.2 branch, from November 14th 2019. I don't have a git commit id I'm afraid, but the timing suggests it was https://github.com/aircrack-ng/rtl8812au/commit/ac911b45c82606546ae4b30904204171c7ddd050