aircrack-ng/rtl8812au #481: Frame Injection on VirtualBox

BolSzol, 2019-11-16:
Hello, I encountered a problem with frame injection on VB Kali 2019.4,   i can get my usb WiFi adapter(TP Link archer T9UHv2 with realtek rtl8814au driver) into monitor mode with 'iwconfig wlan0 mode monior' and 'airodump-ng wlan0' works fine, i can see networks around but when i try ' airoplay-ng -9 wlan0' im finding AP's but in injection i see 'no answer' and 0 packets are send I also cant run deauthetication and cant get a handshake :( 

Any ideas what might help? Is it a problem with VirtualBox or Virtual Machine? 
Disclaimer: I have installed vb guest additions so this is not a problem

blshkv, 2019-11-27:
It is not advisable to use a vm for any tricky USB operations.
USB2 might have timing issues, USB3 doesn't work properly.

btw, try to reconnect your wifi adapter into USB2.

P.S. you need to try the same setup on bare metal. If works, file a bug report at Oracle.

kimocoder, 2019-11-27:
I've disabled USB3 anyway. It's forced down to USB2 for now, duo to bad code (modeswitch)

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)