aircrack-ng/rtl8812au #486: Kernel 5.3.0-23-generic on Ubuntu hangs/crashes with v5.6.4.2

pdelagrave, 2019-11-21:
Hi,

Tried the latest code 3b0b7b989dd90e3debaa576ddbbbb0dedfce248b on Ubuntu with kernel v5.3.0 and the module loads fine but everything start hanging afterwards, even making it impossible to do a clean reboot.

* No secure boot issue.
* No compilation issue.
* Device is a Alfa AC1200 - AWUS036ACM.
* This is the only adapter on the machine.
* This is not a virtual machine.
* Tried both `make, insmod` and the `dkms` method.
* Using Ubuntu 19.10 if that changes anything.
* Had the same problem on a Linux Mint machine using a 4.15.0 kernel.

Here is the relevant part from `dmesg`:
```
13:24:43 desktop kernel: [  340.003483] usbcore: registered new interface driver 88XXau
13:24:49 desktop kernel: [  346.741328] usb 1-6: new high-speed USB device number 15 using xhci_hcd
13:24:49 desktop kernel: [  346.889810] usb 1-6: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
13:24:49 desktop kernel: [  346.889815] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
13:24:49 desktop kernel: [  346.889819] usb 1-6: Product: 802.11n NIC
13:24:49 desktop kernel: [  346.889821] usb 1-6: Manufacturer: Realtek
13:24:49 desktop kernel: [  346.889824] usb 1-6: SerialNumber: 123456
13:24:50 desktop kernel: [  347.012327] usb 1-6: 88XXau 00:a0:aa:aa:ba:ab hw_info[d7]
13:24:50 desktop systemd[1]: Starting Load/Save RF Kill Switch Status...
13:24:50 desktop iwd[1069]: Wiphy: 0, Name: phy0
13:24:50 desktop iwd[1069]: #011Permanent Address: 00:00:00:00:00:00
13:24:50 desktop iwd[1069]: #011Bands: 2.4 GHz 5 GHz
13:24:50 desktop iwd[1069]: #011Ciphers: CCMP TKIP BIP
13:24:50 desktop iwd[1069]: #011Supported iftypes: ad-hoc station ap
13:24:50 desktop mtp-probe: checking bus 1, device 15: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6"
13:24:50 desktop mtp-probe: bus: 1, device: 15 was not an MTP device
```

Any suggestion to how and where to collect more logs would be appreciated.
Thanks

terencode, 2019-11-22:
Same with an archer T4U, I remember seeing something along ` "worker wlan0" is taking a long time`

eugeneware, 2019-12-15:
I can confirm the same issue with Kal Linux running 5.3.0:

```
uname -a
Linux kali 5.3.0-kali2-amd64 #1 SMP Debian 5.3.9-3kali1 (2019-11-20) x86_64 GNU/Linux
```
I'm seeing it with multiple wifi cards running the rtl8812au chipset.

I can connect to wifi, but when I try to put it into monitor mode the machine halts.

gpolitis, 2020-04-30:
I can also confirm the same issue with Ubuntu 18.04.4 LTS:

> uname -a
> Linux ubuntu-desktop 5.3.0-28-generic #30~18.04.1-Ubuntu SMP Fri Jan 17 06:14:09 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

I have a couple of TP-Link AC600 devices, one connecting to my WiFi and another acting as a Hotspot.

> lsusb
> Bus 001 Device 003: ID 2357:011f
> Bus 001 Device 002: ID 2357:011f

The hangs started happening randomly when I configured my second adapter to act as a Hotspot.

ikegami-t, 2020-05-16:
The PC hang up issue is caused on Ubuntu 20.04 Linux kernel 5.4.0-29-generic with the 8812au driver when connected to SSID by Buffalo - WI-U3-866D.
Also the issue was not happened on Ubuntu 20.04 Linux kernel 5.7.0-rc3+ built locally.
So it seems that the issue is fixed by the latest kernel driver.
Now I am confirm again with the latest kernel.

ikegami-t, 2020-05-16:
Just confirmed again as the 8812au driver works correctly with the kernel 5.7.0-rc5+ without any error.

ikegami-t, 2020-05-17:
Not caused on the kernel 5.6.1 built locally.

ikegami-t, 2020-05-17:
Caused on the kernel 5.5.1 built locally so seems that fixed by the kernel 5.6.1.

ikegami-t, 2020-05-17:
Hi,
Just confirmed that the issue has been fixed by the patch in 5.6.1 below.
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/net/wireless?h=v5.6.1&id=9951ebfcdf2b97dbb28a5d930458424341e61aa2
Also it is applied to the v5.5.8 maintenance version.

conqueror8888, 2021-05-11:
> I'm sorry that my effort is considered as spamming.
> May I ask where I can find a list of the Maintainers to this project?
> The only one I found by now is kimocoder and he is maintaining somewhat 600 projects.
> when I look at how old some open issues are I think he has better things to do than closing old issues.
> So I don't think I'm spamming anything by politely asking the issue openers to consider closing there stuff.
> I, at the moment, have the time to remind them and with more issues closed one can focuse on the important still opened ones.

Yo bro, u were said to be spamming for a good reason!

Your email client burst 50+ identical messages into this list. You should have it checked out!

No one here is giving you a hard time at all!

Peace! ✌

closed 2021-05-11 by kimocoder

cedricbambooza, 2021-05-11:
@cyrus104  why are you following me?