aircrack-ng/rtl8812au #487: Tried to install 5.6.4.2 with dkms but got errors...

Doppel-D, 2019-11-21:
Does anyone has a clue?

Linux amilo 4.9.0-11-686-pae #1 SMP Debian 4.9.189-3+deb9u2 (2019-11-11) i686 GNU/Linux
```
`/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c: In function ‘rtw_alloc_etherdev_with_old_priv’:
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2392:9: error: ‘struct net_device’ has no member named ‘min_mtu’
  pnetdev->min_mtu = WLAN_MIN_ETHFRM_LEN;
         ^~
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2394:9: error: ‘struct net_device’ has no member named ‘max_mtu’
  pnetdev->max_mtu = WLAN_DATA_MAXLEN;
         ^~
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c: In function ‘rtw_alloc_etherdev’:
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2417:9: error: ‘struct net_device’ has no member named ‘min_mtu’
  pnetdev->min_mtu = WLAN_MIN_ETHFRM_LEN;
         ^~
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c:2419:9: error: ‘struct net_device’ has no member named ‘max_mtu’
  pnetdev->max_mtu = WLAN_DATA_MAXLEN;
         ^~
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.c: At top level:
cc1: warning: unrecognized command line option ‘-Wno-cast-function-type’
/usr/src/linux-headers-4.9.0-11-common/scripts/Makefile.build:309: die Regel für Ziel „/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.o“ scheiterte
make[4]: *** [/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/osdep_service.o] Fehler 1
make[4]: *** Es wird auf noch nicht beendete Prozesse gewartet...
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/os_intfs.c: In function ‘rtw_init_netdev’:
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/os_intfs.c:1594:9: error: ‘struct net_device’ has no member named ‘min_mtu’
  pnetdev->min_mtu = WLAN_MIN_ETHFRM_LEN;
         ^~
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/os_intfs.c:1596:9: error: ‘struct net_device’ has no member named ‘max_mtu’
  pnetdev->max_mtu = WLAN_DATA_MAXLEN;
         ^~
/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/os_intfs.c: At top level:
cc1: warning: unrecognized command line option ‘-Wno-cast-function-type’
/usr/src/linux-headers-4.9.0-11-common/scripts/Makefile.build:309: die Regel für Ziel „/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/os_intfs.o“ scheiterte
make[4]: *** [/var/lib/dkms/rtl8812au/5.6.4.2/build/os_dep/linux/os_intfs.o] Fehler 1
/usr/src/linux-headers-4.9.0-11-common/Makefile:1543: die Regel für Ziel „_module_/var/lib/dkms/rtl8812au/5.6.4.2/build“ scheiterte
make[3]: *** [_module_/var/lib/dkms/rtl8812au/5.6.4.2/build] Fehler 2
Makefile:152: die Regel für Ziel „sub-make“ scheiterte
make[2]: *** [sub-make] Fehler 2
Makefile:8: die Regel für Ziel „all“ scheiterte
make[1]: *** [all] Fehler 2
make[1]: Verzeichnis „/usr/src/linux-headers-4.9.0-11-686-pae“ wird verlassen
Makefile:2245: die Regel für Ziel „modules“ scheiterte
make: *** [modules] Fehler 2
```

terencode, 2019-11-25:
@Doppel-D I think your kernel is too old. Try upgrading it.

kimocoder, 2019-11-26:
I know what it is. I'll add a handler for your kernel this weekend.

Doppel-D, 2019-11-26:
Thank you for your help! Kernel upgrading would be a measure of last resort because I know it will get me into sweat, tears & pain ;- )

kimocoder, 2019-11-26:
This issue should be resolved. Please update/compile/install/reboot :+1:

closed 2019-11-26 by kimocoder

Doppel-D, 2022-07-21:
Thank you very much!