aircrack-ng/rtl8812au #488: rtw_get_scch_by_cch_offset+0x71/0xb8 [88XXau] -  Ubuntu 18.04 drops wifi connection.

jkzharry, 2019-11-25:
Hi
Any reason why i'm seeing these stack traces in syslog, multiple times within a second. WIFI association will drop subsequently and system requires reboot.

Nov 24 22:11:05 dsu1804_03 kernel: [  874.375826] ------------[ cut here ]------------
Nov 24 22:11:05 dsu1804_03 kernel: [  874.375952] WARNING: CPU: 3 PID: 1675 at /home/dsu1804_03id01/rtl8812au/core/rtw_rf.c:215 rtw_get_scch_by_cch_offset+0x71/0xb8 [88XXau]
Nov 24 22:11:05 dsu1804_03 kernel: [  874.375954] Modules linked in: 88XXau(OE) cfg80211 intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm input_leds irqbypass intel_cstate dell_wmi d
ell_smbios snd_hda_codec_hdmi dcdbas intel_rapl_perf snd_hda_codec_realtek snd_hda_codec_generic snd_hda_intel sparse_keymap serio_raw intel_wmi_thunderbolt dell_wmi_descriptor mei_me snd_hda_codec wmi_
bmof snd_hda_core snd_hwdep snd_pcm mei snd_timer intel_pch_thermal mac_hid snd soundcore ie31200_edac shpchp acpi_pad sch_fq_codel ib_iser rdma_cm iw_cm ib_cm ib_core iscsi_tcp libiscsi_tcp libiscsi sc
si_transport_iscsi ip_tables x_tables autofs4 btrfs zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear crct10dif_
pclmul crc32_pclmul ghash_clmulni_intel
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376042]  pcbc aesni_intel i915 hid_generic e1000e aes_x86_64 crypto_simd i2c_algo_bit glue_helper cryptd ptp uas drm_kms_helper psmouse pps_core usbhid syscopyarea
 sysfillrect ahci sysimgblt fb_sys_fops usb_storage hid libahci drm wmi video [last unloaded: cfg80211]
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376080] CPU: 3 PID: 1675 Comm: RTW_CMD_THREAD Tainted: G        W  OE    4.15.0-70-generic #79-Ubuntu
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376166] RIP: 0010:rtw_get_scch_by_cch_offset+0x71/0xb8 [88XXau]
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376168] RSP: 0018:ffffaf450221bdf8 EFLAGS: 00010282
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376172] RAX: 0000000000000024 RBX: 0000000000000000 RCX: 0000000000000000
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376174] RDX: 0000000000000000 RSI: ffff9c886dd96498 RDI: ffff9c886dd96498
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376177] RBP: ffffaf450221be00 R08: 0000000000001a65 R09: 0000000000000004
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376178] R10: ffffaf450221be78 R11: 0000000000000001 R12: ffffaf45020ed000
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376180] R13: 0000000000000002 R14: 0000000000000000 R15: 000000000000006e
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376184] FS:  0000000000000000(0000) GS:ffff9c886dd80000(0000) knlGS:0000000000000000
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376186] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376188] CR2: 00007f9d95d682f8 CR3: 000000038b40a003 CR4: 00000000003606e0
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376191] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376193] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376194] Call Trace:
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376316]  rtw_hal_set_chnl_bw+0x11b/0x137 [88XXau]
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376397]  set_channel_bwmode+0xab/0xe7 [88XXau]
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376470]  rtw_set_chbw_hdl+0x40/0x49 [88XXau]
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376518]  rtw_cmd_thread+0x1a2/0x3f7 [88XXau]
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376585]  ? rtw_set_external_auth_status+0xff/0xff [88XXau]
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376591]  kthread+0x121/0x140
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376640]  ? rtw_stop_cmd_thread+0x3f/0x3f [88XXau]
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376644]  ? kthread_create_worker_on_cpu+0x70/0x70
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376651]  ret_from_fork+0x35/0x40
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376654] Code: fe 02 74 49 8d 57 da 80 fa 89 77 4f 84 c0 74 4b 8d 47 02 83 ef 02 80 fb 02 0f 45 c7 5b 5d c3 48 c7 c7 90 9f dc c0 e8 84 e4 bd cd <0f> 0b 89 d8 eb eb 
8d 47 02 83 ef 02 80 fb 02 0f 45 c7 eb dd 8d 
Nov 24 22:11:05 dsu1804_03 kernel: [  874.376729] ---[ end trace fded9c76ef3a35a7 ]---
Nov 24 22:11:06 dsu1804_03 kernel: [  875.013212] ------------[ cut here ]------------

cdeletre, 2019-12-16:
Hi,

I've got similar warnings from dmesg on Kali 2019.4 AMD64 running in a virtual machine with VMWare Fusion 10.5.  At the moment I'm only using a Alfa AWUS1900 in monitor mode with Kismet so I can't say about disconnection. Kismet doesn't complain about this as far as I can see.

I'm using a dkms install from `ebaed2` commit and the lastest vmware tools.
```
[10103.693412] WARNING: CPU: 1 PID: 1251 at /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_rf.c:215 rtw_get_scch_by_cch_offset+0x6c/0xb2 [88XXau]
[10103.693415] Modules linked in: 88XXau(O) cfg80211 rfkill vmw_vsock_vmci_transport vsock fuse intel_rapl_msr intel_rapl_common vmw_balloon intel_rapl_perf joydev serio_raw pcspkr snd_ens1371 snd_ac97_codec ac97_bus gameport snd_rawmidi snd_seq_device snd_pcm sg snd_timer snd soundcore vmw_vmci battery evdev ac binfmt_misc sunrpc ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic vmw_pvscsi vmxnet3 sd_mod sr_mod cdrom hid_generic usbhid hid ata_generic crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel aes_x86_64 crypto_simd cryptd glue_helper psmouse xhci_pci xhci_hcd ahci libahci e1000 ehci_pci vmwgfx uhci_hcd ttm ehci_hcd drm_kms_helper usbcore usb_common ata_piix mptspi mptscsih mptbase libata scsi_transport_spi drm scsi_mod i2c_piix4 button
[10103.693467] CPU: 1 PID: 1251 Comm: RTW_CMD_THREAD Tainted: G        W  O      5.3.0-kali2-amd64 #1 Debian 5.3.9-3kali1
[10103.693470] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 05/19/2017
[10103.693522] RIP: 0010:rtw_get_scch_by_cch_offset+0x6c/0xb2 [88XXau]
[10103.693526] Code: fe 02 74 48 8d 57 da 80 fa 89 77 4e 84 c0 74 4a 8d 47 02 83 ef 02 80 fb 02 0f 45 c7 5b c3 48 c7 c7 60 03 95 c0 e8 20 0e 65 e4 <0f> 0b 89 d8 eb ec 8d 47 02 83 ef 02 80 fb 02 0f 45 c7 eb de 8d 47
[10103.693528] RSP: 0018:ffff96ef008a7e10 EFLAGS: 00010246
[10103.693531] RAX: 0000000000000024 RBX: 0000000000000000 RCX: 0000000000000000
[10103.693533] RDX: 0000000000000000 RSI: ffff8be5bbc57688 RDI: ffff8be5bbc57688
[10103.693535] RBP: 0000000000000002 R08: ffff8be5bbc57688 R09: 0000000000000033
[10103.693536] R10: 5d20657265682074 R11: 2d2d2d2d2d2d2d2d R12: 0000000000000000
[10103.693538] R13: ffff96ef01171000 R14: 0000000000000097 R15: 000000000000009b
[10103.693541] FS:  0000000000000000(0000) GS:ffff8be5bbc40000(0000) knlGS:0000000000000000
[10103.693543] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[10103.693545] CR2: 0000558124d96100 CR3: 000000006ce22002 CR4: 00000000003606e0
[10103.693559] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[10103.693561] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[10103.693563] Call Trace:
[10103.693638]  rtw_hal_set_chnl_bw+0x117/0x11c [88XXau]
[10103.693687]  set_channel_bwmode+0x94/0xda [88XXau]
[10103.693736]  ? rtw_set_external_auth_status+0xf7/0xf7 [88XXau]
[10103.693776]  rtw_set_chbw_hdl+0x36/0x42 [88XXau]
[10103.693802]  rtw_cmd_thread+0x295/0x3ed [88XXau]
[10103.693810]  kthread+0xf9/0x130
[10103.693835]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[10103.693839]  ? kthread_park+0x80/0x80
[10103.693843]  ret_from_fork+0x35/0x40
[10103.693848] ---[ end trace 6d42af9fcab43d09 ]---
```

Here are more dmesg output if it can helps to identify the cause
[dmesg-fusion10.5-alfaAWUS1900-ebaed2.txt](https://github.com/aircrack-ng/rtl8812au/files/3968126/dmesg-fusion10.5-alfaAWUS1900-ebaed2.txt)

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)