aircrack-ng/rtl8812au #489: kernel 5.3.10: unable to enable monitor mode

blshkv, 2019-11-26:
using the latest commit: 3b0b7b989dd90e3debaa576ddbbbb0dedfce248b
Bus 001 Device 003: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
sys-kernel/pentoo-sources-5.3.10

command:
```
airmon-ng start wlan1

PHY     Interface       Driver          Chipset

phy0    wlan0           iwlwifi         Intel Corporation Centrino Advanced-N 6205 [Taylor Peak] (rev 34)
phy1    wlan1           88XXau          Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
 (monitor mode enabled)
Error setting channel: command failed: Device or resource busy (-16)
Error -16 likely means your card was set back to station mode by something.
Removing non-monitor wlan1 interface...
 (monitor mode disabled)
```
crash:
```
Nov 26 15:35:53 [kernel] ------------[ cut here ]------------
Nov 26 15:35:53 [kernel] WARNING: CPU: 0 PID: 2333 at /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_mlme_ext.c:12291 survey_timer_hdl+0x1df/0x26f [88XXau]
Nov 26 15:35:53 [kernel] Modules linked in: ctr ccm ipv6 nf_defrag_ipv6 uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videodev videobuf2_common 88XXau(O) usbhid dm_mod dax iwldvm mac80211 x86_pkg_temp_thermal iwlwifi kvm_intel kvm irqbypass efivars serio_raw i2c_i801 sdhci_pci cfg80211 cqhci sdhci mmc_core xhci_pci ehci_pci e1000e ehci_hcd xhci_hcd thinkpad_acpi ledtrig_audio rfkill efivarfs
Nov 26 15:35:53 [kernel] CPU: 0 PID: 2333 Comm: RTW_CMD_THREAD Tainted: G           O    T 5.3.10-pentoo #1
Nov 26 15:35:53 [kernel] Hardware name: LENOVO 2330A54/2330A54, BIOS G2ETB2WW (2.72 ) 04/11/2018
Nov 26 15:35:53 [kernel] RIP: 0010:survey_timer_hdl+0x1df/0x26f [88XXau]
Nov 26 15:35:53 [kernel] Code: ff 1e 06 00 48 8b 44 24 08 65 48 33 04 25 28 00 00 00 0f 85 9d 00 00 00 48 83 c4 10 5b c3 48 c7 c7 50 9e 9e c0 e8 99 50 79 f5 <0f> 0b e9 0b ff ff ff c6 44 24 01 00 48 8d 54 24 01 be 5a 00 00 00
Nov 26 15:35:53 [kernel] RSP: 0018:ffffa6fb406cfe60 EFLAGS: 00010246
Nov 26 15:35:53 [kernel] RAX: 0000000000000024 RBX: ffffa6fb40205000 RCX: 0000000000000000
Nov 26 15:35:53 [kernel] RDX: 0000000000000000 RSI: ffff9c4c160163c8 RDI: 00000000ffffffff
Nov 26 15:35:53 [kernel] RBP: ffff9c4c0720c200 R08: 00000000000002e9 R09: 0000000000000001
Nov 26 15:35:53 [kernel] R10: 0000000000000000 R11: 0000000000000001 R12: ffff9c4c0a28c548
Nov 26 15:35:53 [kernel] R13: ffffa6fb40206370 R14: ffff9c4c0a3fbd40 R15: ffffffffc092f120
Nov 26 15:35:53 [kernel] FS:  0000000000000000(0000) GS:ffff9c4c16000000(0000) knlGS:0000000000000000
Nov 26 15:35:53 [kernel] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Nov 26 15:35:53 [kernel] CR2: 00007f63adc5bfc8 CR3: 000000020f20a002 CR4: 00000000001606f0
Nov 26 15:35:53 [kernel] Call Trace:
Nov 26 15:35:53 [kernel]  disconnect_hdl+0x4c/0xc2 [88XXau]
Nov 26 15:35:53 [kernel]  rtw_cmd_thread+0x290/0x3e8 [88XXau]
Nov 26 15:35:53 [kernel]  ? _raw_spin_lock_irqsave+0x14/0x40
Nov 26 15:35:53 [kernel]  kthread+0xf3/0x130
Nov 26 15:35:53 [kernel]  ? rtw_stop_cmd_thread+0x34/0x34 [88XXau]
Nov 26 15:35:53 [kernel]  ? kthread_park+0x80/0x80
Nov 26 15:35:53 [kernel]  ret_from_fork+0x35/0x40
Nov 26 15:35:53 [kernel] ---[ end trace bf0ce95814e41e78 ]---

```

kimocoder, 2019-11-26:
@blshkv 

Try setting monitor mode manually with "iw" and report back please. Thanks

blshkv, 2019-11-27:
I have tried it. There is no crash after enabling it manually, however ```airodump-ng mon1``` does not show any networks. So it doesn't really work.

It also segfaults on removal:
```'iw dev mon1 del
Segmentation fault
```
```
Nov 27 09:03:32 [NetworkManager] <info>  [1574816612.0941] exiting (success)
Nov 27 09:03:32 [dbus-daemon] [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Nov 27 09:04:21 [kernel] systemd-udevd[3386]: mon1: Process 'net.sh mon1 stop' failed with exit code 1.
Nov 27 09:04:21 [kernel] =============================================================================
Nov 27 09:04:21 [kernel] BUG kmalloc-512 (Tainted: G           O    T): Object already free
Nov 27 09:04:21 [kernel] -----------------------------------------------------------------------------
Nov 27 09:04:21 [kernel] Disabling lock debugging due to kernel taint
Nov 27 09:04:21 [kernel] INFO: Allocated in alloc_netdev_mqs+0x1c5/0x3b0 age=122301 cpu=2 pid=3166
Nov 27 09:04:21 [kernel] ___slab_alloc.isra.0.constprop.0+0x38/0x50
Nov 27 09:04:21 [kernel] ___kmalloc+0x231/0x270
Nov 27 09:04:21 [kernel] _alloc_netdev_mqs+0x1c5/0x3b0
Nov 27 09:04:21 [kernel] _cfg80211_rtw_set_default_mgmt_key+0x3294/0x358f [88XXau]
Nov 27 09:04:21 [kernel] _nl80211_new_interface+0x1ca/0x450 [cfg80211]
Nov 27 09:04:21 [kernel] _genl_family_rcv_msg+0x1ee/0x470
Nov 27 09:04:21 [kernel] _genl_rcv_msg+0x42/0x90
Nov 27 09:04:21 [kernel] _netlink_rcv_skb+0x70/0x140
Nov 27 09:04:21 [kernel] _genl_rcv+0x1f/0x30
Nov 27 09:04:21 [kernel] _netlink_unicast+0x175/0x200
Nov 27 09:04:21 [kernel] _netlink_sendmsg+0x201/0x3c0
Nov 27 09:04:21 [kernel] _sock_sendmsg+0x29/0x40
Nov 27 09:04:21 [kernel] ____sys_sendmsg+0x302/0x370
Nov 27 09:04:21 [kernel] ___sys_sendmsg+0x7c/0xd0
Nov 27 09:04:21 [kernel] _do_syscall_64+0x43/0xf0
Nov 27 09:04:21 [kernel] _entry_SYSCALL_64_after_hwframe+0x44/0xa9
Nov 27 09:04:21 [kernel] INFO: Freed in free_netdev+0x13/0x110 age=0 cpu=1 pid=3385
Nov 27 09:04:21 [kernel] _free_netdev+0x13/0x110
Nov 27 09:04:21 [kernel] _rtw_ndev_destructor+0x22/0x24 [88XXau]
Nov 27 09:04:21 [kernel] _netdev_run_todo+0x1b1/0x290
Nov 27 09:04:21 [kernel] _nl80211_post_doit+0x5d/0x70 [cfg80211]
Nov 27 09:04:21 [kernel] _genl_family_rcv_msg+0x208/0x470
Nov 27 09:04:21 [kernel] _genl_rcv_msg+0x42/0x90
Nov 27 09:04:21 [kernel] _netlink_rcv_skb+0x70/0x140
Nov 27 09:04:21 [kernel] _genl_rcv+0x1f/0x30
Nov 27 09:04:21 [kernel] _netlink_unicast+0x175/0x200
Nov 27 09:04:21 [kernel] _netlink_sendmsg+0x201/0x3c0
Nov 27 09:04:21 [kernel] _sock_sendmsg+0x29/0x40
Nov 27 09:04:21 [kernel] ____sys_sendmsg+0x302/0x370
Nov 27 09:04:21 [kernel] ___sys_sendmsg+0x7c/0xd0
Nov 27 09:04:21 [kernel] _do_syscall_64+0x43/0xf0
Nov 27 09:04:21 [kernel] _entry_SYSCALL_64_after_hwframe+0x44/0xa9
Nov 27 09:04:21 [kernel] INFO: Slab 0x00000000b8efd44f objects=38 used=31 fp=0x0000000011b8ab39 flags=0x8000000000010201
Nov 27 09:04:21 [kernel] INFO: Object 0x0000000011b8ab39 @offset=16968 fp=0x0000000084c564ba
Nov 27 09:04:21 [kernel] Redzone 000000007ea1b8fe: bb bb bb bb bb bb bb bb                          ........
Nov 27 09:04:21 [kernel] Object 0000000011b8ab39: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 00000000be420548: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 00000000f37fe2c7: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 00000000bebd7486: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 00000000927275a6: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 000000002f6f6cef: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 0000000038fcd43e: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 0000000080eaa618: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 0000000086f6e1ab: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 0000000044637ef1: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 00000000360c63fe: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 00000000124f1cc6: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 000000007d115fbf: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 000000006bdd939d: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 000000002c24bb82: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 00000000ebe2b4e4: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 00000000403f5412: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 000000009102d081: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 000000000e09317e: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Nov 27 09:04:21 [kernel] Object 0000000016811d0c: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
[... 330 lines cut ...]
[... cut ...]
Nov 27 09:04:26 [kernel] RDX: 00007ffc30896490 RSI: 000000004020645d RDI: 000000000000000e
Nov 27 09:04:26 [kernel] RBP: 000000004020645d R08: 0000000000000000 R09: 0000000000000000
Nov 27 09:04:26 [kernel] R10: 00007ffc30896510 R11: 0000000000000246 R12: 0000564c4d1d3e90
Nov 27 09:04:26 [kernel] R13: 000000000000000e R14: 000000000000005c R15: 00007fd85fcddd6f
Nov 27 09:04:26 [kernel] FIX kmalloc-192: Restoring 0x00000000302e2d33-0x0000000081824061=0x6b
Nov 27 09:04:26 [kernel] FIX kmalloc-192: Marking all objects used
```

ZeroChaos-, 2019-11-27:
dupe with https://github.com/aircrack-ng/rtl8812au/issues/490

kimocoder, 2019-11-30:
The driver doesn't support VIF, that's why it crashes when you use "iw del" and "iw add" but i recognize "_timer_hdl" from not long ago, so I'll check that anyway

kimocoder, 2019-11-30:
It's been like this all the way.
Source (and a commit that fixes adding, not unplug/removing though (segr))

https://github.com/astsam/rtl8812au/pull/26

kimocoder, 2020-01-18:
@blshkv This issue should be resolved. If not, come over to this other thread at #493

closed 2020-01-18 by kimocoder