aircrack-ng/rtl8812au #490: 5.6.4.2 fails to set channel with iw

ZeroChaos-, 2019-11-26:
excerpt from airmon-ng start output
```
++ iw dev 036ac2 set channel 10
+ IW_ERROR='command failed: Device or resource busy (-16)'
+ '[' -n 'command failed: Device or resource busy (-16)' ']'
+ printf '\nError setting channel: command failed: Device or resource busy (-16)\n'

Error setting channel: command failed: Device or resource busy (-16)
+ printf 'command failed: Device or resource busy (-16)'
+ grep -q '(-16)'
+ printf 'Error -16 likely means your card was set back to station mode by something.\n'
Error -16 likely means your card was set back to station mode by something.
```
The card is in fact *not* in station mode, it is in monitor mode but set channel still doesn't work.
setting by freq also appears to not work
```
mammon ~ # iw dev 036ac2 set freq 2412     
command failed: Device or resource busy (-16)
mammon ~ # iw dev 036ac2 set freq 2412 HT20
command failed: Device or resource busy (-16)
```
setting via wext appears to work fine
```
mammon ~ # iwconfig 036ac1 channel 1
mammon ~ # iwconfig 036ac1 channel 2
```

kimocoder, 2019-11-26:
Noted. Will have a look this weekend

ZerBea, 2019-11-27:
@kimocoder do you remember my mails, a few weeks ago (screenshot with airodump-ng and hcxdumtool in terminal 1 and iw output in terminal 2)? iw set (virtual) monitor mode by NETLINK MESSAGES, but running ioctl() system calls (e.g. to change frequency) failed. I noticed that behavior on both drivers (rtl8812au and rtl8188eus).
And I've noticed that behavior vice versa. iw show manged, but interface is in monitor mode (in this case monitor mode was set by hcxdumptool). I think I have sent this screenshot  (screenshot with airodump-ng and hcxdumtool in terminal 1 and iw output in terminal 2), too.
In my opinion NETLINK is not a good choice to set monitor mode:
https://github.com/aircrack-ng/rtl8812au/issues/476#issuecomment-558523331

ZerBea, 2019-11-27:
Setting monitor mode by ioctl() system calls is very simple and can be done, running three simple calls.
set interface down:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L3952
set monitor mode:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L3966
set interface up:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L3985
Between this calls are some functions to test/make sure is interface is really in monitor mode.

At this point:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L4020
we have a full working monitor mode interface that can be addressed by AF_PACKET socket.
Setting channel by ioctl() is working fine, too:

victorhooi, 2019-11-27:
I think I might be hitting this issue as well. This is on a Raspberry Pi3+, running Raspbian Buster.

I've successfully installed the latest branch of the rtl8812au driver:

```
git clone https://github.com/aircrack-ng/rtl8812au.git
cd rtl8812au
sed -i ' s/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
sed -i 's/^dkms build/ARCH=arm dkms build/' dkms-install.sh
sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
sudo ./dkms-install.sh
```

However, when I try to start a new monitoring interface, I get a "Device or resource busy (-16)" message:
```
$ sudo airmon-ng check kill

Killing these processes:

  PID Name
  323 wpa_supplicant
  403 wpa_supplicant
  422 wpa_supplicant
  530 dhcpcd

$ sudo airmon-ng start wlan1

PHY	Interface	Driver		Chipset

phy0	wlan0		brcmfmac	Broadcom 43430
phy1	wlan1		88XXau		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
 (monitor mode enabled)
Error setting channel: command failed: Device or resource busy (-16)
Error -16 likely means your card was set back to station mode by something.
Removing non-monitor wlan1 interface...
```

Any ideas what's going on?

ZeroChaos-, 2019-11-27:
I wrote airmon-ng so I can say that the first comment from me shows exactly what is going on, and that is that channel setting doesn't work with iw and that is expected.  it does work with iwconfig, so you can just "iwconfig wlan1 mode monitor" and it "works" if you want.  This bug is requesting that channel control via iw be fixed/implemented.

The best and most reliable version I've tested is still 5.2.20 branch.  I just completed some basic tests of it yesterday and it was working well.

Again, this bug is specific to requesting nl80211 channel setting support.

ZerBea, 2019-11-27:
Taking a closer look at the driver code here:
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/os_dep/linux/ioctl_linux.c#L1360
I noticed that I don't reach
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/os_dep/linux/ioctl_linux.c#L1382
if monitor mode was set by iw via NETLINK MESSAGE,
either we go to exit here:
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/os_dep/linux/ioctl_linux.c#L1371
or here:
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/os_dep/linux/ioctl_linux.c#L1376
because parts of the init failed.

kimocoder, 2019-11-27:
I'll take a look. The TODO is pretty full, one more things on the list is find out where the frame injection code (and iterator) wen't after adding it to v5.2.20(.2), Realtek hided it somewhere, and it's unstable. I just need to diff branch v5.2.20 versus v5.3.4 and see what they did with it then I'm game

ZerBea, 2019-11-28:
Great, thnkns. Than we can take the"modified/new" code for rtl8188eus, too, because they are very similar.

kimocoder, 2019-11-28:
Yes they're basicly the same sort of :+1:

ZerBea, 2019-11-28:
Unfortunately, I have not found out, yet, why the initialization process of the device is different, when we use NL80211_IFTYPE_MONITOR (done by iw via NETLINK)
instead of IW_MODE_MONITOR (done by iwconfig and hcxdumptool)
If that failed, we're going back to IW_MODE_MASTER.

kimocoder, 2020-01-25:
Channel setting works with "iw" now, BUT.. the "iw dev channel switch" doesn't work though.
Closing this, setting the channel is operative, but switching seems off (unsure, barely tested here).

closed 2020-01-25 by kimocoder