aircrack-ng/rtl8812au #491: Compilation fails on raspberry pi c

solsticedhiver, 2019-11-27:
When compiling on rpi1, I got the error;

```
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.85-1-ARCH/build M=/var/lib/dkms/rtl88xxau/r972.2e294ae/build  modules
make[1]: Entering directory '/usr/lib/modules/4.19.85-1-ARCH/build'
  CC [M]  /var/lib/dkms/rtl88xxau/r972.2e294ae/build/core/rtw_cmd.o
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
make[2]: *** [scripts/Makefile.build:304: /var/lib/dkms/rtl88xxau/r972.2e294ae/build/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1525: _module_/var/lib/dkms/rtl88xxau/r972.2e294ae/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.19.85-1-ARCH/build'
make: *** [Makefile:2245: modules] Error 2

```

main error being **error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together**

kimocoder, 2019-11-27:
I'll set up all my RPI's this weekend and will fix 'em all. Just hold on few days

solsticedhiver, 2019-11-27:
Ok. It was working fine some time ago like 467f8ff

bartgrefte, 2019-12-02:
Same problem on a RPi 2 with Raspbian Buster, kernel 4.19.75-v7+.

Chisight, 2019-12-29:
Part of the problem seems to be with the Makefile:
# Raspbian kernel is with soft-float.
# 'softfp' allows FP instructions, but no FP on function call interfaces
EXTRA_CFLAGS += -mfloat-abi=softfp
This is false, Raspbian exists only because Debian used softfp in their earlier releases and Raspbian was the hardfp recompile.
http://raspbian.com/RaspbianFAQ#What_is_Raspbian.3F
Unfortunately, fixing this Makefile doesn't resolve the problem.

netskink, 2020-02-19:
i'm trying to build for a pi4 a rtl8812au.  I noticed in the makefile there is a CONFIG_RTL8814A=y switch, which later seems to enable the CONFIG_MP_VHT_HW_TX_MODE setting.  I set the config above to =n and rebuilt.  I'm trying this because building as is while working, does not work later when I try to set it as an access point using hostapd.  I won't get any errors building or loading the module.  It even seems to connect to wifi as a client, but doing the same setup as an x86 laptop with this same usb dongle works.

solsticedhiver, 2020-07-16:
This has been fixed, no ?

closed 2020-07-16 by solsticedhiver