aircrack-ng/rtl8812au #493: Unable to run "airmon-ng start" - Error setting channel: command failed: Device or resource busy (-16)

victorhooi, 2019-11-28:
I have a Raspberry Pi3+, running Raspbian Buster.

I've successfully installed the latest branch of the rtl8812au driver:

```
git clone https://github.com/aircrack-ng/rtl8812au.git
cd rtl8812au
sed -i ' s/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
pi@rp3-ang:~/code/rtl8812au $ sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
sed -i 's/^dkms build/ARCH=arm dkms build/' dkms-install.sh
sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
sudo ./dkms-install.sh
```

However, when I try to start a new monitoring interface, I get a "Device or resource busy (-16)" message:
```
pi@rp3-ang:~/code/PwrDeauther $ sudo airmon-ng check kill

Killing these processes:

  PID Name
  323 wpa_supplicant
  403 wpa_supplicant
  422 wpa_supplicant
  530 dhcpcd

pi@rp3-ang:~/code/PwrDeauther $
pi@rp3-ang:~/code/PwrDeauther $
pi@rp3-ang:~/code/PwrDeauther $ sudo airmon-ng start wlan1

PHY	Interface	Driver		Chipset

phy0	wlan0		brcmfmac	Broadcom 43430
phy1	wlan1		88XXau		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
 (monitor mode enabled)
Error setting channel: command failed: Device or resource busy (-16)
Error -16 likely means your card was set back to station mode by something.
Removing non-monitor wlan1 interface...
```

Any ideas what's going on?

blshkv, 2019-11-30:
dup of https://github.com/aircrack-ng/rtl8812au/issues/489

dront78, 2020-01-05:
same on 4.19.92 archlinux with rtl8814au

blshkv, 2020-01-06:
btw, you should start with killing of running processes as it is highlighted in the output

kimocoder, 2020-01-18:
Yeah. You will have to kill those processes running with interference, and I noticed sometimes, run the 'airmon-ng start <adapter>' command two (2) times, but I run it on all my devices here. RPI 1,2,3,4, Android, computers .. So I know it should be ok now. On newer kernels after 5.4.x the delay on injection/laggish behavior is gone too.

kimocoder, 2020-01-18:
@victorhooi and I need more outputs.. "airmon-ng" (screenshot), "modinfo 88XXau > modinfo.txt" "dmesg > dmesg.txt" would be a good start

blshkv, 2020-01-19:
I'm facing this issue with kernel 5.3 only (regression). I migrated to 5.4.13 and it started working again.

closed 2020-01-25 by kimocoder

yusma1122, 2023-12-14:
omg same case here, is this problem has been resolved ?