aircrack-ng/rtl8812au #494: general-regs-only (non dkms)

howudodat, 2019-11-29:
getting the below error when building on a Gateworks Newport (arm64)

```
root@bionic-newport:~/rtl8812au# make -j4
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.14.4-g9945530/build M=/root/rtl8812au  modules
make[1]: Entering directory '/root/linux-newport'
...
/root/rtl8812au/core/rtw_mp.c: In function 'rtfloor':
/root/rtl8812au/core/rtw_mp.c:31:5: error: '-mgeneral-regs-only' is incompatible with floating-point argument
 int rtfloor(float x)
     ^~~~~~~

```
I saw the instructions in the readme, but that seems to be for dkms and not for the Makefile
```
$ sed -i 's/^dkms build/ARCH=arm dkms build/' dkms-install.sh
$ sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
```

jstruebel, 2020-01-03:
I am getting this error also on the latest commit in the v5.3.4 branch building with dkms for a NanoPi NEO2 (arm64) under Armbian Buster with kernel 4.19.63-sunxi64.

I'm using the CONFIG_PLATFORM_ARM64_RPI = y option in the Makefile.

Per the information in https://github.com/aircrack-ng/rtl8812au/issues/230#issuecomment-451482192 I have the EXTRA_CFLAGS += -mfloat-abi=softfp option disabled (I added ifeq ($(CONFIG_PLATFORM_ARM_RPI), y) to the else clause instead of commenting it out).

I've tried searching through the code on github and I can't find where that option is getting set. Any help is greatly appreciated.

kimocoder, 2020-01-05:
This issue should be resolved. Please update, compile/build/install/reboot and report back. Thanks

closed 2020-01-05 by kimocoder

jstruebel, 2020-01-05:
Build successful and iw list now provides information. Thank you!