aircrack-ng/rtl8812au #495: missing sys/socket.h?

antonvh, 2019-11-29:
Compiling on a Rpi 1, running debian stretch. 

I symlinked /usr/src/linux-headers-4.14.114-ev3dev-2.3.4-rpi into /lib/modules/4.14.114-ev3dev-2.3.4-rpi/build

There's a socket.h here:
/usr/include/arm-linux-gnueabihf/sys/

Should I point the compiler there somehow? How?

```
robot@ev3dev:~/rtl8812au$ sudo make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.14.114-ev3dev-2.3.4-rpi/build M=/home/robot/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.14.114-ev3dev-2.3.4-rpi'
  CC [M]  /home/robot/rtl8812au/core/rtw_cmd.o
In file included from ./include/linux/netfilter.h:8:0,
                 from ./include/linux/netfilter/nf_conntrack_tuple_common.h:6,
                 from ./include/linux/netfilter/nf_conntrack_dccp.h:29,
                 from ./include/net/netns/conntrack.h:11,
                 from ./include/net/net_namespace.h:27,
                 from ./include/linux/netdevice.h:42,
                 from /home/robot/rtl8812au/include/osdep_service_linux.h:30,
                 from /home/robot/rtl8812au/include/osdep_service.h:54,
                 from /home/robot/rtl8812au/include/drv_types.h:26,
                 from /home/robot/rtl8812au/core/rtw_cmd.c:17:
./include/linux/if.h:28:54: fatal error: sys/socket.h: No such file or directory
 #include <sys/socket.h>   /* for struct sockaddr.  */
                                                      ^
compilation terminated.
scripts/Makefile.build:326: recipe for target '/home/robot/rtl8812au/core/rtw_cmd.o' failed
make[2]: *** [/home/robot/rtl8812au/core/rtw_cmd.o] Error 1
Makefile:1534: recipe for target '_module_/home/robot/rtl8812au' failed
make[1]: *** [_module_/home/robot/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.14.114-ev3dev-2.3.4-rpi'
Makefile:2245: recipe for target 'modules' failed
make: *** [modules] Error 2

```

ZerBea, 2019-11-29:
https://www.raspberrypi.org/forums/viewtopic.php?t=230541
should answer your question.

Maybe Debian has a good reason to split binaries and development packages....

kimocoder, 2019-12-16:
@ZerBea has right. Follow the RPI forums for more information. "apt-get install libc6-dev"

closed 2019-12-16 by kimocoder

antonvh, 2019-12-16:
That doesn’t work for me. I did that.

ZerBea, 2019-12-16:
Is the include path ok?
https://gcc.gnu.org/onlinedocs/cpp/Search-Path.html

is socket.h where we expected it: /usr/include/sys/socket.h ?

$ find /usr/include/ -name socket.h
/usr/include/asm-generic/socket.h
/usr/include/wireshark/wsutil/socket.h
/usr/include/libnl3/netlink/socket.h
/usr/include/gtkmm-3.0/gtkmm/socket.h
/usr/include/guile/2.2/libguile/socket.h
/usr/include/giomm-2.4/giomm/socket.h
/usr/include/gnutls/socket.h
/usr/include/bits/socket.h
/usr/include/linux/socket.h
/usr/include/wx-3.0/wx/socket.h
/usr/include/sys/socket.h
/usr/include/pcap/socket.h
/usr/include/asm/socket.h