aircrack-ng/rtl8812au #498: Possible dead lock with v5.6.4.2 + kernel 5.3.13

yshui, 2019-11-30:
Processes start to hang almost immediately after the module is loaded.

Call trace:

```
Nov 30 00:04:22 node-02 kernel: sysrq: Show Blocked State
Nov 30 00:04:22 node-02 kernel:   task                        PC stack   pid father
Nov 30 00:04:22 node-02 kernel: iwd             D    0  1158      1 0x000041a0
Nov 30 00:04:22 node-02 kernel: Call Trace:
Nov 30 00:04:22 node-02 kernel:  ? __schedule+0x27f/0x6d0
Nov 30 00:04:22 node-02 kernel:  schedule+0x43/0xd0
Nov 30 00:04:22 node-02 kernel:  schedule_preempt_disabled+0x14/0x20
Nov 30 00:04:22 node-02 kernel:  __mutex_lock.isra.0+0x27d/0x530
Nov 30 00:04:22 node-02 kernel:  wiphy_unregister+0x63/0x380 [cfg80211]
Nov 30 00:04:22 node-02 kernel:  rtw_wiphy_unregister+0xa/0xb [88XXau]
Nov 30 00:04:22 node-02 kernel:  rtw_os_ndev_unregister+0x6a/0xb1 [88XXau]
Nov 30 00:04:22 node-02 kernel:  cfg80211_rtw_del_virtual_intf+0x4d/0xa2 [88XXau]
Nov 30 00:04:22 node-02 kernel:  nl80211_del_interface+0x4d/0x130 [cfg80211]
Nov 30 00:04:22 node-02 kernel:  genl_family_rcv_msg+0x1f3/0x470
Nov 30 00:04:22 node-02 kernel:  genl_rcv_msg+0x47/0x90
Nov 30 00:04:22 node-02 kernel:  ? __kmalloc_node_track_caller+0x1c5/0x330
Nov 30 00:04:22 node-02 kernel:  ? genl_family_rcv_msg+0x470/0x470
Nov 30 00:04:22 node-02 kernel:  netlink_rcv_skb+0x75/0x140
Nov 30 00:04:22 node-02 kernel:  genl_rcv+0x24/0x40
Nov 30 00:04:22 node-02 kernel:  netlink_unicast+0x177/0x1f0
Nov 30 00:04:22 node-02 kernel:  netlink_sendmsg+0x204/0x3d0
Nov 30 00:04:22 node-02 kernel:  sock_sendmsg+0x5e/0x60
Nov 30 00:04:22 node-02 kernel:  __sys_sendto+0x120/0x190
Nov 30 00:04:22 node-02 kernel:  __x64_sys_sendto+0x25/0x30
Nov 30 00:04:22 node-02 kernel:  do_syscall_64+0x4e/0x110
Nov 30 00:04:22 node-02 kernel:  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Nov 30 00:04:22 node-02 kernel: RIP: 0033:0x7f7d1c04d700
Nov 30 00:04:22 node-02 kernel: Code: Bad RIP value.
Nov 30 00:04:22 node-02 kernel: RSP: 002b:00007fffe048ef78 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
Nov 30 00:04:22 node-02 kernel: RAX: ffffffffffffffda RBX: 000055bea4b91890 RCX: 00007f7d1c04d700
Nov 30 00:04:22 node-02 kernel: RDX: 0000000000000028 RSI: 000055bea4ba4560 RDI: 0000000000000004
Nov 30 00:04:22 node-02 kernel: RBP: 000055bea4ba1e20 R08: 0000000000000000 R09: 0000000000000000
Nov 30 00:04:22 node-02 kernel: R10: 0000000000000000 R11: 0000000000000246 R12: 00007fffe048efd0
Nov 30 00:04:22 node-02 kernel: R13: 00007fffe048efcc R14: 000055bea4b97380 R15: 000055bea4b917a8
Nov 30 00:04:22 node-02 kernel: systemd-udevd   D    0 11883    762 0x000041a0
Nov 30 00:04:22 node-02 kernel: Call Trace:
Nov 30 00:04:22 node-02 kernel:  ? __schedule+0x27f/0x6d0
Nov 30 00:04:22 node-02 kernel:  schedule+0x43/0xd0
Nov 30 00:04:22 node-02 kernel:  schedule_preempt_disabled+0x14/0x20
Nov 30 00:04:22 node-02 kernel:  __mutex_lock.isra.0+0x27d/0x530
Nov 30 00:04:22 node-02 kernel:  dev_ioctl+0x3aa/0x576
Nov 30 00:04:22 node-02 kernel:  sock_do_ioctl+0xee/0x190
Nov 30 00:04:22 node-02 kernel:  sock_ioctl+0x281/0x3f0
Nov 30 00:04:22 node-02 kernel:  do_vfs_ioctl+0x43d/0x6c0
Nov 30 00:04:22 node-02 kernel:  ? syscall_trace_enter+0x1f2/0x2e0
Nov 30 00:04:22 node-02 kernel:  ksys_ioctl+0x5e/0x90
Nov 30 00:04:22 node-02 kernel:  __x64_sys_ioctl+0x16/0x20
Nov 30 00:04:22 node-02 kernel:  do_syscall_64+0x4e/0x110
Nov 30 00:04:22 node-02 kernel:  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Nov 30 00:04:22 node-02 kernel: RIP: 0033:0x7f5123f4f25b
Nov 30 00:04:22 node-02 kernel: Code: Bad RIP value.
Nov 30 00:04:22 node-02 kernel: RSP: 002b:00007ffe68354778 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
Nov 30 00:04:22 node-02 kernel: RAX: ffffffffffffffda RBX: 0000557d81a2c8a8 RCX: 00007f5123f4f25b
Nov 30 00:04:22 node-02 kernel: RDX: 00007ffe68354780 RSI: 0000000000008946 RDI: 0000000000000006
Nov 30 00:04:22 node-02 kernel: RBP: 00007ffe68354938 R08: 0000557d81be92f0 R09: 0000000000000000
Nov 30 00:04:22 node-02 kernel: R10: 00007f512312b7c0 R11: 0000000000000246 R12: 0000557d81be92f0
Nov 30 00:04:22 node-02 kernel: R13: 00007ffe68354780 R14: 0000557d81bea200 R15: 0000000000000007
Nov 30 00:04:22 node-02 kernel: sudo            D    0 11889   1840 0x00000080
Nov 30 00:04:22 node-02 kernel: Call Trace:
Nov 30 00:04:22 node-02 kernel:  ? __schedule+0x27f/0x6d0
Nov 30 00:04:22 node-02 kernel:  schedule+0x43/0xd0
Nov 30 00:04:22 node-02 kernel:  schedule_preempt_disabled+0x14/0x20
Nov 30 00:04:22 node-02 kernel:  __mutex_lock.isra.0+0x27d/0x530
Nov 30 00:04:22 node-02 kernel:  __netlink_dump_start+0x54/0x1e0
Nov 30 00:04:22 node-02 kernel:  ? rtnl_fill_ifinfo+0x1020/0x1020
Nov 30 00:04:22 node-02 kernel:  rtnetlink_rcv_msg+0x296/0x3c0
Nov 30 00:04:22 node-02 kernel:  ? rtnl_fill_ifinfo+0x1020/0x1020
Nov 30 00:04:22 node-02 kernel:  ? rtnl_calcit.isra.0+0x120/0x120
Nov 30 00:04:22 node-02 kernel:  netlink_rcv_skb+0x75/0x140
Nov 30 00:04:22 node-02 kernel:  netlink_unicast+0x177/0x1f0
Nov 30 00:04:22 node-02 kernel:  netlink_sendmsg+0x204/0x3d0
Nov 30 00:04:22 node-02 kernel:  sock_sendmsg+0x5e/0x60
Nov 30 00:04:22 node-02 kernel:  __sys_sendto+0x120/0x190
Nov 30 00:04:22 node-02 kernel:  __x64_sys_sendto+0x25/0x30
Nov 30 00:04:22 node-02 kernel:  do_syscall_64+0x4e/0x110
Nov 30 00:04:22 node-02 kernel:  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Nov 30 00:04:22 node-02 kernel: RIP: 0033:0x7f3d9f2ef85a
Nov 30 00:04:22 node-02 kernel: Code: Bad RIP value.
Nov 30 00:04:22 node-02 kernel: RSP: 002b:00007ffc6979e928 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
Nov 30 00:04:22 node-02 kernel: RAX: ffffffffffffffda RBX: 00007ffc6979fa90 RCX: 00007f3d9f2ef85a
Nov 30 00:04:22 node-02 kernel: RDX: 0000000000000014 RSI: 00007ffc6979f9c0 RDI: 0000000000000003
Nov 30 00:04:22 node-02 kernel: RBP: 00007ffc6979fa10 R08: 00007ffc6979f980 R09: 000000000000000c
Nov 30 00:04:22 node-02 kernel: R10: 0000000000000000 R11: 0000000000000246 R12: 00007ffc6979f980
Nov 30 00:04:22 node-02 kernel: R13: 00007ffc6979f9c0 R14: 0000000000000000 R15: 00007ffc6979e930
```

yshui, 2019-11-30:
Looks like I am not the only one: https://aur.archlinux.org/packages/rtl88xxau-aircrack-dkms-git#comment-716905

This user reports the bad commit is somewhere between 73f0a88 and a8ec357. If I could find time, I would try bisecting it.

kimocoder, 2019-11-30:
I don't know anything about other OS'en than Debian and Kali / NetHunter, so support for them is allways a challenge, also to keep all compiler out there satisfied 💯

terencode, 2019-11-30:
Same as #486 I think.

mokkurkalve, 2019-12-02:
Hello, I'm the user  yshui are referring to above. I still have same problem with latest commit, it hangs my system. r918.73f0a88 works fine. I'm on Arch Linux x86_64 current with kernel 5.3.13-arch1-1 and compiler  gcc 9.2.0-4. I could try to help out debugging this if somebody gives me some pointers as to what to do (holds my hand most likely...).

mokkurkalve, 2020-01-12:
I tried again with r1002.34258e7 (with linux 5.4.10.arch1-1 and gcc 9.2.0-4). Same problem, various system functions crashes or freezes to the point where system is unusable. Reverting to r918.73f0a88, and all is good.

ericsaintetienne, 2020-01-13:
73f0a88 still exhibit the same kernel oops:
```
[   28.429410] ------------[ cut here ]------------
[   28.429555] WARNING: CPU: 2 PID: 2362 at net/wireless/nl80211.c:3153 nl80211_send_chandef+0x14b/0x160 [cfg80211]
[   28.429714] Modules linked in: xt_MASQUERADE nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_REJECT nf_reject_ipv6 ip6t_rpfilter ipt_REJECT nf_reject_ipv4 xt_conntrack ebtable_nat ebtable_broute ip6table_nat ip6table_mangle i
p6table_raw ip6table_security iptable_nat nf_nat iptable_mangle iptable_raw iptable_security nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_set nfnetlink ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter overlay intel_rapl_
msr intel_rapl_common raid456 async_raid6_recov async_memcpy x86_pkg_temp_thermal async_pq intel_powerclamp async_xor coretemp async_tx xor kvm_intel raid6_pq raid0 kvm irqbypass 88XXau(OE) crct10dif_pclmul crc32_pclmul iTCO_wdt ghash_clm
ulni_intel iTCO_vendor_support intel_cstate mei_hdcp eeepc_wmi intel_uncore cfg80211 asus_wmi joydev sparse_keymap mxm_wmi i2c_i801 intel_rapl_perf wmi_bmof mei_me mei acpi_pad nfsd auth_rpcgss nfs_acl lockd grace sunrpc ip_tables xfs lib
crc32c rfkill hid_logitech_hidpp hid_logitech_dj i915
[   28.429811]  i2c_algo_bit drm_kms_helper mptsas drm crc32c_intel scsi_transport_sas r8169 mptscsih serio_raw mptbase wmi video
[   28.431366] CPU: 2 PID: 2362 Comm: RTW_CMD_THREAD Tainted: G           OE     5.4.8-200.fc31.x86_64 #1
[   28.431530] Hardware name: System manufacturer System Product Name/H110I-PLUS D3, BIOS 1802 05/26/2016
[   28.431747] RIP: 0010:nl80211_send_chandef+0x14b/0x160 [cfg80211]
[   28.431851] Code: 00 00 be a1 00 00 00 48 89 ef 89 44 24 04 e8 0c ca 8c cd 85 c0 0f 84 7b ff ff ff 41 bc 97 ff ff ff e9 70 ff ff ff 31 c0 eb a7 <0f> 0b 41 bc ea ff ff ff e9 5f ff ff ff e8 33 ec 48 cd 0f 1f 00 0f
[   28.432137] RSP: 0018:ffffb89841453d40 EFLAGS: 00010246
[   28.432231] RAX: 0000000000000000 RBX: ffffb89841453dc8 RCX: 0000000000000098
[   28.432356] RDX: 0000000078e12300 RSI: 00000000ffff30bc RDI: ffffb89841453dc8
[   28.432477] RBP: ffff9343dcdd4400 R08: 0000000000000eac R09: ffff9343e4c6401c
[   28.432597] R10: 0000000000000000 R11: 0000000000000000 R12: ffffb89841453dc8
[   28.432715] R13: 0000000000000000 R14: ffff9343dcdd4400 R15: ffff9343e4c64014
[   28.433708] FS:  0000000000000000(0000) GS:ffff9343e6a80000(0000) knlGS:0000000000000000
[   28.434574] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   28.435488] CR2: 00007f7034000010 CR3: 000000021260a003 CR4: 00000000003606e0
[   28.436373] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[   28.437278] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[   28.438170] Call Trace:
[   28.439033]  nl80211_ch_switch_notify.constprop.0+0xcd/0x170 [cfg80211]
[   28.439934]  rtw_cfg80211_ch_switch_notify+0x138/0x147 [88XXau]
[   28.440861]  rtw_chk_start_clnt_join+0x96/0x157 [88XXau]
[   28.441752]  ? rtw_chk_start_clnt_join+0x157/0x157 [88XXau]
[   28.442676]  join_cmd_hdl+0x267/0x373 [88XXau]
[   28.443582]  rtw_cmd_thread+0x44a/0x611 [88XXau]
[   28.444496]  kthread+0xf9/0x130
[   28.445397]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[   28.446266]  ? kthread_park+0x90/0x90
[   28.447216]  ret_from_fork+0x35/0x40
[   28.448106] ---[ end trace d4125400b44d74d9 ]---
```

Plus it outputs a lot of debug on the kernel buffer, all starting with `RTW:` for example:
```
[   28.449462] RTW: ==>PHY_SwitchWirelessBand8814() 5G
[   28.451604] RTW: <==PHY_SwitchWirelessBand8814():Switch Band OK.
[   28.453017] RTW: SCMapping: pHalData->current_channel_bw 1, pHalData->nCur40MhzPrimeSC 1 
[   28.577809] RTW: OnBeacon: beacon keys ready
[   28.578744] RTW: link to Broadcom AP
[   28.579718] RTW: start auth
[   28.580628] RTW: issue_auth
[   28.581511] RTW: +xmitframe_complete
[   28.583527] RTW: OnAuthClient
[   28.584500] RTW: auth success, start assoc
[   28.585432] RTW: network.SupportedRates[0]=8C
[   28.586354] RTW: network.SupportedRates[1]=12
[   28.587227] RTW: network.SupportedRates[2]=98
[   28.588126] RTW: network.SupportedRates[3]=24
[   28.588976] RTW: network.SupportedRates[4]=B0
[   28.589767] RTW: network.SupportedRates[5]=48
[   28.590560] RTW: network.SupportedRates[6]=60
[   28.591355] RTW: network.SupportedRates[7]=6C
[   28.592209] RTW: bssrate_len = 8
[   28.593058] RTW: +xmitframe_complete
[   28.594987] RTW: OnAssocRsp
```
an much more!
That's on Fedora 31 with kernel version `5.4.8-200.fc31.x86_64`, 88XXau module build using `dkms`.

mokkurkalve, 2020-01-14:
Mm. I cherrypicked a patch to turn off debug. But r918.73f0a88 have been good to me this far. The problems occurs from some later commit I haven't been able to pinpoint.
My details are here: https://aur.archlinux.org/packages/rtl88xxau-aircrack-dkms-git/#comment-716905

yshui, 2020-02-14:
OK, this is going to sound ridiculous, but I am able to make the dead lock go away by making this change:

```diff
diff --git a/Makefile b/Makefile
--- a/Makefile
+++ b/Makefile
@@ -189,7 +189,7 @@ endif

 ifeq ($(CONFIG_RTL8812A)_$(CONFIG_RTL8821A)_$(CONFIG_RTL8814A), y_y_y)

-EXTRA_CFLAGS += -DDRV_NAME=\"88XXau\"
+EXTRA_CFLAGS += -DDRV_NAME=\"rtl88XXau\"
 ifeq ($(CONFIG_USB_HCI), y)
 USER_MODULE_NAME = 88XXau
 endif
```
I have no idea why.

<details>

<summary>More information on the deadlock</summary>

```
============================================
WARNING: possible recursive locking detected
5.5.3-locktest #29 Tainted: G           O    T
--------------------------------------------
iwd/1355 is trying to acquire lock:
ffffffffabb2da20 (rtnl_mutex){+.+.}, at: wiphy_unregister+0x35/0x460

but task is already holding lock:
ffffffffabb2da20 (rtnl_mutex){+.+.}, at: nl80211_pre_doit+0x100/0x1a0

other info that might help us debug this:
 Possible unsafe locking scenario:
       CPU0
       ----
  lock(rtnl_mutex);
  lock(rtnl_mutex);

 *** DEADLOCK ***
 May be due to missing lock nesting notation
2 locks held by iwd/1355:
 #0: ffffffffabb337c8 (cb_lock){++++}, at: genl_rcv+0x10/0x30
 #1: ffffffffabb2da20 (rtnl_mutex){+.+.}, at: nl80211_pre_doit+0x100/0x1a0

stack backtrace:
CPU: 1 PID: 1355 Comm: iwd Tainted: G           O    T 5.5.3-locktest #29
Hardware name: System manufacturer System Product Name/ROG STRIX X470-F GAMING, BIOS 5406 11/13/2019
Call Trace:
 dump_stack+0x71/0xa0
 validate_chain.cold+0x122/0x15f
 __lock_acquire+0x3a2/0x790
 lock_acquire+0xbf/0x1f0
 ? wiphy_unregister+0x35/0x460
 __mutex_lock+0x88/0x910
 ? wiphy_unregister+0x35/0x460
 ? lockdep_hardirqs_on+0xea/0x180
 ? wiphy_unregister+0x35/0x460
 wiphy_unregister+0x35/0x460
 ? rtw_cfg80211_indicate_scan_done+0x6e/0x95 [88XXau]
 rtw_wiphy_unregister+0x15/0x3f [88XXau]
 rtw_os_ndev_unregister+0x65/0xad [88XXau]
 cfg80211_rtw_del_virtual_intf+0x6a/0x114 [88XXau]
 nl80211_del_interface+0x5f/0x220
 genl_rcv_msg+0x18d/0x400
 ? genl_family_rcv_msg_attrs_parse+0x130/0x130
 netlink_rcv_skb+0x44/0x110
 genl_rcv+0x1f/0x30
 netlink_unicast+0x173/0x250
 netlink_sendmsg+0x232/0x450
 __sys_sendto+0x1cc/0x1f0
 ? lockdep_hardirqs_on+0xea/0x180
 ? syscall_trace_enter+0x17c/0x340
 ? syscall_trace_enter+0x17c/0x340
 __x64_sys_sendto+0x20/0x30
 do_syscall_64+0x59/0x270
 entry_SYSCALL_64_after_hwframe+0x49/0xbe
```

</details>

Looks to me this deadlock would _always_ happen, even in v5.3.4. But it doesn't. 

I bisected the offending commit to fa35b99b425bc93ed, and further narrowed it down to the `DRV_NAME` change.

I am currently completely discombobulated.

yshui, 2020-02-14:
One possible explanation is that `cfg80211_rtw_del_virtual_intf` is just not called at all when `DRV_NAME` is not `88XXau`. From what I can tell that `iwd` is trying to delete the interface. Maybe it only does that when the driver name is `88XXau`?

yshui, 2020-02-14:
Ah, I was correct. `iwd` has [special treatment](https://git.kernel.org/pub/scm/network/wireless/iwd.git/tree/src/manager.c#n81) for drivers named `rtl88*` to stop them from crashing the kernel, because they are just so broken.

mokkurkalve, 2020-02-15:
yshui's advice to add
`[General]
UseDefaultInterface=true`
to /etc/iwd/main.conf
seem to have solved my long standing problem. :-)

aep, 2021-01-08:
73f0a88 doesnt compile for me, and openwrt does not use iwd.

did anyone find a different workaround for this yet?

ohhai, 2021-02-10:
Just saw something similar on Fedora 33, kernel 5.10.13-200.fc33.x86_64+debug
Using branch v5.6.4.2 on commit 059e06a "Merge pull request #814 from CGarces/patch-3"
Happens immediately after module insert

```
[   31.536044] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)
[ 4998.068952] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[ 4998.069390] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[ 4998.074561] 88XXau: loading out-of-tree module taints kernel.
[ 4998.075242] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
[ 4998.910343] usb 3-2.3: 88XXau 08:10:7b:6d:5b:4e hw_info[d7]
[ 4998.925535] usbcore: registered new interface driver rtl88XXau
[ 4998.947132] rtl88XXau 3-2.3:1.0 wlp7s0f3u2u3: renamed from wlan0
[ 4999.977650] ============================================
[ 4999.977652] WARNING: possible recursive locking detected
[ 4999.977656] 5.10.13-200.fc33.x86_64+debug #1 Tainted: G           OE
[ 4999.977657] --------------------------------------------
[ 4999.977660] ifconfig/12762 is trying to acquire lock:
[ 4999.977662] ffff978a4361d570 (pmutex){+.+.}-{3:3}, at: usbctrl_vendorreq+0x9a/0x290 [88XXau]
[ 4999.977707] 
               but task is already holding lock:
[ 4999.977709] ffff978a4361c110 (pmutex){+.+.}-{3:3}, at: netdev_open+0x31/0x60 [88XXau]
[ 4999.977740] 
               other info that might help us debug this:
[ 4999.977742]  Possible unsafe locking scenario:

[ 4999.977743]        CPU0
[ 4999.977744]        ----
[ 4999.977745]   lock(pmutex);
[ 4999.977748]   lock(pmutex);
[ 4999.977750] 
                *** DEADLOCK ***

[ 4999.977751]  May be due to missing lock nesting notation

[ 4999.977753] 2 locks held by ifconfig/12762:
[ 4999.977754]  #0: ffffffffa7fe0df0 (rtnl_mutex){+.+.}-{3:3}, at: devinet_ioctl+0xa6/0x8f0
[ 4999.977760]  #1: ffff978a4361c110 (pmutex){+.+.}-{3:3}, at: netdev_open+0x31/0x60 [88XXau]
[ 4999.977788] 
               stack backtrace:
[ 4999.977791] CPU: 4 PID: 12762 Comm: ifconfig Tainted: G           OE     5.10.13-200.fc33.x86_64+debug #1
[ 4999.977793] Hardware name: ASUS System Product Name/TUF GAMING B550M-PLUS, BIOS 1401 12/03/2020
[ 4999.977794] Call Trace:
[ 4999.977799]  dump_stack+0x8b/0xb0
[ 4999.977803]  __lock_acquire.cold+0x159/0x2ab
[ 4999.977808]  lock_acquire+0xbc/0x360
[ 4999.977833]  ? usbctrl_vendorreq+0x9a/0x290 [88XXau]
[ 4999.977858]  ? usbctrl_vendorreq+0x9a/0x290 [88XXau]
[ 4999.977861]  __mutex_lock+0x7b/0x7e0
[ 4999.977883]  ? usbctrl_vendorreq+0x9a/0x290 [88XXau]
[ 4999.977886]  ? __lock_acquire+0x3ae/0x1eb0
[ 4999.977909]  usbctrl_vendorreq+0x9a/0x290 [88XXau]
[ 4999.977944]  usb_read8+0x2a/0x40 [88XXau]
[ 4999.977975]  rtl8812au_hal_init+0x54/0x1070 [88XXau]
[ 4999.978008]  rtw_hal_init+0x1c/0xd0 [88XXau]
[ 4999.978037]  _netdev_open+0x51/0x230 [88XXau]
[ 4999.978062]  netdev_open+0x39/0x60 [88XXau]
[ 4999.978065]  __dev_open+0xd4/0x1a0
[ 4999.978069]  __dev_change_flags+0x1cb/0x240
[ 4999.978072]  dev_change_flags+0x21/0x60
[ 4999.978075]  devinet_ioctl+0x701/0x8f0
[ 4999.978078]  inet_ioctl+0x18c/0x1d0
[ 4999.978084]  sock_do_ioctl+0x3c/0x130
[ 4999.978088]  ? selinux_file_ioctl+0x132/0x1e0
[ 4999.978091]  sock_ioctl+0x260/0x3e0
[ 4999.978096]  __x64_sys_ioctl+0x83/0xb0
[ 4999.978099]  do_syscall_64+0x33/0x40
[ 4999.978102]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 4999.978105] RIP: 0033:0x7f30af1285db
[ 4999.978107] Code: 89 d8 49 8d 3c 1c 48 f7 d8 49 39 c4 72 b5 e8 1c ff ff ff 85 c0 78 ba 4c 89 e0 5b 5d 41 5c c3 f3 0f 1e fa b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 6d b8 0c 00 f7 d8 64 89 01 48
[ 4999.978109] RSP: 002b:00007ffdc576b258 EFLAGS: 00000202 ORIG_RAX: 0000000000000010
[ 4999.978112] RAX: ffffffffffffffda RBX: 0000000000000041 RCX: 00007f30af1285db
[ 4999.978114] RDX: 00007ffdc576b260 RSI: 0000000000008914 RDI: 0000000000000004
[ 4999.978115] RBP: 00007ffdc576b320 R08: 0000000000000002 R09: 0033753275336630
[ 4999.978117] R10: 00007f30af1a6ac0 R11: 0000000000000202 R12: 00007ffdc576b260
[ 4999.978118] R13: 0000000000000004 R14: 00007ffdc576b320 R15: 0000000000000000
[ 5001.396862] IPv6: ADDRCONF(NETDEV_CHANGE): wlp7s0f3u2u3: link becomes ready
[ 5001.510298] 8021q: 802.1Q VLAN Support v1.8
```

ohhai, 2021-03-26:
Still present on 5.11.8-200.fc33.x86_64+debug and latest v5.6.4.2 (b65dcf4  "Merge pull request #823 from astsam2/v5.6.4.2")
Should I create separate issue for it?

```
[  334.705194] kworker/dying (149) used greatest stack depth: 11712 bytes left
[ 1315.541576] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[ 1315.541759] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[ 1315.544895] 88XXau: loading out-of-tree module taints kernel.
[ 1315.545946] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
[ 1316.380296] usb 3-2.1: 88XXau 08:10:7b:6d:5b:4e hw_info[d7]
[ 1316.390141] usbcore: registered new interface driver rtl88XXau
[ 1316.404272] rtl88XXau 3-2.1:1.0 wlp7s0f3u2u1: renamed from wlan0

[ 1317.426705] ============================================
[ 1317.426706] WARNING: possible recursive locking detected
[ 1317.426707] 5.11.8-200.fc33.x86_64+debug #1 Tainted: G           OE    
[ 1317.426709] --------------------------------------------
[ 1317.426710] ifconfig/7357 is trying to acquire lock:
[ 1317.426711] ffff9874475dd670 (pmutex){+.+.}-{3:3}, at: usbctrl_vendorreq+0x9a/0x290 [88XXau]
[ 1317.426751] 
               but task is already holding lock:
[ 1317.426752] ffff9874475dc110 (pmutex){+.+.}-{3:3}, at: netdev_open+0x31/0x60 [88XXau]
[ 1317.426784] 
               other info that might help us debug this:
[ 1317.426785]  Possible unsafe locking scenario:

[ 1317.426785]        CPU0
[ 1317.426786]        ----
[ 1317.426786]   lock(pmutex);
[ 1317.426788]   lock(pmutex);
[ 1317.426789] 
                *** DEADLOCK ***

[ 1317.426790]  May be due to missing lock nesting notation

[ 1317.426790] 2 locks held by ifconfig/7357:
[ 1317.426792]  #0: ffffffff9dfe2d30 (rtnl_mutex){+.+.}-{3:3}, at: devinet_ioctl+0xa6/0x8f0
[ 1317.426797]  #1: ffff9874475dc110 (pmutex){+.+.}-{3:3}, at: netdev_open+0x31/0x60 [88XXau]
[ 1317.426823] 
               stack backtrace:
[ 1317.426825] CPU: 0 PID: 7357 Comm: ifconfig Tainted: G           OE     5.11.8-200.fc33.x86_64+debug #1
[ 1317.426827] Hardware name: ASUS System Product Name/TUF GAMING B550M-PLUS, BIOS 1401 12/03/2020
[ 1317.426828] Call Trace:
[ 1317.426830]  dump_stack+0x8b/0xb0
[ 1317.426834]  __lock_acquire.cold+0x159/0x2ab
[ 1317.426839]  lock_acquire+0xc8/0x380
[ 1317.426842]  ? usbctrl_vendorreq+0x9a/0x290 [88XXau]
[ 1317.426871]  ? save_trace+0x3f/0x2f0
[ 1317.426874]  ? usbctrl_vendorreq+0x9a/0x290 [88XXau]
[ 1317.426894]  __mutex_lock+0x7b/0x7e0
[ 1317.426896]  ? usbctrl_vendorreq+0x9a/0x290 [88XXau]
[ 1317.426916]  usbctrl_vendorreq+0x9a/0x290 [88XXau]
[ 1317.426936]  usb_read8+0x2a/0x40 [88XXau]
[ 1317.426967]  rtl8812au_hal_init+0x54/0x1070 [88XXau]
[ 1317.426995]  rtw_hal_init+0x1c/0xd0 [88XXau]
[ 1317.427024]  _netdev_open+0x51/0x230 [88XXau]
[ 1317.427050]  netdev_open+0x39/0x60 [88XXau]
[ 1317.427073]  __dev_open+0xd4/0x1a0
[ 1317.427076]  __dev_change_flags+0x1cb/0x240
[ 1317.427078]  dev_change_flags+0x21/0x60
[ 1317.427080]  devinet_ioctl+0x701/0x8f0
[ 1317.427083]  inet_ioctl+0x18c/0x1d0
[ 1317.427086]  sock_do_ioctl+0x3c/0x130
[ 1317.427088]  ? selinux_file_ioctl+0x132/0x1e0
[ 1317.427091]  sock_ioctl+0x25e/0x360
[ 1317.427094]  __x64_sys_ioctl+0x83/0xb0
[ 1317.427097]  do_syscall_64+0x33/0x40
[ 1317.427099]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 1317.427101] RIP: 0033:0x7fdc6db5b5db
[ 1317.427103] Code: 89 d8 49 8d 3c 1c 48 f7 d8 49 39 c4 72 b5 e8 1c ff ff ff 85 c0 78 ba 4c 89 e0 5b 5d 41 5c c3 f3 0f 1e fa b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 6d b8 0c 00 f7 d8 64 89 01 48
[ 1317.427105] RSP: 002b:00007fff18ee3738 EFLAGS: 00000202 ORIG_RAX: 0000000000000010
[ 1317.427107] RAX: ffffffffffffffda RBX: 0000000000000041 RCX: 00007fdc6db5b5db
[ 1317.427108] RDX: 00007fff18ee3740 RSI: 0000000000008914 RDI: 0000000000000004
[ 1317.427110] RBP: 00007fff18ee3800 R08: 0000000000000002 R09: 0031753275336630
[ 1317.427111] R10: 00007fdc6dbd9ac0 R11: 0000000000000202 R12: 00007fff18ee3740
[ 1317.427112] R13: 0000000000000004 R14: 00007fff18ee3800 R15: 0000000000000000
[ 1318.874822] IPv6: ADDRCONF(NETDEV_CHANGE): wlp7s0f3u2u1: link becomes ready
[ 1318.951391] 8021q: 802.1Q VLAN Support v1.8
[ 1448.821303] kworker/dying (35) used greatest stack depth: 11312 bytes left
```

ohhai, 2021-04-11:
Also exists with v5.7.0 + kernel 5.11.11

```
[ 3405.667652] 88XXau: loading out-of-tree module taints kernel.
[ 3405.669055] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
[ 3406.535634] usbcore: registered new interface driver rtl88xxau
[ 3406.565507] rtl88xxau 3-2.1:1.0 wlp7s0f3u2u1: renamed from wlan0

[ 3407.594493] ============================================
[ 3407.594495] WARNING: possible recursive locking detected
[ 3407.594498] 5.11.11-200.fc33.x86_64+debug #1 Tainted: G           OE    
[ 3407.594501] --------------------------------------------
[ 3407.594503] ifconfig/5133 is trying to acquire lock:
[ 3407.594506] ffff8a5f84d8d998 (pmutex){+.+.}-{3:3}, at: usbctrl_vendorreq+0x98/0x295 [88XXau]
[ 3407.594597] 
               but task is already holding lock:
[ 3407.594600] ffff8a5f84d8c110 (pmutex){+.+.}-{3:3}, at: netdev_open+0x37/0x59 [88XXau]
[ 3407.594672] 
               other info that might help us debug this:
[ 3407.594674]  Possible unsafe locking scenario:

[ 3407.594676]        CPU0
[ 3407.594677]        ----
[ 3407.594679]   lock(pmutex);
[ 3407.594682]   lock(pmutex);
[ 3407.594685] 
                *** DEADLOCK ***

[ 3407.594687]  May be due to missing lock nesting notation

[ 3407.594689] 2 locks held by ifconfig/5133:
[ 3407.594692]  #0: ffffffff86fe2e50 (rtnl_mutex){+.+.}-{3:3}, at: devinet_ioctl+0xa6/0x8f0
[ 3407.594703]  #1: ffff8a5f84d8c110 (pmutex){+.+.}-{3:3}, at: netdev_open+0x37/0x59 [88XXau]
[ 3407.594772] 
               stack backtrace:
[ 3407.594774] CPU: 5 PID: 5133 Comm: ifconfig Tainted: G           OE     5.11.11-200.fc33.x86_64+debug #1
[ 3407.594778] Hardware name: ASUS System Product Name/TUF GAMING B550M-PLUS, BIOS 1401 12/03/2020
[ 3407.594781] Call Trace:
[ 3407.594785]  dump_stack+0x8b/0xb0
[ 3407.594792]  __lock_acquire.cold+0x159/0x2ab
[ 3407.594800]  lock_acquire+0xc8/0x380
[ 3407.594807]  ? usbctrl_vendorreq+0x98/0x295 [88XXau]
[ 3407.594873]  ? save_trace+0x3f/0x2f0
[ 3407.594879]  ? usbctrl_vendorreq+0x98/0x295 [88XXau]
[ 3407.594943]  __mutex_lock+0x7b/0x7e0
[ 3407.594948]  ? usbctrl_vendorreq+0x98/0x295 [88XXau]
[ 3407.595014]  usbctrl_vendorreq+0x98/0x295 [88XXau]
[ 3407.595060]  usb_read8+0x2d/0x37 [88XXau]
[ 3407.595127]  _rtw_read8+0x1b/0x1c [88XXau]
[ 3407.595164]  rtl8812au_hal_init+0x54/0x108d [88XXau]
[ 3407.595223]  rtw_hal_init+0x35/0xd7 [88XXau]
[ 3407.595284]  _netdev_open+0x54/0x1bd [88XXau]
[ 3407.595342]  netdev_open+0x3f/0x59 [88XXau]
[ 3407.595387]  __dev_open+0xd4/0x1a0
[ 3407.595392]  __dev_change_flags+0x1cb/0x240
[ 3407.595397]  dev_change_flags+0x21/0x60
[ 3407.595401]  devinet_ioctl+0x701/0x8f0
[ 3407.595406]  inet_ioctl+0x18c/0x1d0
[ 3407.595411]  sock_do_ioctl+0x3c/0x130
[ 3407.595416]  ? selinux_file_ioctl+0x132/0x1e0
[ 3407.595421]  sock_ioctl+0x25e/0x360
[ 3407.595426]  __x64_sys_ioctl+0x83/0xb0
[ 3407.595431]  do_syscall_64+0x33/0x40
[ 3407.595435]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 3407.595439] RIP: 0033:0x7f9df08455db
[ 3407.595442] Code: 89 d8 49 8d 3c 1c 48 f7 d8 49 39 c4 72 b5 e8 1c ff ff ff 85 c0 78 ba 4c 89 e0 5b 5d 41 5c c3 f3 0f 1e fa b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 6d b8 0c 00 f7 d8 64 89 01 48
[ 3407.595446] RSP: 002b:00007ffc080c6468 EFLAGS: 00000202 ORIG_RAX: 0000000000000010
[ 3407.595450] RAX: ffffffffffffffda RBX: 0000000000000041 RCX: 00007f9df08455db
[ 3407.595452] RDX: 00007ffc080c6470 RSI: 0000000000008914 RDI: 0000000000000004
[ 3407.595454] RBP: 00007ffc080c6530 R08: 0000000000000002 R09: 0031753275336630
[ 3407.595456] R10: 00007f9df08c3ac0 R11: 0000000000000202 R12: 00007ffc080c6470
[ 3407.595458] R13: 0000000000000004 R14: 00007ffc080c6530 R15: 0000000000000000
[ 3409.059450] IPv6: ADDRCONF(NETDEV_CHANGE): wlp7s0f3u2u1: link becomes ready
[ 3409.181107] 8021q: 802.1Q VLAN Support v1.8
```

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)

ohhai, 2021-05-11:
Still exists on 5.11.18 and this commit:
  `1c9d034 (HEAD -> v5.7.0, origin/v5.7.0) Updated MESH code (picked from the future) * Merged an updated MESH code from Realtek's G6 RTK driver   (used to build/test the RTW88 drivers)`

yjdwbj, 2023-01-21:
I also have the same problem.

```sh
[ 9882.058910] 88XXau: loading out-of-tree module taints kernel.
[ 9882.473394] usb 1-1.4: 88XXau e8:4e:06:5b:fa:98 hw_info[107]
[ 9882.519240] usbcore: registered new interface driver rtl88XXau

[ 9882.907153] ============================================
[ 9882.912472] WARNING: possible recursive locking detected
[ 9882.917787] 5.15.88-lcy-230115 #3 Tainted: G           O     
[ 9882.923536] --------------------------------------------
[ 9882.928850] dhcpcd/568 is trying to acquire lock:
[ 9882.933560] c60fd08c (pmutex){+.+.}-{3:3}, at: usbctrl_vendorreq+0x78/0x254 [88XXau]
[ 9882.941825] 
               but task is already holding lock:
[ 9882.947664] c60fc0d8 (pmutex){+.+.}-{3:3}, at: netdev_open+0x2c/0x4c [88XXau]
[ 9882.955235] 
               other info that might help us debug this:
[ 9882.961763]  Possible unsafe locking scenario:

[ 9882.967684]        CPU0
[ 9882.970132]        ----
[ 9882.972580]   lock(pmutex);
[ 9882.975382]   lock(pmutex);
[ 9882.978184] 
                *** DEADLOCK ***

[ 9882.984104]  May be due to missing lock nesting notation

[ 9882.990894] 2 locks held by dhcpcd/568:
[ 9882.994733]  #0: c18c6120 (rtnl_mutex){+.+.}-{3:3}, at: devinet_ioctl+0xc8/0x8f8
[ 9883.002167]  #1: c60fc0d8 (pmutex){+.+.}-{3:3}, at: netdev_open+0x2c/0x4c [88XXau]
[ 9883.010168] 
               stack backtrace:
[ 9883.014530] CPU: 0 PID: 568 Comm: dhcpcd Tainted: G           O      5.15.88-lcy-230115 #3
[ 9883.022802] Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[ 9883.029337] [<c0112b40>] (unwind_backtrace) from [<c010cb8c>] (show_stack+0x10/0x14)
[ 9883.037107] [<c010cb8c>] (show_stack) from [<c0f8cd28>] (dump_stack_lvl+0x58/0x70)
[ 9883.044692] [<c0f8cd28>] (dump_stack_lvl) from [<c018f444>] (__lock_acquire+0x14c8/0x2e00)
[ 9883.052977] [<c018f444>] (__lock_acquire) from [<c01917fc>] (lock_acquire+0x130/0x40c)
[ 9883.060908] [<c01917fc>] (lock_acquire) from [<c0f9fc04>] (__mutex_lock+0x90/0xa34)
[ 9883.068584] [<c0f9fc04>] (__mutex_lock) from [<c0fa05c4>] (mutex_lock_nested+0x1c/0x24)
[ 9883.076603] [<c0fa05c4>] (mutex_lock_nested) from [<bf0fc2b8>] (usbctrl_vendorreq+0x78/0x254 [88XXau])
[ 9883.086344] [<bf0fc2b8>] (usbctrl_vendorreq [88XXau]) from [<bf12ed58>] (usb_read8+0x40/0x6c [88XXau])
[ 9883.096451] [<bf12ed58>] (usb_read8 [88XXau]) from [<bf142b84>] (rtl8812au_hal_init+0x6c/0x106c [88XXau])
[ 9883.106821] [<bf142b84>] (rtl8812au_hal_init [88XXau]) from [<bf1117ac>] (rtw_hal_init+0x18/0xdc [88XXau])
[ 9883.117270] [<bf1117ac>] (rtw_hal_init [88XXau]) from [<bf0faa3c>] (_netdev_open+0x4c/0x200 [88XXau])
[ 9883.127290] [<bf0faa3c>] (_netdev_open [88XXau]) from [<bf0fac24>] (netdev_open+0x34/0x4c [88XXau])
[ 9883.137134] [<bf0fac24>] (netdev_open [88XXau]) from [<c0c26720>] (__dev_open+0xe4/0x1a4)
[ 9883.145727] [<c0c26720>] (__dev_open) from [<c0c26bac>] (__dev_change_flags+0x18c/0x224)
[ 9883.153832] [<c0c26bac>] (__dev_change_flags) from [<c0c26c58>] (dev_change_flags+0x14/0x44)
[ 9883.162281] [<c0c26c58>] (dev_change_flags) from [<c0d21b38>] (devinet_ioctl+0x6c4/0x8f8)
[ 9883.170476] [<c0d21b38>] (devinet_ioctl) from [<c0d23f54>] (inet_ioctl+0x1c8/0x2fc)
[ 9883.178146] [<c0d23f54>] (inet_ioctl) from [<c0bf4d10>] (sock_ioctl+0x4e0/0x618)
[ 9883.185559] [<c0bf4d10>] (sock_ioctl) from [<c02f0ca0>] (sys_ioctl+0x5a8/0xec4)
[ 9883.192888] [<c02f0ca0>] (sys_ioctl) from [<c0100080>] (ret_fast_syscall+0x0/0x1c)
[ 9883.200473] Exception stack(0xc33fbfa8 to 0xc33fbff0)
[ 9883.205533] bfa0:                   bee257cc 01c60150 00000009 00008914 bee257cc bee25a60
[ 9883.213719] bfc0: bee257cc 01c60150 b6fd387c 00000036 bee25a60 01c60150 bee2581c 01c6015c
[ 9883.221899] bfe0: 004cce58 bee257c4 00497a3f b6f42f98
[ 9883.865436] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

```

kimocoder, 2023-05-01:
the issue with "iwd" crash should now be fixed. reopen if not.

closed 2023-05-01 by kimocoder