aircrack-ng/rtl8812au #499: No network found using v5.6.4.2/v5.6.4.1

yshui, 2019-11-30:
A scan shows no results. If I downgrade to v5.3.4, I can see the APs and I can connect to them just fine.

In dmesg, I see lines like:

`RTW: _alloc_mgtxmitframe(wlan0) alloc xmitbuf fail`

```
[ 3645.379145] RTW: ===>PHY_GetTxPowerIndexBase: INVALID Rate(0x02).
[ 3645.379232] RTW: ===>PHY_GetTxPowerIndexBase: INVALID Rate(0x04).
[ 3645.379313] RTW: ===>PHY_GetTxPowerIndexBase: INVALID Rate(0x0b).
```
```
[ 3855.377317] RTW: rtl8812_sreset_linked_status_check REG_RXDMA_STATUS:0xeaeaeaea
[ 3857.510363] RTW: rtl8812_sreset_xmit_status_check REG_TXDMA_STATUS:0xeaeaeaea
[ 3857.510365] RTW: rtl8812_sreset_xmit_status_check tx hang
```

Not sure if they are related.

kimocoder, 2019-11-30:
OS, kernel, which chipset? (since er got 3)

yshui, 2019-11-30:
Linux 5.3.13,  T2U Nano

NullbutC00L, 2019-12-01:
Same problem here zero APs found

epinter, 2019-12-02:
See https://github.com/aircrack-ng/rtl8812au/issues/422 and https://github.com/aircrack-ng/rtl8812au/pull/504

yshui, 2019-12-03:
@epinter I don't think #504 is related, my VID:PID is `2357:011e`

yshui, 2020-02-15:
Looks like this problem has disappeared from the latest v5.6.4.2 branch. Closing.

closed 2020-02-15 by yshui