aircrack-ng/rtl8812au #503: ASUS USB-AC68: low speed RX/TX

AdrianBan, 2019-12-01:
Hi all,

I've installed the latest git driver and first tests I could get about 150Mbps RX and 200Mbps TX. But now I can't go more than 32Mbps and 75-80Mbps in the same conditions.
Initially I said that is connected on USB 2.0, but is still USB 3.0:

```
# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 8, If 0, Class=Vendor Specific Class, Driver=88XXau, 5000M
```

dmesg:
```
[44360.941664] usbcore: deregistering interface driver 88XXau
[44361.241058] usb 2-2: reset high-speed USB device number 14 using xhci_hcd
[44476.334970] usb 2-2: 88XXau 04:d9:f5:16:16:d2 hw_info[d8]
[44477.092704] usb 2-2: USB disconnect, device number 14
[44477.092719] usbcore: registered new interface driver 88XXau
[44477.220586] usb 4-2: new SuperSpeed Gen 1 USB device number 8 using xhci_hcd
[44477.240991] usb 4-2: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[44477.241151] usb 4-2: New USB device found, idVendor=0b05, idProduct=1853, bcdDevice= 0.00
[44477.241154] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[44477.241157] usb 4-2: Product: 802.11ac NIC
[44477.241159] usb 4-2: Manufacturer: Realtek
[44477.241161] usb 4-2: SerialNumber: 123456
[44477.364160] usb 4-2: 88XXau 04:d9:f5:16:16:d2 hw_info[d8]
[44477.378737] 88XXau 4-2:1.0 wlx04d9f51616d2: renamed from wlan0
```
OS: Debian Buster tested with Kernel 4.19.0-6-amd64 and 5.2.0-0.bpo.3-amd64.
When I got the high speeds were with kernel 4.19.0-6-amd64, then I've booted in 5.2.0-0.bpo.3-amd64 and the speed dropped to 32Mbps. After neither on 4.19.0-6-amd64 I can reach more than 32Mbps.
What could be the issue?

AdrianBan, 2020-01-25:
After upgrading the kernel to 5.4.0 and latest driver, seems that the speed is going over 100Mbps. Very strange behavior with 5.2.0 and 4.19.0. Before starting the new kernel, I was making another test with kernel 5.2.0 and the speed was no more than 50Mbps.

[image]

kimocoder, 2020-01-25:
You said that it's connected to USB 2.0, but is still USB 3.0? I can't really understand that line. And the driver is forced down from SS by using the usb switch in usb_intfs.c for a reason.. and information is lacking in this report whatsoever, there's no point showing me a speedtest, for all I know your location is on the moon and you pinging Jupiter for all I know..

"modinfo 88XXau", "dmesg", "iw phy info", 
what chipset.. etc etc etc

AdrianBan, 2020-01-26:
I will try to avoid your irony. I did tests directly between my router, which is also a Linux x86 PC with Intel NICs.
Maybe I was expressed my self wrongly : first time when I've connected the device, the driver was set to USB 2.0 mode. I had to set rtw_switch_usb_mode to 1 to force the device to work in USB 3.0. When the device worked only in USB 2.0,the speed results were similar like I've mentioned in my post. 
Today I've installed the kernel 5.4.0, updated the driver using git pull and then I've recompiled the driver. Before rebooting in 5.4.0 I did few tests to see if is still behave with low speeds and the problem was the same.
Rebooting the new kernel (and NOT changing any thin in module.d/88xxau.conf file!) the speed was going up.
The chipset is Realtek RTL8814AU. Tomorrow I will give you the rest of the information. I will reboot in kernel 5.2.0 and I will check if the issue is still there on 5.2.0 and 4.19.

kimocoder, 2020-01-26:
That's better, yeah sorry for that btw, a bit cranky, been quite lots to do.
I can explain you something about the these drivers, that people don't keep in mind..

These drivers is a "skeleton" originally, from Realtek.. a so called multichip driver 'base'.
So.. we wanted monitor mode & frame injection for use with Kali Linux (that's was the main goal at least), so into the rtl8812au base, we've added the rtl8821au and rtl8814au into it.. but when rtl8812au gets a new release from Realtek, the other chipsets HAL and PHYDM still will be hanging behind.. and that's the situation for rtl8814au. The HAL/PHYDM was ripped out from v4.3.20(21) .. it's old, dirty.. have no words. Therefor support for the rtl8814au will not be included in the future, it takes time to actually change all the functions/params to make it fit. And I won't start cleaning that stuff, not Realtek either probably. So if your on the 8814au.. you should any other adapter really.

But I'll take your request seriously, and I have all supported chipsets, so I'll setup a test environment and will go through 'em all anyway, so thanks for the heads up/giving me the idea :1st_place_medal:

kimocoder, 2020-01-26:
Take a look below here.. 8814au code to the left, 8812au to the right.. same file, the HAL init..
notice the copyright years too. there's lots of fixes it missed, not just one or two unfortunately :-1: 

[image]

kimocoder, 2020-03-08:
It is what it is for now, we're updating the base day by day again now.. kernel v5.4 has a chandef regression caused by kernel. Just a heads up. Stay tuned.

closed 2020-03-08 by kimocoder