aircrack-ng/rtl8812au #506: Error installing dkms in Raspberry Pi 4

acdcjavier, 2019-12-06:
Hello
I've followed all the steps for installing the latest drive into my Raspberry Pi 4, but I got thos error when I performed sudo ./dkms-install.sh

```
pi@raspberrypi:~/rtl8812au $ sudo ./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.6.4.2/source ->
                 /usr/src/rtl8812au-5.6.4.2

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.19.86-v7l+ cannot be found at
/lib/modules/4.19.86-v7l+/build or /lib/modules/4.19.86-v7l+/source.
Error! echo
Your kernel headers for kernel 4.19.86-v7l+ cannot be found at
/lib/modules/4.19.86-v7l+/build or /lib/modules/4.19.86-v7l+/source.
Finished running dkms install steps.
```

Using make installation got this error:

`pi@raspberrypi:~/rtl8812au $ sudo make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.86-v7l+/build M=/home/pi/rtl8812au  modules
make[1]: *** /lib/modules/4.19.86-v7l+/build: No existe el fichero o el directorio.  Alto.
make: *** [Makefile:2245: modules] Error 2
`

Any solution? thanks

ZerBea, 2019-12-06:
Kernel headers are missing and you have to install them.
Assuming you're running Raspian, read more here
https://www.raspberrypi.org/documentation/linux/kernel/headers.md
I suggest to install Arch:
https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-4
and you will learn much about Linux

acdcjavier, 2019-12-06:
> Kernel headers are missing and you have to install them.
> Assuming you're running Raspian, read more here
> https://www.raspberrypi.org/documentation/linux/kernel/headers.md
> I suggest to install Arch:
> https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-4
> and you will learn much about Linux

Thanks but headers already are installed

ZerBea, 2019-12-06:
Do they match to the installed kernel?
$ uname -r

acdcjavier, 2019-12-06:
> Do they match to the installed kernel?
> $ uname -r
> vs
> $ ls -la /lib/modules
> 
> or
> $ dpkg -s raspberrypi-kernel | grep Version
> vs
> $ dpkg -s raspberrypi-kernel-headers | grep Version

The first one got it:

`root@javier:/home/pi# uname -r
4.19.86-v7l+
root@javier:/home/pi# ls -la /lib/modules
total 40
drwxr-xr-x 10 root root 4096 dic  5 20:52 .
drwxr-xr-x 18 root root 4096 dic  5 20:50 ..
drwxr-xr-x  3 root root 4096 dic  6 11:45 4.19.75+
drwxr-xr-x  3 root root 4096 dic  6 11:45 4.19.75-v7+
drwxr-xr-x  3 root root 4096 dic  6 11:45 4.19.75-v7l+
drwxr-xr-x  3 root root 4096 dic  5 17:56 4.19.75-v8+
drwxr-xr-x  3 root root 4096 dic  5 20:52 4.19.86+
drwxr-xr-x  3 root root 4096 dic  5 20:53 4.19.86-v7+
drwxr-xr-x  3 root root 4096 dic  5 20:52 4.19.86-v7l+
drwxr-xr-x  3 root root 4096 dic  5 20:52 4.19.86-v8+
root@javier:/home/pi# 

`

The second:

`root@javier:/home/pi# dpkg -s raspberrypi-kernel | grep Version
Version: 1.20190925+1-1
root@javier:/home/pi# dpkg -s raspberrypi-kernel-headers | grep Version
Version: 1.20190925+1-1
root@javier:/home/pi# 
`

ZerBea, 2019-12-06:
That's looking good. Now we have to check for the sub directories (build and/or source).
Does one of them exists?
$ ls -la /lib/modules/4.19.86-v7l+/

acdcjavier, 2019-12-08:
> ls -la /lib/modules/4.19.86-v7l+/

Hello

Yes, it  exists

`pi@javier:~ $  ls -la /lib/modules/4.19.86-v7l+/

total 2140
drwxr-xr-x  3 root root   4096 dic  5 20:52 .
drwxr-xr-x 10 root root   4096 dic  5 20:52 ..
drwxr-xr-x 11 root root   4096 dic  5 20:52 kernel
-rw-r--r--  1 root root 543018 dic  5 20:52 modules.alias
-rw-r--r--  1 root root 561702 dic  5 20:52 modules.alias.bin
-rw-r--r--  1 root root  12111 dic  5 20:52 modules.builtin
-rw-r--r--  1 root root  13379 dic  5 20:52 modules.builtin.bin
-rw-r--r--  1 root root 185405 dic  5 20:52 modules.dep
-rw-r--r--  1 root root 256726 dic  5 20:52 modules.dep.bin
-rw-r--r--  1 root root    302 dic  5 20:52 modules.devname
-rw-r--r--  1 root root  62689 dic  5 20:52 modules.order
-rw-r--r--  1 root root    379 dic  5 20:52 modules.softdep
-rw-r--r--  1 root root 233253 dic  5 20:52 modules.symbols
-rw-r--r--  1 root root 286483 dic  5 20:52 modules.symbols.bin

pi@javier:~ $ 
`

ZerBea, 2019-12-08:
Unfortunately neither "build" nor "source" exists in your tree. So dkms isn't able to build the driver.
For example my tree:
drwxr-xr-x 21 zerobeat users    500  8. Dez 19:05 build
              1 zerobeat users      0  8. Dez 19:05 extramodules
drwxr-xr-x  9 zerobeat users    180  2. Dez 18:49 kernel
-rw-r--r--  1 zerobeat users 638541  2. Dez 18:52 modules.alias
-rw-r--r--  1 zerobeat users 661826  2. Dez 18:52 modules.alias.bin
-rw-r--r--  1 zerobeat users   9376 30. Nov 19:55 modules.builtin
-rw-r--r--  1 zerobeat users  11541  2. Dez 18:52 modules.builtin.bin
-rw-r--r--  1 zerobeat users 276137  2. Dez 18:52 modules.dep
-rw-r--r--  1 zerobeat users 372137  2. Dez 18:52 modules.dep.bin
-rw-r--r--  1 zerobeat users    226  2. Dez 18:52 modules.devname
-rw-r--r--  1 zerobeat users  89309 30. Nov 19:55 modules.order
-rw-r--r--  1 zerobeat users    379  2. Dez 18:52 modules.softdep
-rw-r--r--  1 zerobeat users 247964  2. Dez 18:52 modules.symbols
-rw-r--r--  1 zerobeat users 303841  2. Dez 18:52 modules.symbols.bin

As you can see, the headers are present here:
drwxr-xr-x 21 zerobeat users    500  8. Dez 19:05 build

ZerBea, 2019-12-08:
looks that they are dead:
https://github.com/raspberrypi/linux/issues/3279

ZerBea, 2019-12-08:
Take a look here:
https://archlinuxarm.org/packages/armv7h/linux-raspberrypi4-headers
Download the header package 
http://mirror.archlinuxarm.org/armv7h/core/linux-raspberrypi4-headers-4.19.86-1-armv7h.pkg.tar.xz
Unpack and take a look inside the build tree you can imagine what files you're missing:
rwxr-xr-x   3 zerobeat users   4096 30. Nov 19:53 arch
drwxr-xr-x   3 zerobeat users   4096 30. Nov 19:53 block
drwxr-xr-x   2 zerobeat users   4096 30. Nov 19:53 certs
-rw-r--r--   1 zerobeat users 178453 30. Nov 19:53 .config
drwxr-xr-x   4 zerobeat users   4096 30. Nov 19:53 crypto
drwxr-xr-x 131 zerobeat users   4096 30. Nov 19:53 drivers
drwxr-xr-x  63 zerobeat users   4096 30. Nov 19:53 fs
drwxr-xr-x  29 zerobeat users   4096 30. Nov 19:53 include
drwxr-xr-x   2 zerobeat users   4096 30. Nov 19:53 init
-rw-r--r--   1 zerobeat users    563 30. Nov 19:53 Kconfig
drwxr-xr-x  10 zerobeat users   4096 30. Nov 19:53 kernel
drwxr-xr-x   4 zerobeat users   4096 30. Nov 19:53 lib
-rw-r--r--   1 zerobeat users  60216 30. Nov 19:53 Makefile
drwxr-xr-x   2 zerobeat users   4096 30. Nov 19:53 mm
-rw-r--r--   1 zerobeat users 933550 30. Nov 19:53 Module.symvers
drwxr-xr-x  63 zerobeat users   4096 30. Nov 19:53 net
drwxr-xr-x   2 zerobeat users   4096 30. Nov 19:53 samples
drwxr-xr-x  14 zerobeat users   4096 30. Nov 19:53 scripts
drwxr-xr-x  10 zerobeat users   4096 30. Nov 19:53 security
drwxr-xr-x  25 zerobeat users   4096 30. Nov 19:53 sound
drwxr-xr-x   2 zerobeat users   4096 30. Nov 19:53 .tmp_versions
drwxr-xr-x   2 zerobeat users   4096 30. Nov 19:53 usr
drwxr-xr-x   4 zerobeat users   4096 30. Nov 19:53 virt

acdcjavier, 2019-12-09:
> looks that your installation is broken:
> [raspberrypi/linux#3279](https://github.com/raspberrypi/linux/issues/3279)
> [respeaker/seeed-voicecard#190](https://github.com/respeaker/seeed-voicecard/issues/190)

I've perfomed a fresh Rasbian Installation but still got error:

`pi@raspberrypi:~/rtl8812au $ sudo ./dkms-install.sh

About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.6.4.2/source ->
                 /usr/src/rtl8812au-5.6.4.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.19.75-v7l+ KSRC=/lib/modules/4.19.75-v7l+/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.75-v7l+ (armv7l)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.19.75-v7l+ KSRC=/lib/modules/4.19.75-v7l+/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.75-v7l+ (armv7l)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.
Finished running dkms install steps.
`

ZerBea, 2019-12-09:
You're on a wrong kernel version:  4.19.75-v7l+
Actual kernel is 4.19.86-v7l+

Please attach 
/var/lib/dkms/rtl8812au/5.6.4.2/build/make.log
to see what's going on there.

Or avoid complete dkms stuff.
Compile driver and add copy it to modules directory

ZerBea, 2019-12-09:
dkms is a never ending story. If you not want to avoid it:
https://raspberrypi.stackexchange.com/questions/93677/install-drivers-for-rtl8812au-for-raspibian-kernel-4-14-79-v7-or-4-14-79-87-v7
https://github.com/aircrack-ng/rtl8812au/issues/408

acdcjavier, 2019-12-09:
Finally i've installed the driver succesfull 
Thanks for your help

ZerBea, 2019-12-09:
Good to hear that. This chipset isn't bad and the driver has big potential.

acdcjavier, 2019-12-09:
> Good to hear that. This chipset isn't bad and the driver has big potential.

Yes, I already know this chipset, because I use it on my PC using Ubuntu 19, and recenlty i've bought a raspberry

ZerBea, 2019-12-09:
And the Raspberry is a nice system, too. I'm running several of them Pi 1: A+, B+, Zero wh and Zero.
https://github.com/ZerBea/hcxdumptool/wiki/Penetration-testing-system-2

turbofiesta, 2020-01-02:
I have a similar problem. I try tried many solutions but failed
Thanks for your help

System:
Raspberry Pi 4
Kernel 4.19.81-Re4son-v8l+
Description:    Kali GNU/Linux Rolling
Release:        2019.4
-------------
```
_sudo ./dkms-install.sh_
sudo: unable to resolve host bobas: Name or service not known
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.6.4.2/source ->
                 /usr/src/rtl8812au-5.6.4.2

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.19.81-Re4son-v8l+ cannot be found at
/lib/modules/4.19.81-Re4son-v8l+/build or /lib/modules/4.19.81-Re4son-v8l+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Error! echo
Your kernel headers for kernel 4.19.81-Re4son-v8l+ cannot be found at
/lib/modules/4.19.81-Re4son-v8l+/build or /lib/modules/4.19.81-Re4son-v8l+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Finished running dkms install steps.
```
-----

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)

mauricecyril, 2022-06-12:
I have the same issue.

phpony, 2023-02-16:
All you need is to run `apt-get dist-upgrade` and then `reboot` to get both latest kernel running and latest kernel headers installed. If you just run `apt install raspberrypi-kernel-headers ` - you'll get latest headers few versions ahead of running kernel.