aircrack-ng/rtl8812au #507: need help with older commit

logan001, 2019-12-07:
Can someone fix [this](https://github.com/aircrack-ng/rtl8812au/commit/2d0547c99fa029db403fe42192be9907e134b59c) commit to work with 5.4/5.5 kernels? i don't want anything major to change. this commit is still the best for my needs but it does not compile on 5+ kernels. i don't know what to change to make it work.

This is a random commit i pick a while back when i noticed that my wireless performance was downgraded after update. 

So please help me if know how to.

CGarces, 2019-12-07:
Hi.

Can you please detail the error?
The device from https://github.com/aircrack-ng/rtl8812au/commit/2d0547c99fa029db403fe42192be9907e134b59c already exist on the last branch.
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/os_dep/linux/usb_intf.c#L218

logan001, 2019-12-07:
did you try branch 5.3.4? 
i don't know why my link goes to 5.6.4.2. i link it from 5.3.4 commits. well whatever.
this is the log

> DKMS make.log for rtl8812au-5.3.4 for kernel 5.2.21-39-tkg-pds (x86_64)
> Sat 07 Dec 2019 11:24:24 PM EET
> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.2.21-39-tkg-pds/build M=/var/lib/dkms/rtl8812au/5.3.4/build  modules
> make[1]: Entering directory '/usr/lib/modules/5.2.21-39-tkg-pds/build'
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_cmd.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_security.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_debug.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_io.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ioctl_set.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ioctl_query.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ieee80211.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_mlme.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_mlme_ext.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_mi.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_wlan_util.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_vht.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_pwrctrl.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rf.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_recv.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_sta_mgt.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ap.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/mesh/rtw_mesh.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/mesh/rtw_mesh_pathtbl.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/mesh/rtw_mesh_hwmp.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_xmit.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_p2p.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rson.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_tdls.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_br_ext.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_iol.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_sreset.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_btcoex_wifionly.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_btcoex.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_beamforming.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_odm.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rm.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rm_fsm.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/efuse/rtw_efuse.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/osdep_service.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/os_intfs.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/usb_intf.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/usb_ops_linux.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_linux.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/xmit_linux.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/mlme_linux.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/recv_linux.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_cfg80211.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_cfgvendor.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/wifi_regd.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_android.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_rhashtable.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_mp.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_intf.o
> /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/os_intfs.c:1414:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
>  1414 |  .ndo_select_queue = rtw_select_queue,
>       |                      ^~~~~~~~~~~~~~~~
> /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/os_intfs.c:1414:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_com.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_com_phycfg.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_phy.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_dm.o
>   CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_dm_acs.o
> cc1: some warnings being treated as errors
> make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/os_intfs.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_get_channel’:
[... 8 lines cut ...]
> In file included from ./include/linux/uaccess.h:11,
>                  from ./include/linux/crypto.h:21,
>                  from ./include/crypto/hash.h:11,
[... cut ...]
> /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_android.c:660:7: note: each undeclared identifier is reported only once for each function it appears in
> make[2]: *** [scripts/Makefile.build:278: /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_android.o] Error 1
> make[1]: *** [Makefile:1601: _module_/var/lib/dkms/rtl8812au/5.3.4/build] Error 2
> make[1]: Leaving directory '/usr/lib/modules/5.2.21-39-tkg-pds/build'
> make: *** [Makefile:1650: modules] Error 2

right now i'm not on linux 5.4.due to some test. but the result will probably be similar

CGarces, 2019-12-08:
But why you are using an old and not maintained branch?
There is any reason to not use the last version ?

logan001, 2019-12-08:
yeap. i get better speed for AP than the latest branch/commits which refuse to even create ap in 5ghz. this commit is still the best among any newer versions i've tried (i didn't try every one of the commits). i'll keep trying the newer version from time to time but in the mean time i prefer to have a version that is actually working with good speeds.

CGarces, 2019-12-08:
I'm confused, I don't know the code that you are trying to test.
Please compile 5.3.4 from last commit (1887e5cbf772cd248a005eb687a5fd867fee41db), it must support 5.4 kernel as requested.

kimocoder, 2019-12-08:
I'm looking into the kernel v5.4 support.
Beginning with 5.4.0-rc3, a macro is defined for "fallthrough". For
that reason, the attribute statement must use __fallthrough__ instead

logan001, 2019-12-08:
> I'm confused, I don't know the code that you are trying to test.
> Please compile 5.3.4 from last commit ([1887e5c](https://github.com/aircrack-ng/rtl8812au/commit/1887e5cbf772cd248a005eb687a5fd867fee41db)), it must support 5.4 kernel as requested.

you misunderstood. the "code" has nothing to do with the rtl driver. i'm having some problems with a hard disk after i update to 5.4 and i was trying to figure out if the kernel was the issue or driver failure. anyway as i said before that commit that i pointed out is the one that i currently use on 4.19 and has great speeds for both upload/download in ap mode on rtl8814au. i would like to be able to use it on 5.x but it won't compile as i show you the logs. i'll try to use this commit you linked and see how it work in 1-2 hours when i can restart computer.

logan001, 2019-12-08:
[here](https://drive.google.com/open?id=1ViLxchHMWyuHpsdsyVoraIM23OPkgEXj) is the difference between the commit you pointed out and the one that i use

logan001, 2019-12-14:
so anybody that can make that 5.3.4 commit to work with linux 5.4 (and posible 5.5)?

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)