aircrack-ng/rtl8812au #509: 88XXau cause delay on system suspend

MR-Diamond, 2019-12-09:
Info of my System: Linux Manjaro XFCE x64, kernel 4.14.155-1-MANJARO, wifi USB adapter TP-Link Archer T9UH, driver is installed from AUR repo: `rtl88xxau-aircrack-dkms-git  r978.ebaed2f-1` (v5.6.4.1).
Presence of  `/etc/modprobe.d/88XXau.conf` with the content `options 88XXau rtw_switch_usb_mode=1`

iw dev wlan0 link:
```
Connected to xx:xx:xx:xx:xx:xx (on wlan0)
	SSID: MY_SSID
	freq: 5180
	signal: -39 dBm
	tx bitrate: 1300.0 MBit/s

	bss flags:	short-slot-time
	dtim period:	0
	beacon int:	100
```
iwconfig wlan0
```
wlan0     IEEE 802.11AC  ESSID:"MY_SSID"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.18 GHz  Access Point: xx:xx:xx:xx:xx:xx   
          Bit Rate:1.3 Gb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=92/100  Signal level=61/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

The device is on 5 Ghz, very stable, always fixed at 1300.0 MBit/s also under heavy network load.

Today when I put my laptop to sleep, it took about 7 seconds, strange, because is always immediate (1 second), I repeated this action various times and always took about 7 seconds for complete the suspending.

After various tests I discovered that before suspend I have to `modprobe -r 88XXau`: in this way the situation is back to normal: the laptop is able again to suspend in one second.

For now I'm using the following workaround:
I made two systemd services: the first one is triggered when i suspend the laptop: `modprobe -r 88XXau`, the second is triggered on resume: `modprobe 88XXau`

Maybe there is some other module option to insert in `/etc/modprobe.d/88XXau.conf`?

MR-Diamond, 2019-12-10:
I noticed another delay: when I reboot the laptop.
Usually, when I press the restart button in the XFCE Panel, the laptop reach the bios screen in 6 seconds.
Since I'm using TP-Link Archer T9UH with the 88XXau driver, instead, it reach the bios screen in 9 seconds.

If, before to do the reboot I do `rmmod 88xxau`, the laptop back to the previous behaviour (reach the bios screen in 6 seconds)
**If I don't** `rmmod 88XXau`, it back to reach the bios screen in 9 seconds.

Some info from the journal (i run `journalctl | grep 88XXau` after reboot, when I see the Desktop)

**Reboot without rmmod** (delayed):

```
dic 10 15:26:33 Probook kernel: usb 2-3: 88XXau XX:XX:XX:XX:XX:XX hw_info[d8]
dic 10 15:26:33 Probook kernel: usbcore: registered new interface driver 88XXau
dic 10 15:26:33 Probook NetworkManager[397]: <info>  [1575987993.3077] rfkill0: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/ieee80211/phy0/rfkill0) (driver 88XXau)
```

**Reboot with rmmod** (not delayed):

```
dic 10 15:32:01 Probook kernel: usb 4-3: 88XXau XX:XX:XX:XX:XX:XX hw_info[d8]
dic 10 15:32:01 Probook kernel: usbcore: registered new interface driver 88XXau
dic 10 15:32:01 Probook NetworkManager[376]: <info>  [1575988321.3493] rfkill0: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/ieee80211/phy0/rfkill0) (driver 88XXau)
dic 10 15:32:32 Probook sudo[2418]:     dave : TTY=unknown ; PWD=/home/dave ; USER=root ; COMMAND=/usr/bin/rmmod 88XXau
dic 10 15:32:32 Probook kernel: usbcore: deregistering interface driver 88XXau
```

Obviously I've done several times these tests: I can confirm that if I rmmod the 88XXau driver the reboot is quick; if I don't rmmod the 88XXau driver, there is a delay: and this occurs also on shutdown.

Doppel-D, 2019-12-25:
I probably had the same issue until I tried stable v5.3.4 which is working fine for me.

MR-Diamond, 2019-12-27:
> I probably had the same issue until I tried stable v5.3.4 which is working fine for me.

no dice: I downgraded to v5.3.4 and the issue (delay on suspend, reboot and shutdown) persists; furthermore, despite the fact that using v5.3.4 the device connects with the same stability, I have higher ping latency and oddly when I run `iw dev wlan0 link` it reports "not connected", despite the fact that `iwconfig wlan0` reports that all is fine.

Do you have the same device with the same revision? My TP-Link Archer T9UH is ver 2.0.

Anyway: when I installed this driver from the AUR repo, the maintainer reported that:

>" iyanmv commented on 2019-11-04 22:02	
I have updated the PKGBUILD just to push pkgver to last commit.  [...]  (v5.6.4.1)"
https://aur.archlinux.org/packages/rtl88xxau-aircrack-dkms-git/#pinned-714920

And currrently in such page the last update is reported to `Last Updated: 	2019-11-04 22:33`

But when I do `modinfo 88XXau` I got `version:        v5.6.4.2_35491.20191025`

Doppel-D, 2019-12-27:
My device is a Zyxel NWD6605 with a RTL8812AU. Are you sure that you were really running 5.3.4?

MR-Diamond, 2019-12-27:
> Are you sure that you were really running 5.3.4?

Yes. I checked with modinfo indicating v5.3.4

Doppel-D, 2019-12-27:
You can still be called lucky, because my delay was around 2 minutes (old CoreDuo 2x2.1Ghz) AND I always got a passphrase error before I downgraded ;- )

kimocoder, 2020-03-08:
I have removed some AUTOSUSPEND code, can someone test the latest git (of branch v5.6.4.2) and see if the suspend issue is resolved?

kimocoder, 2020-03-08:
Use the latest branch v5.6.4.2, the v5.6.4.1 haven't for many parches because Realtek pushed the update not long after v5.6.4.1.

closed 2020-03-08 by kimocoder

MR-Diamond, 2020-03-09:
Ok: I've rebuilt from https://github.com/aircrack-ng/rtl8812au.git#branch=v5.6.4.2
and the delay while suspending the system, has gone. But the delay while rebooting and shutting down, still persists.