aircrack-ng/rtl8812au #511: Set Tx Power (8B26)      Operation not permitted

w4cky, 2019-12-11:
I have problem with my ALFA AWUS1900:

```
Bus 001 Device 003: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
 root@kali  ~  iwconfig wlan0 txpower 30                                                                                      
### Error for wireless request "Set Tx Power" (8B26) :
    SET failed on device wlan0 ; Operation not permitted.

```

why?

kimocoder, 2019-12-11:
You didn't follow the README.md ..
Don't use "iwconfig" but "iw" !

And lots of information is missing here, which branch e.g?

w4cky, 2019-12-12:
Sorry @kimocoder .
I use branch v5.6.4.2 and kali linux 2019.4

```
iwconfig wlan1 | grep -i power
Power Managment: off 

```

when i use command
sudo iw wlan1 set txpower fixed 3000
command failed: Operation not permitted (-1) 

when i want change region 
sudo iw reg set US
command failed: Operation not permitted (-1)

closed 2019-12-12 by w4cky