aircrack-ng/rtl8812au #513: NetworkManager doesn't open

dsime42, 2019-12-13:
lsusb sees the dongal
lsmod shows that it was seen and modules loaded

but log shows 
Dec 12 11:53:07 flamebai kernel: usb 2-2.3: new high-speed USB device number 5 using ehci-pci
Dec 12 11:53:07 flamebai kernel: usb 2-2.3: New USB device found, idVendor=7392, idProduct=a833, bcdDevice= 0.00
Dec 12 11:53:07 flamebai kernel: usb 2-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 12 11:53:07 flamebai kernel: usb 2-2.3: Product: Edimax AC1750 USB
Dec 12 11:53:07 flamebai kernel: usb 2-2.3: Manufacturer: Realtek
Dec 12 11:53:07 flamebai kernel: usb 2-2.3: SerialNumber: 123456
Dec 12 11:53:07 flamebai mtp-probe[3394]: checking bus 2, device 5: "/sys/devices/pci0000:00/0000:00:13.2/usb2/2-2/2-2.3"
Dec 12 11:53:07 flamebai mtp-probe[3394]: bus: 2, device: 5 was not an MTP device
Dec 12 11:53:07 flamebai mtp-probe[3398]: checking bus 2, device 5: "/sys/devices/pci0000:00/0000:00:13.2/usb2/2-2/2-2.3"
Dec 12 11:53:07 flamebai mtp-probe[3398]: bus: 2, device: 5 was not an MTP device
Dec 12 11:53:07 flamebai baloo_file.desktop[2952]: UdevQt: unhandled device action "bind"

dsime42, 2019-12-13:
I thought that perhaps if I nNarrowed the driver to the chip it would be more successful
so I changed the Makefile to compile only for the 8814  and not the 8812 or the 8821
########################## WIFI IC ############################
CONFIG_RTL8812A = n
CONFIG_RTL8821A = n
CONFIG_RTL8814A = y

This resulted in these two error while compiling
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "recvbuf2recvframe" [/root/EW-7833UAC/driver/rtl8812au-5.6.4.2/8814au.ko] undefined!
ERROR: "_dbg_dump_tx_info" [/root/EW-7833UAC/driver/rtl8812au-5.6.4.2/8814au.ko] undefined!

and interestingly I found that in  ./hal/rtl8814a/usb/usb_ops_linux.c
The call definition for recvbuf2recvframe is commented out

#if 0
int recvbuf2recvframe(PADAPTER padapter, void *ptr)

kimocoder, 2020-01-20:
Don't do that, our drivers are linked together in one module, called 88XXau. However, I'll take a look, because AWUS1900 is testing ok.

kimocoder, 2020-01-20:
Tested the AWUS1900 in Ubuntu 19.10 x64 now, and NetworkManager works just fine. You need to check your setup I'd belive.

kimocoder, 2020-03-08:
NetworkManager and all is working as it should. Branch v5.6.4.2 is the latest, I've pushed some fixes to it, so update and format/check your setup/system.

[image]

closed 2020-03-08 by kimocoder