aircrack-ng/rtl8812au #514: `iw dev del` hangs on OpenWRT

Alabate, 2019-12-13:
Hi,

I'm trying to use this driver on an rtl8812au based usb board with OpenWrt 18.06.5 for x86_64. I've got no problem to build it after 28a8d26f11918b18871cb7ee6ba0cff329f1b7ee which brings openwrt build compatiblity. 

When I boot with the driver loaded and the board connected, the next time I try to do a simple `iw dev`, it will hang forever. However, if I connect the board afterboot, I can do an `iw dev` but generally after trying to configure it will hang at a moment.

I know that, In the OpenWRT wireless configuration script, it will delete and recreate the wifi interface with the right mode. So I tried to execute the `del` command when connecting the board after boot, and I can "produce" a hang with this single command: 

```
iw dev wlan0 del
```

Maybe this bug is not related to OpenWRT and can be produce everywhere, because not a lot of system tries to delete the interface. So it's maybe just a deadlock on the delete?

In latest commit ebaed2facba7bc96a3e664985b34011a896f2620, it will just hang without any message, but on an older commit 6ed82d886c18bcc7826fcbe8ca91f5fa645eb11a (with openwrt build fix 28a8d26f11918b18871cb7ee6ba0cff329f1b7ee cherry-picked), it will print this before hanging:

```
[   84.928053] RTW: cfg80211_rtw_del_virtual_intf(wlan0) unregister ndev
[   84.934571] RTW: rtw_wdev_unregister(wdev=ffff888071af6c00)
[   84.940617] RTW: cfg80211_rtw_get_txpower
[   84.983346] RTW: rtw_ndev_notifier_call(wlan0) state:6
[   84.988584] RTW: rtw_ndev_uninit(wlan0) if1
[   84.993948] RTW: rtw_wiphy_unregister(phy0)
[   84.998409] RTW: Vendor: Unregister RTW cfg80211 vendor interface
```

I tried this on following commits with the same result:
* 6ed82d886c18bcc7826fcbe8ca91f5fa645eb11a (2019-11-09) (with 28a8d26f11918b18871cb7ee6ba0cff329f1b7ee cherry picked)
* 73f0a8810da1f3dd84991cf3efda405cbab43f89 (2019-11-11) (with 28a8d26f11918b18871cb7ee6ba0cff329f1b7ee cherry picked)
* 2e294ae3695be71ea14a8513f23b763797902b91 (2019-11-26)
* ebaed2facba7bc96a3e664985b34011a896f2620 (2019-12-08)

If I can give you any information, or try a patch or something, tell me I will do it as soon as possible. Thank for your work!

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)

Alabate, 2021-05-11:
Well, i'm not working on this hardware anymore, so i cannot check if it is solved or not. I close it because this is not an issue for me anymore, but if it is for someone, feel free to re-open it.

WilliamKSokol, 2023-10-04:
I am having this issue still:
```
root@OpenWrt:~# iw dev wlan0 del
command failed: No error information (-524)
```
I am on raspberry pi 4b and I have this problem consistently