aircrack-ng/rtl8812au #516: Modprobe fails: Skipping invalid relocation target

itay-grudev, 2019-12-15:
The following is my `dmesg` log when the card is inserted. I tried both building it from source and DKMS - same result. I'm not sure what causes the error and my udnerstanfing of kernel drivers ends here.

```
[ 1556.965625] usb 1-2: new high-speed USB device number 11 using xhci_hcd
[ 1557.114261] usb 1-2: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[ 1557.114262] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1557.114263] usb 1-2: Product: 802.11n NIC
[ 1557.114264] usb 1-2: Manufacturer: Realtek
[ 1557.114264] usb 1-2: SerialNumber: 123456
[ 1557.156819] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 000000003c5ef4fe, val ffffffffc15cab78
[ 1557.219473] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 000000003c5ef4fe, val ffffffffc15cab78
```
```
Linux 4.19.0-5-amd64 #1 SMP Debian 4.19.37-5+deb10u2 (2019-08-08) x86_64 GNU/Linux
```
Any clues on how to address this problem?

itay-grudev, 2019-12-15:
I just recompiled it and tried `insmod` directly on the `.ko` file and it failed with:

```
ERROR: could not insert module /var/lib/dkms/rtl8812au/5.6.4.2/build/8812au.ko: Unknown symbol in module
```

asteriosos, 2020-05-09:
I got the same on a kali linux just now. 

kali 5.5.0 amd64

asteriosos, 2020-05-09:
i got it, before i had to load the module mac80211

Ludo444, 2020-05-16:
Same issue here

`dmesg`:
```
[ 1682.164901] usb 1-3: new high-speed USB device number 8 using xhci_hcd
[ 1682.333625] usb 1-3: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[ 1682.333657] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1682.333676] usb 1-3: Product: 802.11ac NIC
[ 1682.333687] usb 1-3: Manufacturer: Realtek
[ 1682.333698] usb 1-3: SerialNumber: 123456
[ 1682.363388] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 00000000cdab4977, val ffffffffc121b06c
[ 1682.388763] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 00000000570c301c, val ffffffffc143906c
[ 1770.968632] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 00000000e9796573, val ffffffffc165706c
```

Manual installation `insmod /var/lib/dkms/realtek-rtl88xxau/5.6.4.2~20200313/4.19.0-5-amd64/x86_64/module/88XXau.ko`:

```
insmod: ERROR: could not insert module /var/lib/dkms/realtek-rtl88xxau/5.6.4.2~20200313/4.19.0-5-amd64/x86_64/module/88XXau.ko: Invalid module format
```

Ludo444, 2020-05-16:
Same problem here

`dmesg`:
```
[ 2111.571424] usb 1-3: USB disconnect, device number 8
[ 2117.467726] usb 1-3: new high-speed USB device number 9 using xhci_hcd
[ 2117.647704] usb 1-3: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[ 2117.647731] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2117.647750] usb 1-3: Product: 802.11ac NIC
[ 2117.647761] usb 1-3: Manufacturer: Realtek
[ 2117.647772] usb 1-3: SerialNumber: 123456
[ 2117.678760] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 00000000265866ea, val ffffffffc1a857c8
[ 2117.715189] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 000000003e11bfb6, val ffffffffc1f897c8
[ 2294.881740] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 000000009775d90f, val ffffffffc21c806c
[ 2552.849153] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 000000004626b0af, val ffffffffc121db54
[ 2808.408423] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 000000002194768c, val ffffffffc150bb54
```

Manual `insmod /var/lib/dkms/rtl8812au/5.7.0/4.19.0-5-amd64/x86_64/module/88XXau.ko`:
```
insmod: ERROR: could not insert module /var/lib/dkms/rtl8812au/5.7.0/4.19.0-5-amd64/x86_64/module/88XXau.ko: Invalid module format
```

Ludo444, 2020-05-16:
Ok, so turns out something got corrupted on my system with linux headers. It was compiling for different kernel version that was running. I upgraded the kernel and headers and it seems working fine.

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)

pushkaraj-g-patil, 2021-12-31:
> Ok, so turns out something got corrupted on my system with linux headers. It was compiling for different kernel version than what was running. I upgraded the kernel and headers and it seems working fine.

@Ludo444 : If possible, Could you please help me with recovery steps, I tried but still issue persist

Ludo444, 2022-01-06:
@pushkaraj-g-patil i think i just installed latest kernel+headers, booted with new kernel and then recompiled the package