aircrack-ng/rtl8812au #522: 5.3.4 missining device in iw list

logan001, 2019-12-21:
With branch 5.3.4 commit 44cda227c31e757cf2835db8607ea8974241ac19 seems that iwconfig show my device 
```
wlp10s0f3u3  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

```

but i can't find my device in iw list anymore and of course i can't create AP anymore.

zand999, 2019-12-24:
I too am having a similar problem on Manjaro. I am running through VirtualBox.

~ % iwconfig
enp0s3    no wireless extensions.

enp0s8    no wireless extensions.

enp0s11u1  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lo        no wireless extensions.

~ %  iw dev enp0s11u1 set type monitor
command failed: No such device (-19)

jstruebel, 2019-12-28:
I'm having the same problem. I'm compiling the drivers on a NanoPi NEO2 under Armbian 5.92 (Kernel 4.19.63) using the CONFIG_PLATFORM_ARM64_RPI since that seemed like it might be the closest to my platform.

As an additional datapoint, I compiled and installed on a Raspberry Pi 3B under the latest Raspbian and it does show up in iw list. Maybe something is broken/missing when we build for 64-bit?

jstruebel, 2019-12-28:
One additional thing in my systems. On the Raspberry Pi an lsmod shows that cfg80211 depends on 88XXau, however on the NanoPi NEO2 cfg80211 shows no dependencies.

Just to be clear on my configurations:
NanoPi NEO2 - aarch64, compiled with CONFIG_PLATFORM_ARM64_RPI = y
Rasbperry Pi 3B - armv7l, compiled with CONFIG_PLATFORM_ARM_RPI = y

kimocoder, 2020-01-02:
I'll take a look at this right away

kimocoder, 2020-01-02:
The issue is now resolved. Please update and compile/build/install again.

closed 2020-01-02 by kimocoder

jstruebel, 2020-01-03:
Can confirm that this issue is resolved, however I'm now having a slightly different issue described in https://github.com/aircrack-ng/rtl8812au/issues/494#issuecomment-570462461.