aircrack-ng/rtl8812au #523: TP-Link Archer slow speeds

RoundRobin2, 2019-12-26:
Hi, I just recently got hold of a Archer T9UH Euv1 wireless adapter and have been trying to get to work under Fedora 31.
With the latest branch 5.6.4.2 I only get around 30Mbps transfer speed.
My existing wireless adapter an Asus N53 can get 100Mbps which is expected.
I noticed there were several branches, so I tried each one in turn going starting with the highest version number and going down.
I found that the version 5.2.20 gets me the 100Mbps transfer speed i was expecting.
So I will use that version for now.

I have included some infomation that I was using to try and debug the problem below.
<details><summary>dmesg output</summary>
[    0.000000] microcode: microcode updated early to revision 0x27, date = 2019-02-26
[    0.000000] Linux version 5.3.16-300.fc31.x86_64 (mockbuild@bkernel04.phx2.fedoraproject.org) (gcc version 9.2.1 20190827 (Red Hat 9.2.1-1) (GCC)) #1 SMP Fri Dec 13 17:59:04 UTC 2019
[    0.000000] Command line: BOOT_IMAGE=(hd2,gpt2)/vmlinuz-5.3.16-300.fc31.x86_64 root=/dev/mapper/fedora00-root ro resume=/dev/mapper/fedora00-swap rd.lvm.lv=fedora00/root rd.lvm.lv=fedora00/swap rhgb quiet rd.driver.blacklist=nouveau systemd.unified_cgroup_hierarchy=0 pci=nomsi
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000ca6b8fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ca6b9000-0x00000000ca6bffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000ca6c0000-0x00000000cab13fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cab14000-0x00000000caf79fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000caf7a000-0x00000000dd721fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dd722000-0x00000000dd92afff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dd92b000-0x00000000dd943fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dd944000-0x00000000dde7bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dde7c000-0x00000000deffefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000defff000-0x00000000deffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000041effffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0xca132018-0xca151a57] usable ==> usable
[    0.000000] e820: update [mem 0xca132018-0xca151a57] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000059000-0x000000000009efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x000000000009ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x00000000ca132017] usable
[    0.000000] reserve setup_data: [mem 0x00000000ca132018-0x00000000ca151a57] usable
[    0.000000] reserve setup_data: [mem 0x00000000ca151a58-0x00000000ca6b8fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000ca6b9000-0x00000000ca6bffff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000ca6c0000-0x00000000cab13fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000cab14000-0x00000000caf79fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000caf7a000-0x00000000dd721fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000dd722000-0x00000000dd92afff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000dd92b000-0x00000000dd943fff] ACPI data
[    0.000000] reserve setup_data: [mem 0x00000000dd944000-0x00000000dde7bfff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000dde7c000-0x00000000deffefff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000defff000-0x00000000deffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x000000041effffff] usable
[    0.000000] efi: EFI v2.31 by American Megatrends
[    0.000000] efi:  ESRT=0xdef53818  ACPI=0xdd930000  ACPI 2.0=0xdd930000  SMBIOS=0xf04c0  MPS=0xfd510 
[    0.000000] secureboot: Secure boot disabled
[    0.000000] SMBIOS 2.7 present.
[... 2311 lines cut ...]
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
</details>

<details><summary>dkms status</summary>
nvidia, 440.44, 5.3.16-300.fc31.x86_64, x86_64: installed

rtl8812au, 5.6.4.2, 5.3.16-300.fc31.x86_64, x86_64: installed
</details>

ericsaintetienne, 2020-01-13:
Do you still experience this issue? With the latest I get 9 MB/s (72 Mbit/s) but that's the maximum my network allows. In any case well above 30 Mbps.

RoundRobin2, 2020-01-14:
The the issue does appear to have been resolved. Great work!

closed 2020-01-14 by RoundRobin2

harunyasar, 2020-01-24:
I was dealing with the same problem until I have changed the current frequency into 5GHz. I have applied the following instructions and it has been solved even though I couldn't choose the signal with a %85 rate.

```console
merlin@avalon:~$ nmcli device wifi list
```

```
IN-USE  SSID                 MODE   CHAN  RATE        SIGNAL  BARS  SECURITY         
        Virgin Media         Infra  11    130 Mbit/s  97      ▂▄▆█  WPA2 802.1X      
        VM3918602            Infra  11    130 Mbit/s  85      ▂▄▆█  WPA1 WPA2        
*       VM3918602            Infra  44    405 Mbit/s  70      ▂▄▆_  WPA1 WPA2
```

And I have checked the best channel by using that command below.

```console
merlin@avalon:~$ iwlist wlx503eaa2422d0 channel
```

```
wlx503eaa2422d0  32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
          Current Frequency:5.22 GHz (Channel 44)
```

I was not able to switch into the fastest signal above somehow but when I got a speed test it showed up an accurate result.

```console
merlin@avalon:~$ sudo iwconfig <wlan> mode Managed
merlin@avalon:~$ sudo iwconfig <wlan> channel <new channel>
```

I hope this works for you too.

eric-saintetienne, 2020-01-24:
After switching to 5GHz It's normal you couldn't switch to the channel with 85% because ch11 is 2.4GHz.

The speed problem has been resolved already in the latest. Not sure what you're trying to solve here with those `iwconfig` commands.

harunyasar, 2020-01-24:
I'm not trying to solve anything. I have just shared my experience because I had a slow internet connection with TP-LINK Archer T9UH AC1900 on Elementary OS. When I switched into 5GHz it is solved.

> The speed problem has been resolved already in the latest.

I don't think so because I use the latest version and it does not work for me.

eric-saintetienne, 2020-01-24:
This ticket has been closed. Please open a new ticket with the specifics of your setup. Thanks.