aircrack-ng/rtl8812au #525: ERROR: pcap_activate status -1 in 5.6.4.2

ismailjaweedahmed, 2019-12-29:
Using an ALFA AWUS1900 

Facing issues with wash. Below are the error details. 

root@kali:~/Downloads# wash -i wlan0
[X] ERROR: pcap_activate status -1
[X] PCAP: generic error code
couldn't get pcap handle, exiting

root@kali:~/Downloads# lsusb | grep Realtek
Bus 001 Device 003: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter

root@kali:~/Downloads# iwconfig
wlan0     IEEE 802.11b  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency:2.462 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=-100 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

kimocoder, 2020-01-05:
Yes, you will have to update "Reaver" and "Wash"
The repo is over at https://github.com/t6x/reaver-wps-fork-t6x/

Git clone, compile/build, install. Then WASH should be working fine again.

closed 2020-01-05 by kimocoder

pineturtle, 2020-01-05:
@kimocoder Thank you very much. It is working now (y).

Bozzan, 2020-05-20:
Can you please explain exact commands to type in as I get an error. Needless to say I am a beginner

Dileesha-abilash, 2022-07-15:
just type  sudo airodump-ng -i <your adapter>