aircrack-ng/rtl8812au #526: hostapd-wpe is not working with Alfa card AWUS036ACH (rtl8812au)

Zatyam, 2019-12-30:
I am running v2.9 of hostapd-wpe on Kali Linux:

**hostapd-WPE v2.9
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
-----------------------------------------------------
WPE (Wireless Pwnage Edition)
This version has been cleverly modified to target
wired and wireless users.
Twitter: @aircrackng**

My Kali Linux version is:

**Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	2019.4
Codename:	kali-rolling
Linux kali 5.3.0-kali3-amd64 #1 SMP Debian 5.3.15-1kali1 (2019-12-09) x86_64 GNU/Linux**

I've installed the rtl8812au drivers on the Kali using the following steps:

- apt install realtek-rtl88xxau-dkms
- git clone https://github.com/aircrack-ng/rtl8812au.git
- ./dkms-install.sh

I'm able to switch the Alfa card to monitor mode by **airmon-ng start wlan0**. However, whenever I'm trying to set up a rogue access point through hostapd-wpe through the command **hostapd-wpe hostapd-wpe.conf**, I get the following error:

**Configuration file: hostapd-wpe.conf
nl80211: deinit ifname=wlan1 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan1: interface state UNINITIALIZED->DISABLED
wlan1: AP-DISABLED 
wlan1: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan1 wasn't started**

Even when my interface is up and running in monitor mode:
**wlan1     IEEE 802.11b  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency:2.457 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=-100 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0**

Is there anything that I need to change in the hostapd-wpe.conf file? Please help.

Mister-X-, 2019-12-30:
I don't think so but you have to kill network managers first `airmon-ng check kill` and put the interface back in managed mode.

Zatyam, 2020-01-01:
Okay, I found that the hostapd-wpe works and it even captures the challenge-response if I keep the Alfa card in managed mode. Thank you so much!

closed 2020-01-01 by Zatyam