aircrack-ng/rtl8812au #533: 88XXau driver crash (5.6.4.2) on 5.4 kernel (debian testing + TP-LINK T2U Nano)

mgondium, 2020-01-09:
88XXau driver (5.6.4.2) crashes on 5.4 kernel from debian testing. Wireless device is a USB TP-LINK T2U Nano v1. (lsusb "ID 2357:011e TP-Link 802.11ac WLAN Adapter").
Connection to 2.4GHz AP works but 5GHz usually fails, crashes on both.
All good on kernel 4.9.

Connecting to 2.4Ghz AP (success)
(kernel taint because of nvidia driver)

```
[   52.344652] ------------[ cut here ]------------
[   52.344686] WARNING: CPU: 0 PID: 1097 at net/wireless/nl80211.c:3153 nl80211_send_chandef+0x14b/0x160 [cfg80211]
[   52.344686] Modules linked in: 88XXau(OE) ctr ccm bbswitch(OE) cpufreq_powersave cpufreq_userspace cpufreq_conservative binfmt_misc nls_ascii nls_cp437 vfat fat intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm snd_hda_codec_hdmi irqbypass crct10dif_pclmul crc32_pclmul ath9k snd_hda_codec_realtek ath9k_common ath9k_hw snd_hda_codec_generic ledtrig_audio ath uvcvideo videobuf2_vmalloc videobuf2_memops snd_hda_intel videobuf2_v4l2 snd_intel_nhlt mac80211 videobuf2_common ghash_clmulni_intel cryptd efi_pstore intel_cstate videodev snd_hda_codec intel_uncore asus_nb_wmi snd_hda_core joydev asus_wmi intel_rapl_perf sparse_keymap serio_raw mc wmi_bmof snd_hwdep efivars pcspkr iTCO_wdt iTCO_vendor_support snd_pcm rtsx_pci_ms sg cfg80211 watchdog snd_timer memstick ipmi_devintf rfkill snd mei_me ipmi_msghandler libarc4 mei soundcore ac evdev asus_wireless efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic sr_mod sd_mod cdrom i915
[   52.344721]  ahci libahci libata mxm_wmi rtsx_pci_sdmmc mmc_core xhci_pci xhci_hcd i2c_algo_bit ehci_pci ehci_hcd drm_kms_helper crc32c_intel r8169 usbcore scsi_mod i2c_i801 psmouse lpc_ich realtek libphy rtsx_pci mfd_core drm usb_common wmi battery video button
[   52.344736] CPU: 0 PID: 1097 Comm: RTW_CMD_THREAD Tainted: P           OE     5.4.0-2-amd64 #1 Debian 5.4.8-1
[   52.344737] Hardware name: ASUSTeK COMPUTER INC. X550CL/X550CL, BIOS X550CL.207 07/10/2014
[   52.344753] RIP: 0010:nl80211_send_chandef+0x14b/0x160 [cfg80211]
[   52.344756] Code: 00 00 be a1 00 00 00 48 89 ef 89 44 24 04 e8 fc 59 64 e3 85 c0 0f 84 7b ff ff ff 41 bc 97 ff ff ff e9 70 ff ff ff 31 c0 eb a7 <0f> 0b 41 bc ea ff ff ff e9 5f ff ff ff e8 b3 dd 2a e3 0f 1f 00 0f
[   52.344757] RSP: 0018:ffffb7ae409ebd80 EFLAGS: 00010246
[   52.344759] RAX: 0000000000000000 RBX: ffffb7ae409ebe08 RCX: 00000000000000ae
[   52.344760] RDX: 00000000402677ec RSI: 00000000ffff25a6 RDI: ffffb7ae409ebe08
[   52.344761] RBP: ffff994fa4b41a00 R08: 0000000000000eac R09: ffff994fa2d0f01c
[   52.344761] R10: 0000000000000000 R11: 0000000000000000 R12: ffffb7ae409ebe08
[   52.344762] R13: 0000000000000000 R14: ffff994fa4b41a00 R15: ffff994fa2d0f014
[   52.344764] FS:  0000000000000000(0000) GS:ffff994faac00000(0000) knlGS:0000000000000000
[   52.344765] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   52.344766] CR2: 000056366c2fc000 CR3: 000000011f640006 CR4: 00000000001606f0
[   52.344767] Call Trace:
[   52.344787]  nl80211_ch_switch_notify.constprop.0+0xcd/0x170 [cfg80211]
[   52.344842]  rtw_cfg80211_ch_switch_notify+0x138/0x147 [88XXau]
[   52.344877]  ? rtw_chk_start_clnt_join+0x79/0x79 [88XXau]
[   52.344908]  rtw_chk_start_clnt_join+0x72/0x79 [88XXau]
[   52.344939]  join_cmd_hdl+0x267/0x373 [88XXau]
[   52.344959]  rtw_cmd_thread+0x295/0x3ed [88XXau]
[   52.344964]  kthread+0xf9/0x130
[   52.344984]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[   52.344987]  ? kthread_park+0x90/0x90
[   52.344990]  ret_from_fork+0x35/0x40
[   52.344993] ---[ end trace 0b4425838f503254 ]---
```

Connecting to 5GHz AP (success)

```
[  148.028438] ------------[ cut here ]------------
[  148.028472] WARNING: CPU: 0 PID: 1097 at net/wireless/nl80211.c:3153 nl80211_send_chandef+0x14b/0x160 [cfg80211]
[  148.028473] Modules linked in: 88XXau(OE) ctr ccm bbswitch(OE) cpufreq_powersave cpufreq_userspace cpufreq_conservative binfmt_misc nls_ascii nls_cp437 vfat fat intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm snd_hda_codec_hdmi irqbypass crct10dif_pclmul crc32_pclmul ath9k snd_hda_codec_realtek ath9k_common ath9k_hw snd_hda_codec_generic ledtrig_audio ath uvcvideo videobuf2_vmalloc videobuf2_memops snd_hda_intel videobuf2_v4l2 snd_intel_nhlt mac80211 videobuf2_common ghash_clmulni_intel cryptd efi_pstore intel_cstate videodev snd_hda_codec intel_uncore asus_nb_wmi snd_hda_core joydev asus_wmi intel_rapl_perf sparse_keymap serio_raw mc wmi_bmof snd_hwdep efivars pcspkr iTCO_wdt iTCO_vendor_support snd_pcm rtsx_pci_ms sg cfg80211 watchdog snd_timer memstick ipmi_devintf rfkill snd mei_me ipmi_msghandler libarc4 mei soundcore ac evdev asus_wireless efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic sr_mod sd_mod cdrom i915
[  148.028507]  ahci libahci libata mxm_wmi rtsx_pci_sdmmc mmc_core xhci_pci xhci_hcd i2c_algo_bit ehci_pci ehci_hcd drm_kms_helper crc32c_intel r8169 usbcore scsi_mod i2c_i801 psmouse lpc_ich realtek libphy rtsx_pci mfd_core drm usb_common wmi battery video button
[  148.028522] CPU: 0 PID: 1097 Comm: RTW_CMD_THREAD Tainted: P        W  OE     5.4.0-2-amd64 #1 Debian 5.4.8-1
[  148.028523] Hardware name: ASUSTeK COMPUTER INC. X550CL/X550CL, BIOS X550CL.207 07/10/2014
[  148.028540] RIP: 0010:nl80211_send_chandef+0x14b/0x160 [cfg80211]
[  148.028543] Code: 00 00 be a1 00 00 00 48 89 ef 89 44 24 04 e8 fc 59 64 e3 85 c0 0f 84 7b ff ff ff 41 bc 97 ff ff ff e9 70 ff ff ff 31 c0 eb a7 <0f> 0b 41 bc ea ff ff ff e9 5f ff ff ff e8 b3 dd 2a e3 0f 1f 00 0f
[  148.028544] RSP: 0018:ffffb7ae409ebd80 EFLAGS: 00010246
[  148.028545] RAX: 0000000000000000 RBX: ffffb7ae409ebe08 RCX: 00000000000000ae
[  148.028546] RDX: 00000000402677ec RSI: 00000000ffff306c RDI: ffffb7ae409ebe08
[  148.028547] RBP: ffff994fa4b1f500 R08: 0000000000000eac R09: 0000000000000000
[  148.028548] R10: 000000000000001a R11: 0000000000000000 R12: ffffb7ae409ebe08
[  148.028549] R13: 0000000000000000 R14: ffff994fa4b1f500 R15: ffff994fa4f7f014
[  148.028551] FS:  0000000000000000(0000) GS:ffff994faac00000(0000) knlGS:0000000000000000
[  148.028552] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  148.028553] CR2: 000056366c315328 CR3: 000000012053c002 CR4: 00000000001606f0
[  148.028554] Call Trace:
[  148.028574]  nl80211_ch_switch_notify.constprop.0+0xcd/0x170 [cfg80211]
[  148.028628]  rtw_cfg80211_ch_switch_notify+0x138/0x147 [88XXau]
[  148.028663]  ? rtw_chk_start_clnt_join+0x79/0x79 [88XXau]
[  148.028694]  rtw_chk_start_clnt_join+0x72/0x79 [88XXau]
[  148.028726]  join_cmd_hdl+0x267/0x373 [88XXau]
[  148.028747]  rtw_cmd_thread+0x295/0x3ed [88XXau]
[  148.028752]  kthread+0xf9/0x130
[  148.028772]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[  148.028774]  ? kthread_park+0x90/0x90
[  148.028778]  ret_from_fork+0x35/0x40
[  148.028781] ---[ end trace 0b4425838f503255 ]---
```

3skr0, 2020-01-12:
Same, with the last release [(rtl8812au v5.2.20.2~20181213)  ](https://github.com/aircrack-ng/rtl8812au/releases/tag/v1.0-DEV):

```
Jan 12 15:46:32 localhost kernel: [ 9964.769031] ------------[ cut here ]------------
Jan 12 15:46:32 localhost kernel: [ 9964.769033] kernel BUG at /build/linux-hwe-FLYqTt/linux-hwe-5.0.0/mm/slub.c:305!
Jan 12 15:46:32 localhost kernel: [ 9964.769058] invalid opcode: 0000 [#1] SMP PTI
Jan 12 15:46:32 localhost kernel: [ 9964.769060] CPU: 2 PID: 7060 Comm: iw Tainted: G           OE     5.0.0-37-generic #40~18.04.1-Ubuntu
Jan 12 15:46:32 localhost kernel: [ 9964.769061] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.12.1-0-ga5cab58e9a3f-prebuilt.qemu.org 04/01/2014
Jan 12 15:46:32 localhost kernel: [ 9964.769067] RIP: 0010:__slab_free+0x188/0x330
Jan 12 15:46:32 localhost kernel: [ 9964.769069] Code: 00 48 89 c7 fa 66 0f 1f 44 00 00 f0 49 0f ba 2c 24 00 72 68 4d 3b 6c 24 20 74 11 49 0f ba 34 24 00 57 9d 0f 1f 44 00 00 eb a9 <0f> 0b 49 3b 5c 24 28 75 e8 48 8b 45 88 49 89 4c 24 28 49 89 44 24
Jan 12 15:46:32 localhost kernel: [ 9964.769070] RSP: 0018:ffffb2c280c7b8f0 EFLAGS: 00010246
Jan 12 15:46:32 localhost kernel: [ 9964.769072] RAX: ffff968da8764600 RBX: 0000000080100009 RCX: ffff968da8764600
Jan 12 15:46:32 localhost kernel: [ 9964.769073] RDX: ffff968da8764600 RSI: ffffe7ba09a1d900 RDI: ffff968db7403080
Jan 12 15:46:32 localhost kernel: [ 9964.769074] RBP: ffffb2c280c7b990 R08: 0000000000000001 R09: ffffffff9662d9f3
Jan 12 15:46:32 localhost kernel: [ 9964.769075] R10: 0000000000000001 R11: 0000000000000034 R12: ffffe7ba09a1d900
Jan 12 15:46:32 localhost kernel: [ 9964.769075] R13: ffff968da8764600 R14: ffff968db7403080 R15: ffff968da8764600
Jan 12 15:46:32 localhost kernel: [ 9964.769077] FS:  00007f2cfdf3a240(0000) GS:ffff968db7b00000(0000) knlGS:0000000000000000
Jan 12 15:46:32 localhost kernel: [ 9964.769078] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan 12 15:46:32 localhost kernel: [ 9964.769079] CR2: 000056438a616cd0 CR3: 000000026ab8c000 CR4: 00000000000006e0
Jan 12 15:46:32 localhost kernel: [ 9964.769082] Call Trace:
Jan 12 15:46:32 localhost kernel: [ 9964.769096]  ? schedule+0x2c/0x70
Jan 12 15:46:32 localhost kernel: [ 9964.769098]  ? schedule_timeout+0x1db/0x360
Jan 12 15:46:32 localhost kernel: [ 9964.769101]  ? pcpu_free_area+0x1c7/0x2f0
Jan 12 15:46:32 localhost kernel: [ 9964.769104]  ? kvfree+0x33/0x40
Jan 12 15:46:32 localhost kernel: [ 9964.769107]  kfree+0x169/0x180
Jan 12 15:46:32 localhost kernel: [ 9964.769108]  ? kfree+0x169/0x180
Jan 12 15:46:32 localhost kernel: [ 9964.769110]  kvfree+0x33/0x40
Jan 12 15:46:32 localhost kernel: [ 9964.769114]  free_netdev+0x23/0x130
Jan 12 15:46:32 localhost kernel: [ 9964.769117]  netdev_run_todo+0x286/0x300
Jan 12 15:46:32 localhost kernel: [ 9964.769135]  rtnl_unlock+0xe/0x10
Jan 12 15:46:32 localhost kernel: [ 9964.769204]  nl80211_post_doit+0x6a/0x80 [cfg80211]
Jan 12 15:46:32 localhost kernel: [ 9964.769211]  genl_family_rcv_msg+0x231/0x420
Jan 12 15:46:32 localhost kernel: [ 9964.769215]  ? __wake_up_common_lock+0x8e/0xc0
Jan 12 15:46:32 localhost kernel: [ 9964.769217]  genl_rcv_msg+0x4c/0x90
Jan 12 15:46:32 localhost kernel: [ 9964.769219]  ? genl_family_rcv_msg+0x420/0x420
Jan 12 15:46:32 localhost kernel: [ 9964.769220]  netlink_rcv_skb+0x51/0x120
Jan 12 15:46:32 localhost kernel: [ 9964.769222]  genl_rcv+0x28/0x40
Jan 12 15:46:32 localhost kernel: [ 9964.769224]  netlink_unicast+0x1a4/0x250
Jan 12 15:46:32 localhost kernel: [ 9964.769226]  netlink_sendmsg+0x2d1/0x3d0
Jan 12 15:46:32 localhost kernel: [ 9964.769233]  sock_sendmsg+0x3e/0x50
Jan 12 15:46:32 localhost kernel: [ 9964.769235]  ___sys_sendmsg+0x2a0/0x2f0
Jan 12 15:46:32 localhost kernel: [ 9964.769237]  ? sock_destroy_inode+0x2f/0x40
Jan 12 15:46:32 localhost kernel: [ 9964.769240]  ? destroy_inode+0x3e/0x60
Jan 12 15:46:32 localhost kernel: [ 9964.769241]  ? evict+0x139/0x1a0
Jan 12 15:46:32 localhost kernel: [ 9964.769243]  ? _cond_resched+0x19/0x40
Jan 12 15:46:32 localhost kernel: [ 9964.769245]  ? __dentry_kill+0x12d/0x170
Jan 12 15:46:32 localhost kernel: [ 9964.769247]  __sys_sendmsg+0x63/0xa0
Jan 12 15:46:32 localhost kernel: [ 9964.769248]  ? __sys_sendmsg+0x63/0xa0
Jan 12 15:46:32 localhost kernel: [ 9964.769250]  __x64_sys_sendmsg+0x1f/0x30
Jan 12 15:46:32 localhost kernel: [ 9964.769254]  do_syscall_64+0x5a/0x120
Jan 12 15:46:32 localhost kernel: [ 9964.769257]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Jan 12 15:46:32 localhost kernel: [ 9964.769259] RIP: 0033:0x7f2cfd647d04
Jan 12 15:46:32 localhost kernel: [ 9964.769261] Code: 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b5 0f 1f 80 00 00 00 00 48 8d 05 01 dc 2c 00 8b 00 85 c0 75 13 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 54 f3 c3 66 90 41 54 55 41 89 d4 53 48 89 f5
Jan 12 15:46:32 localhost kernel: [ 9964.769262] RSP: 002b:00007ffeb14bd768 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
Jan 12 15:46:32 localhost kernel: [ 9964.769264] RAX: ffffffffffffffda RBX: 000055a8e6d3e990 RCX: 00007f2cfd647d04
Jan 12 15:46:32 localhost kernel: [ 9964.769265] RDX: 0000000000000000 RSI: 00007ffeb14bd7a0 RDI: 0000000000000003
Jan 12 15:46:32 localhost kernel: [ 9964.769265] RBP: 000055a8e6d43d80 R08: 0000000000000001 R09: 000055a8e6d3eab0
Jan 12 15:46:32 localhost kernel: [ 9964.769266] R10: 000055a8e53c10a0 R11: 0000000000000246 R12: 000055a8e6d43ec0
[... cut ...] raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear hid_generic usbhid hid amdgpu(OE) virtio_net virtio_gpu amdkcl(OE) amd_iommu_v2 amdttm(OE) amd_sched(OE) net_failover psmouse ttm i2c_algo_bit virtio_scsi failover ahci libahci drm_kms_helper
Jan 12 15:46:32 localhost kernel: [ 9964.769401]  syscopyarea sysfillrect sysimgblt fb_sys_fops drm
Jan 12 15:46:32 localhost kernel: [ 9964.769412] ---[ end trace 32fcbf8f1021b857 ]---
```

3skr0, 2020-01-12:
Duplicate of #498

kimocoder, 2020-01-12:
@3skr0 Your using a older driver, which is more or less left behind.

Stable branch is v5.3.4
Experimental branch is v5.6.4.2

kimocoder, 2020-01-25:
Closing, as the kernel oops is related to the fact that your using a branch which does not support newer kernels (and is left behind)

mgondium, 2020-01-26:
I am the original reporter and I'm using 5.6.4.2, isn't that where all latest commits go?
I still get the crash on debian kernel 5.4.

kimocoder, 2020-01-26:
I'm on Kali (debian) v5.4 and it's working just fine

mgondium, 2020-01-26:
> I'm on Kali (debian) v5.4 and it's working just fine on RTL8812AU, don't have any regressions no more. But I'll do proper test shortly

Weird, I don't use any other distribution to test on, but I get this on multiple computers.

```
[sun jan 26 16:29:39 2020] ------------[ cut here ]------------
[sun jan 26 16:29:39 2020] WARNING: CPU: 2 PID: 6906 at net/wireless/nl80211.c:3153 nl80211_send_chandef+0x14b/0x160 [cfg80211]
[sun jan 26 16:29:39 2020] Modules linked in: 88XXau(OE) cfg80211 ipt_REJECT nf_reject_ipv4 nft_counter xt_state xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables nfnetlink fuse rfkill btrfs snd_hda_codec_hdmi snd_hda_codec_analog xor snd_hda_codec_generic zstd_compress ledtrig_audio raid6_pq nvidia_drm(POE) libcrc32c zstd_decompress drm_kms_helper coretemp snd_hda_intel snd_intel_nhlt drm snd_hda_codec kvm_intel nvidia_modeset(POE) snd_hda_core kvm snd_hwdep joydev snd_pcm irqbypass serio_raw pcspkr sg snd_timer iTCO_wdt iTCO_vendor_support snd watchdog soundcore x38_edac asus_atk0110 evdev nvidia(POE) ipmi_devintf ipmi_msghandler parport_pc ppdev lp parport ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic uas usb_storage sr_mod cdrom sd_mod hid_generic usbhid hid ahci libahci libata scsi_mod xhci_pci r8169 i2c_i801 lpc_ich xhci_hcd realtek mfd_core libphy ehci_pci uhci_hcd ehci_hcd sky2 usbcore usb_common button [last unloaded: 88XXau]
[sun jan 26 16:29:39 2020] CPU: 2 PID: 6906 Comm: RTW_CMD_THREAD Tainted: P        W  OE     5.4.0-3-amd64 #1 Debian 5.4.13-1
[sun jan 26 16:29:39 2020] Hardware name: System manufacturer P5E/P5E, BIOS 1201    02/19/2009
[sun jan 26 16:29:39 2020] RIP: 0010:nl80211_send_chandef+0x14b/0x160 [cfg80211]
[sun jan 26 16:29:39 2020] Code: 00 00 be a1 00 00 00 48 89 ef 89 44 24 04 e8 cc 53 6c e9 85 c0 0f 84 7b ff ff ff 41 bc 97 ff ff ff e9 70 ff ff ff 31 c0 eb a7 <0f> 0b 41 bc ea ff ff ff e9 5f ff ff ff e8 43 bf 32 e9 0f 1f 00 66
[sun jan 26 16:29:39 2020] RSP: 0018:ffffb410c03a3d80 EFLAGS: 00010246
[sun jan 26 16:29:39 2020] RAX: 0000000000000000 RBX: ffffb410c03a3e08 RCX: 0000000000000010
[sun jan 26 16:29:39 2020] RDX: 00000000c00b37ec RSI: 00000000ffff25a6 RDI: ffffb410c03a3e08
[sun jan 26 16:29:39 2020] RBP: ffff8ba8e913ff00 R08: 0000000000000eac R09: ffff8ba8bee1401c
[sun jan 26 16:29:39 2020] R10: 0000000000000000 R11: 0000000000000000 R12: ffffb410c03a3e08
[sun jan 26 16:29:39 2020] R13: 0000000000000000 R14: ffff8ba8e913ff00 R15: ffff8ba8bee14014
[sun jan 26 16:29:39 2020] FS:  0000000000000000(0000) GS:ffff8ba8f7b00000(0000) knlGS:0000000000000000
[sun jan 26 16:29:39 2020] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[sun jan 26 16:29:39 2020] CR2: 000055a10f348340 CR3: 00000002312b0000 CR4: 00000000000006e0
[sun jan 26 16:29:39 2020] Call Trace:
[sun jan 26 16:29:39 2020]  nl80211_ch_switch_notify.constprop.0+0xcd/0x170 [cfg80211]
[sun jan 26 16:29:39 2020]  rtw_cfg80211_ch_switch_notify+0x138/0x147 [88XXau]
[sun jan 26 16:29:39 2020]  ? rtw_chk_start_clnt_join+0x79/0x79 [88XXau]
[sun jan 26 16:29:39 2020]  rtw_chk_start_clnt_join+0x72/0x79 [88XXau]
[sun jan 26 16:29:39 2020]  join_cmd_hdl+0x267/0x373 [88XXau]
[sun jan 26 16:29:39 2020]  rtw_cmd_thread+0x295/0x3ed [88XXau]
[sun jan 26 16:29:39 2020]  kthread+0xf9/0x130
[sun jan 26 16:29:39 2020]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[sun jan 26 16:29:39 2020]  ? kthread_park+0x90/0x90
[sun jan 26 16:29:39 2020]  ret_from_fork+0x35/0x40
[sun jan 26 16:29:39 2020] ---[ end trace d84a7ce31f5964d1 ]---
[sun jan 26 16:29:39 2020] IPv6: ADDRCONF(NETDEV_CHANGE): wlx503eaacf20f2: link becomes ready
user@debian-c2q-x64:~$ uname -a
Linux debian-c2q-x64 5.4.0-3-amd64 #1 SMP Debian 5.4.13-1 (2020-01-19) x86_64 GNU/Linux
```

kimocoder, 2020-01-26:
Had that om my of known ones, haven't seen it after latest patches, but I'll do some test shortly. Will look at that code again

crispywizard, 2020-02-01:
I get same error when trying to connect via NetworkManager

Linux kali 5.4.0-kali3-amd64 #1 SMP Debian 5.4.13-1kali1 (2020-01-20) x86_64 GNU/Linux
[dmesg.txt](https://github.com/aircrack-ng/rtl8812au/files/4143322/dmesg.txt)
[modinfo.txt](https://github.com/aircrack-ng/rtl8812au/files/4143323/modinfo.txt)

kimocoder, 2020-03-08:
I just pushed some stash for branch v5.6.4.2, it works, but however, the "nl80211_send_chandef" regression is still here. I'll dive into it.

[image]

kimocoder, 2020-03-08:
This error is a kernel bug introduced over here https://github.com/torvalds/linux/commit/f43e5210c739fe76a4b0ed851559d6902f20ceb1#diff-6c2a071a3bad859b6dabb535217e11c4

There's nothing we can do about it, besides wait for a newer kernel release, it doesn't seem to impact functions in the driver, if it does, I haven't noticed it yet.
More information/bug report is found below.

https://bugzilla.kernel.org/show_bug.cgi?id=205131

closed 2020-03-08 by kimocoder

qbarnes, 2020-05-05:
@kimocoder, I don't think this is that bug you're quoting.  I'm getting almost an identical problem running a RHEL 8.2 (4.18.0-193.el8) kernel and I've verified that its source has the f43e521 patch applied.

On branch v5.6.4.2 in os_dep/linux/ioctl_cfg80211.c in function rtw_cfg80211_ch_switch_notify(), I'm not seeing how the local variable `chdef` gets initialized.  Neither functions rtw_chbw_to_cfg80211_chan_def() nor cfg80211_ch_switch_started_notify() are ensuring the variable is initialized before attempting to use its contents.

Do you agree with my assessment?

satmandu, 2020-06-17:
@guspoveda can you paste in the changed code you used to properly initialize ```cfg80211_chan_def chdef```?

I'm seeing this here : https://github.com/aircrack-ng/rtl8812au/blob/76c3cf81fdc71af4338571b6404beb2ba3835a62/os_dep/linux/ioctl_cfg80211.c#L3624
```
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
	struct cfg80211_chan_def *pch_def;
#endif
	struct ieee80211_channel *pch;
	int ret = 0;

#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
	pch_def = (struct cfg80211_chan_def *)(&params->chandef);
	pch = (struct ieee80211_channel *) pch_def->chan;
#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 31))
	pch = (struct ieee80211_channel *)(params->channel);
#endif
```

(This is the same issue I reported again here: https://github.com/aircrack-ng/rtl8812au/issues/660 )

guspoveda, 2020-06-17:
> @guspoveda can you paste in the changed code you used to properly initialize `cfg80211_chan_def chdef`?
> 
> I'm seeing this here :
> 
> https://github.com/aircrack-ng/rtl8812au/blob/76c3cf81fdc71af4338571b6404beb2ba3835a62/os_dep/linux/ioctl_cfg80211.c#L8049
> 
> ```
> #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
> 	struct cfg80211_chan_def *chdef = (struct cfg80211_chan_def *)(&setup->chandef);
> #endif
> 	struct ieee80211_channel *chan;
> 	u8 ch, bw, offset;
> #endif
> ```
> 
> (This is the same issue I reported again here: #660 )

@satmandu, The issue is within the function called rtw_cfg80211_ch_switch_notify.  I've pasted it below.  You need to initialize the struct cfg80211_chan_def chdef.  I've done that below like this:
`struct cfg80211_chan_def chdef = {};`

It should be around line 416 in the file you referenced above.  That is the only location I found this issue.  The lines you reference are pointers, so it doesn't have this issue as they are assigned below to one struct or another.

```
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 5, 0))
u8 rtw_cfg80211_ch_switch_notify(_adapter *adapter, u8 ch, u8 bw, u8 offset, u8 ht)
{
	struct wiphy *wiphy = adapter_to_wiphy(adapter);
	u8 ret = _SUCCESS;

#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
	struct cfg80211_chan_def chdef  = {};

	ret = rtw_chbw_to_cfg80211_chan_def(wiphy, &chdef, ch, bw, offset, ht);
	if (ret != _SUCCESS)
		goto exit;
```

satmandu, 2020-06-18:
Changing that to 
```
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
        struct cfg80211_chan_def chdef = {};

        ret = rtw_chbw_to_cfg80211_chan_def(wiphy, &chdef, ch, bw, offset, ht);
        if (ret != _SUCCESS)
                goto exit;
```

Doesn't appear to solve my issue.

I'm still seeing:
```
[   28.933729] [phydm_la_set_buff_mode] Warning!
[   28.933735] start_addr=(0x0), end_addr=(0x0), buffer_size=(0x0), smp_number_max=(0)
[   30.117336] ------------[ cut here ]------------
[   30.117433] WARNING: CPU: 3 PID: 1772 at net/wireless/nl80211.c:16299 cfg80211_ch_switch_notify+0x160/0x168 [cfg80211]
[   30.117435] Modules linked in: iptable_nat nf_nat nf_conntrack nf_defrag_ipv4 iptable_mangle iptable_filter bluetooth ecdh_generic ecc zram zsmalloc binfmt_misc brcmfmac bcm2835_codec(C) brcmutil raspberrypi_hwmon bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc bcm2835_mmal_vchiq(C) videobuf2_memops videobuf2_v4l2 videobuf2_common vc_sm_cma(C) videodev rpivid_mem mc uio_pdrv_genirq uio sch_fq_codel i2c_dev drm drm_panel_orientation_quirks ip_tables x_tables ipv6 nf_defrag_ipv6 btrfs zstd_decompress zstd_compress xxhash raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor xor_neon raid6_pq raid1 raid0 linear md_mod dm_mirror dm_region_hash dm_log dm_mod squashfs zfs(O) zunicode(O) zlua(PO) zcommon(O) znvpair(O) zavl(O) icp(O) spl(O) zlib_deflate 88XXau(O) sha256_generic libsha256 cfg80211 rfkill spidev i2c_bcm2835 spi_bcm2835
[   30.117505] CPU: 3 PID: 1772 Comm: RTW_CMD_THREAD Tainted: P         C O      5.4.47-v8+ #1322
[   30.117507] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   30.117510] pstate: 60000005 (nZCv daif -PAN -UAO)
[   30.117549] pc : cfg80211_ch_switch_notify+0x160/0x168 [cfg80211]
[   30.117663] lr : rtw_cfg80211_ch_switch_notify+0xf0/0x140 [88XXau]
[   30.117665] sp : ffffffc01bf8bc90
[   30.117668] x29: ffffffc01bf8bc90 x28: ffffffc008c5b430 
[   30.117672] x27: ffffffc008b15794 x26: 0000000000000003 
[   30.117676] x25: ffffff80efcf9300 x24: 0000000000000001 
[   30.117679] x23: ffffff80f5d81000 x22: ffffff80efcf9000 
[   30.117683] x21: ffffff80f5d7b000 x20: ffffffc01bf8bd28 
[   30.117686] x19: ffffffc011597000 x18: 0000000000000000 
[   30.117689] x17: 0000000000000000 x16: 0000000000000000 
[   30.117692] x15: 0000000000000000 x14: 0000000000000008 
[   30.117695] x13: 7f1201c7ffaa009b x12: 0105c00363fffa03 
[   30.117698] x11: 0000000000000004 x10: 0000000000000002 
[   30.117702] x9 : 0000000000000004 x8 : 0000000000000006 
[   30.117705] x7 : 3d00000000000000 x6 : 0000000000000001 
[   30.117708] x5 : ffffff80efcf9418 x4 : ffffff80efcfa758 
[   30.117711] x3 : 0000000000000000 x2 : 0000000000000000 
[   30.117715] x1 : ffffff80f5d81118 x0 : 0000000000000000 
[   30.117719] Call trace:
[   30.117763]  cfg80211_ch_switch_notify+0x160/0x168 [cfg80211]
[   30.117832]  rtw_cfg80211_ch_switch_notify+0xf0/0x140 [88XXau]
[   30.117893]  rtw_chk_start_clnt_join+0x88/0x90 [88XXau]
[   30.117953]  join_cmd_hdl+0x244/0x34c [88XXau]
[   30.118016]  rtw_cmd_thread+0x278/0x3d0 [88XXau]
[   30.118022]  kthread+0xfc/0x128
[   30.118028]  ret_from_fork+0x10/0x1c
[   30.118031] ---[ end trace 96128662945a5b08 ]---
```

I'd note that currently ```cfg80211_chan_def``` appears to be defined in the Linux kernel includes though: https://github.com/torvalds/linux/blob/1b5044021070efa3259f3e9548dc35d1eb6aa844/include/net/cfg80211.h#L647

satmandu, 2020-06-18:
This might be relevant to us: https://github.com/rtlwifi-linux/rtk_wifi_driver_rtl8822ce/commit/2d7489f142caf11d7864f9149e94db58baa60148?diff=split

guspoveda, 2020-06-18:
@satmandu Let's move this over to the open issue #660.  I've noticed also that several static structs defined within os_dep/linux/ioctl_cfg80211.c and initialized using macros here https://github.com/aircrack-ng/rtl8812au/blob/76c3cf81fdc71af4338571b6404beb2ba3835a62/os_dep/linux/ioctl_cfg80211.c#L106 do not initialize all the fields within the structs.  Could this be an issue?