aircrack-ng/rtl8812au #535: Low TX power at start

MJaoune, 2020-01-12:
When using this driver, the device would have low TX power and can only see very close access points. At some times, the device would be able to find my access point which is a little bit further and would connect to it, then the TX power would be good again.

Why wouldn't the device start at full power from the beginning?

Tried changing CONFIG_CALIBRATE_TX_POWER_TO_MAX, but the problem was still there.

kimocoder, 2020-01-12:
I'll need more information than that.
What distro? Kernel? Driver branch?

MJaoune, 2020-01-12:
Gentoo, kernel 5.4.8.

Driver 5.2.20

Using NetworkManager 1.18.4.

kimocoder, 2020-01-12:
Right. Your also using a branch which is left behind. v5.3.4 is stable, but also got issues with range, however, Realtek fixed the issue on branches above, like on v5.6.4.2 (which is experimental)

rafael39matos, 2020-01-13:
When using this driver (Problem doesn't happen in others), he creates Wlan0 (monitor mode) and Wlan1 (managed mode)
both with txpower setted in 12dbm, so I can see only near AP.

Kali 2019.4

5.3.0-kali2-amd64

RTL8812au v5.6.4.2

kimocoder, 2020-01-13:
12dBm is a hardcoded setting in this driver, it's not the real one. And v5.6.4.2 range is quite good, compared to many, many other devices.

rafael39matos, 2020-01-13:
The number of devices detected has been reduced drastically after installed the RTL8812au v5.6.4.2, and I can not set higher Txpower on Wireless Card

MJaoune, 2020-01-13:
> 12dBm is a hardcoded setting in this driver, it's not the real one. And v5.6.4.2 range is quite good, compared to many, many other devices.

What do you mean "hardcoded"? In which source file? Can't we change it?

MJaoune, 2020-01-17:
Update: The problem is even worse in driver branch 5.6.4.2, just compiled commit d523a0e and tested it, and I couldn't get the device to find near routers except for the ones really really close to it, and even those had a weak signal.

Switched back to 5.2.20 to be able to connect again (After trying multiple times).

closed 2020-03-08 by kimocoder