aircrack-ng/rtl8812au #537: AWUS036ACH [X] ERROR: pcap_activate status -1

Error0x00000001, 2020-01-19:
dkms.conf
PACKAGE_NAME="realtek-rtl88xxau"
PACKAGE_VERSION="5.6.4.2~20200111"
##########################################
root@book:/# uname -a
Linux book 5.3.0-kali2-amd64 #1 SMP Debian 5.3.9-3kali1 (2019-11-20) x86_64 GNU/Linux
##########################################
root@book:/# dkms status
rtl8812au, 5.6.4.2, 5.3.0-kali2-amd64, x86_64: installed
##########################################
root@book:/# airmon-ng 

PHY	Interface	Driver		Chipset
phy1	wlan1		88XXau		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
###################
root@book:/# wash -i wlan1
[X] ERROR: pcap_activate status -1
[X] PCAP: generic error code
couldn't get pcap handle, exiting
#######################

kimocoder, 2020-01-19:
This issue was just fixed and pushed to Kali.
Either update from repo or wait for Kali to update it

kimocoder, 2020-01-20:
The fix has been pushed to Kali. Closing issue.

closed 2020-01-20 by kimocoder