aircrack-ng/rtl8812au #538: Error when building for Jetson Nano

michaelwijnands-zuyd, 2020-01-20:
Hi,

I tried to build the drivers for the Jetson Nano. I changed the makefile to set CONFIG_PLATFORM_ARM_JET_NANO = y and CONFIG_PLATFORM_I386_PC = n.

When using sudo ./dkms-install.sh I get the following error:

`About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.6.4.2/source ->
                 /usr/src/rtl8812au-5.6.4.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.9.140-tegra KSRC=/lib/modules/4.9.140-tegra/build...............(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812au: 5.6.4.2 not found
Error! Bad return status for module build on kernel: 4.9.140-tegra (aarch64)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.9.140-tegra KSRC=/lib/modules/4.9.140-tegra/build............(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812au: 5.6.4.2 not found
Error! Bad return status for module build on kernel: 4.9.140-tegra (aarch64)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.
Finished running dkms install steps.`

When using 'make install' I get the following error:

`install -p -m 644 88XXau.ko  /lib/modules/4.9.140-tegra/kernel/drivers/net/wireless/realtek/rtl8812au/
install: cannot stat '88XXau.ko': No such file or directory
Makefile:2242: recipe for target 'install' failed
make: *** [install] Error 1`

Am I doing something wrong?

kimocoder, 2020-01-20:
Don't know. This one below got it working
https://github.com/ThunderMikey/rtl8812au-aircrack-dkms-aarch64

kimocoder, 2020-01-20:
He made the changes below to make it work
https://github.com/ThunderMikey/rtl8812au/commit/ca267cd406b75f8ef1cedf59420b8e412c3713ee

michaelwijnands-zuyd, 2020-01-20:
Tried his steps. Still get the message `ERROR (dkms apport): binary package for rtl8812au: 5.6.4.2 not found
`

kimocoder, 2020-01-20:
Try compiling, not DKMS and post the result here

michaelwijnands-zuyd, 2020-01-20:
```
sudo make
make ARCH=arm64 CROSS_COMPILE= -C /usr/src/linux-headers-4.9.140-tegra-ubuntu18.04_aarch64/kernel-4.9 M=/home/michael/wifi/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.140-tegra-ubuntu18.04_aarch64/kernel-4.9'
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_cmd.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_security.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_debug.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_io.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_mlme.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_mi.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_vht.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_rf.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_chplan.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_recv.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_ap.o
  CC [M]  /home/michael/wifi/rtl8812au/core/mesh/rtw_mesh.o
  CC [M]  /home/michael/wifi/rtl8812au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/michael/wifi/rtl8812au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_xmit.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_p2p.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_rson.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_tdls.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_br_ext.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_iol.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_sreset.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_btcoex_wifionly.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_btcoex.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_beamforming.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_odm.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_rm.o
  CC [M]  /home/michael/wifi/rtl8812au/core/rtw_rm_fsm.o
  CC [M]  /home/michael/wifi/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /home/michael/wifi/rtl8812au/os_dep/osdep_service.o
  CC [M]  /home/michael/wifi/rtl8812au/os_dep/linux/os_intfs.o
/home/michael/wifi/rtl8812au/os_dep/linux/os_intfs.c: In function ‘netdev_vir_if_close’:
/home/michael/wifi/rtl8812au/os_dep/linux/os_intfs.c:2805:2: error: ‘wdev’ undeclared (first use in this function); did you mean ‘cdev’?
  wdev->iftype = NL80211_IFTYPE_MONITOR;
  ^~~~
  cdev
/home/michael/wifi/rtl8812au/os_dep/linux/os_intfs.c:2805:2: note: each undeclared identifier is reported only once for each function it appears in
/home/michael/wifi/rtl8812au/os_dep/linux/os_intfs.c: At top level:
cc1: warning: unrecognized command line option ‘-Wno-cast-function-type’
scripts/Makefile.build:335: recipe for target '/home/michael/wifi/rtl8812au/os_dep/linux/os_intfs.o' failed
make[2]: *** [/home/michael/wifi/rtl8812au/os_dep/linux/os_intfs.o] Error 1
Makefile:1629: recipe for target '_module_/home/michael/wifi/rtl8812au' failed
make[1]: *** [_module_/home/michael/wifi/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.140-tegra-ubuntu18.04_aarch64/kernel-4.9'
Makefile:2236: recipe for target 'modules' failed
make: *** [modules] Error 2
```

kimocoder, 2020-01-20:
Aaah, fixing this after work today. I'll notify you, thanks!

kimocoder, 2020-01-20:
@michaelwijnands-zuyd 

"git pull" (update) and try to compile/build/install again. Removed the "-Wno-cast-function-type" string

michaelwijnands-zuyd, 2020-01-21:
Thanks! It seemed to have finished making, however it did show a warning somewhere:

```
 CC [M]  /home/michael/wifi/rtl8812au/os_dep/linux/os_intfs.o
In file included from /home/michael/wifi/rtl8812au/include/osdep_service.h:54:0,
                 from /home/michael/wifi/rtl8812au/include/drv_types.h:26,
                 from /home/michael/wifi/rtl8812au/os_dep/linux/os_intfs.c:17:
/home/michael/wifi/rtl8812au/os_dep/linux/os_intfs.c: In function ‘rtw_ndev_notifier_call’:
/home/michael/wifi/rtl8812au/include/osdep_service_linux.h:518:71: warning: ‘ndev’ is used uninitialized in this function [-Wuninitialized]
 efine rtw_netdev_priv(netdev) (((struct rtw_netdev_priv_indicator *)netdev_pri(netdev))->priv)
                                                                     ^~~~~~~~~~
/home/michael/wifi/rtl8812au/os_dep/linux/os_intfs.c:1430:21: note: ‘ndev’ was declared here
  struct net_device *ndev;
                     ^~~~
  CC [M]  /home/michael/wifi/rtl8812au/os_dep/linux/usb_intf.o
```

It finished with:
```
LD [M]  /home/michael/wifi/rtl8812au/88XXau.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/michael/wifi/rtl8812au/88XXau.mod.o
  LD [M]  /home/michael/wifi/rtl8812au/88XXau.ko
make[1]: Leaving directory '/u
```sr/src/linux-headers-4.9.140-tegra-ubuntu18.04_aarch64/kernel-4.9'

After that I did a 'sudo make install' and it resulted with:
```
sudo make install
install -p -m 644 88XXau.ko  /lib/modules/4.9.140-tegra/kernel/drivers/net/wireless/
/sbin/depmod -a 4.9.140-tegra

```

However, the WiFi doesn't seem to be working yet. Is there something else I should do?