aircrack-ng/rtl8812au #539: no wifi Interface shows up

edlovesiraq, 2020-01-21:
Situation, 
tldr;  I've used this driver on kali for a bit with no real issues. I updated to kali 5.4 from 5.3 and my boot was full so i purged old kernals but kept the current one.  I've tried to install using make/make install then restart ... no dice. Then i used the install.dkms script, restarted. also no dice.  I can see it in dmesg and lsusb. Light is on and blinking but it does not show up under iwconfig or ifconfig. 
Any help is appreciated, I'm sure this is my fault, but I'm also sure I'm not the only one who will mess this up. 

---------info
uname -r
5.4.0-kali2-amd64
[13:58:57] root@android:~/rtl8812au# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:b009 Realtek Semiconductor Corp. 802.11n WLAN Adapter
Bus 001 Device 003: ID 0408:5365 Quanta Computer, Inc. HP TrueVision HD Camera
Bus 001 Device 007: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Dmesg ...
[ 4383.494585] usb 1-1: USB disconnect, device number 6
[ 4385.695020] usb 1-1: new high-speed USB device number 7 using xhci_hcd
[ 4385.843452] usb 1-1: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[ 4385.843457] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4385.843460] usb 1-1: Product: 802.11ac NIC
[ 4385.843462] usb 1-1: Manufacturer: Realtek
[ 4385.843464] usb 1-1: SerialNumber: 123456
[ 4395.929006] ------------[ cut here ]--------

kimocoder, 2020-01-21:
I need to add v5.4 support. Doing it in a few hours

edlovesiraq, 2020-01-21:
SWEET! Thank you.

kimocoder, 2020-01-22:
@edlovesiraq 

"apt-get update && apt-get install bc"
"git clone https://github.com/aircrack-ng/rtl8812au -b v5.3.4"
"cd rtl8812au"
"make && make install"

edlovesiraq, 2020-01-22:
Works! Thank ou

closed 2020-01-25 by kimocoder

knnurl, 2022-07-18:
> @edlovesiraq
> 
> "apt-get update && apt-get install bc" "git clone https://github.com/aircrack-ng/rtl8812au -b v5.3.4" "cd rtl8812au" "make && make install"

Thank you a lot!!