aircrack-ng/rtl8812au #540: Heads up: Regression stable 5.3.4, wrong passphrase

Doppel-D, 2020-01-22:
With the latest [commit](https://github.com/aircrack-ng/rtl8812au/commit/737dce3cdde8e8c6daab58bcffe4b87a20473eb0) I get the same problems which I had with 5.6.4.2 and which brought me to 5.3.4:
Wrong passphrase although it is correct.

Reinstalled old version, everything ok.

Linux amilo 4.19.0-0.bpo.6-686-pae #1 SMP Debian 4.19.67-2+deb10u2~bpo9+1 (2019-11-12) i686 GNU/Linux

crispywizard, 2020-01-23:
I have the same problem when using the icon (network manager) to connect.  Connecting via command line manually works like a charm though

I used this process to successfully connect: https://www.blackmoreops.com/2014/09/18/connect-to-wifi-network-from-command-line-in-linux/

Doppel-D, 2020-01-23:
The article also mention that there are possibilities to use (if available) network manager command line tools....thanks for the link!

crispywizard, 2020-01-26:
No problem, it would be wonderful if someone could fix this problem rather than having to use this clunky workaround.

kimocoder, 2020-01-26:
Thanks, I'll prioritize this and do some test. I also need you dmesg output with the error (put it in dmesg.txt and upload it here)

Doppel-D, 2020-01-26:
**WIP - dmesg not attached, have to check on thing**

Thank you for looking into it! Attached dmesg from the client.
- the laptop has a built in intel wifi iwl3945 which I "disabled" in rc.local to remove the scanning results in network manager.
`#!/bin/bash
rmmod iwl3945
rmmod iwlegacy 
exit 0`

Here is the syslog from the router:

```Sun Jan 26 22:17:11 2020 daemon.info hostapd: wlan0-1: STA 58:8b:f3:9a:c1:b4 IEEE 802.11: authenticated
Sun Jan 26 22:17:11 2020 daemon.info hostapd: wlan0-1: STA 58:8b:f3:9a:c1:b4 IEEE 802.11: associated (aid 2)
Sun Jan 26 22:17:12 2020 daemon.notice hostapd: wlan0-1: AP-STA-POSSIBLE-PSK-MISMATCH 58:8b:f3:9a:c1:b4
Sun Jan 26 22:17:13 2020 daemon.notice hostapd: wlan0-1: AP-STA-POSSIBLE-PSK-MISMATCH 58:8b:f3:9a:c1:b4
Sun Jan 26 22:17:14 2020 daemon.notice hostapd: wlan0-1: AP-STA-POSSIBLE-PSK-MISMATCH 58:8b:f3:9a:c1:b4
Sun Jan 26 22:17:20 2020 daemon.info hostapd: wlan0-1: STA 58:8b:f3:9a:c1:b4 IEEE 802.11: deauthenticated due to local deauth request`

Doppel-D, 2020-01-28:
Perhaps it is a good idea to wait a few more days because there will be an update to openwrt 19.07.1 which I am using as a router, just to be on the safe side.

crispywizard, 2020-01-30:
So I'm running with latest commits of version of 5.6.4.2 branch and have this problem.  Same problem under both vmware and virtualbox environments.  Connecting to any AP through NetworkManager icon with known good password or even an open AP constantly prompts for a password and refuses to connect.  Connecting from terminal with the process included above works.  Including dmesg from failed connection attempt.
[dmesg.txt](https://github.com/aircrack-ng/rtl8812au/files/4136426/dmesg.txt)

Doppel-D, 2020-01-31:
@crispywizard  Have you tried stable 5.3.4? I am facing perodic pause of transmission....

crispywizard, 2020-02-01:
I tried 5.3.4.  I still can't connect with NetworkManager icon.  When connecting through terminal,  I don't get dmesg errors but wpa_supplicant shows info below.  MYBSSID and SSID have been replaced.  If more info required, let me know.  Kernel: Linux Kali 5.4.0-kali3-amd64 #1 SMP Debian 5.4.13-1kali1 (2020-01-20) x86_64 GNU/Linux

wlan0: CTRL-EVENT-DISCONNECTED bssid=<MY BSSID> reason=0 locally_generated=1
wlan0: CTRL-EVENT-ASSOC-REJECT status_code=1
wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: Trying to associate with <MY BSSID> (SSID='MYSSID' freq=5785 MHz)
FT: Invalid key management type (2)
wlan0: Associated with <MY BSSID>
wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=US
wlan0: WPA: Key negotiation completed with <MY BSSID> [PTK=CCMP GTK=TKIP]
wlan0: CTRL-EVENT-CONNECTED - Connection to <MY BSSID> completed [id=0 id_str=]
wlan0: CTRL-EVENT-DISCONNECTED bssid=<MY BSSID> reason=0 locally_generated=1
wlan0: CTRL-EVENT-ASSOC-REJECT status_code=1
wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: Trying to associate with <MY BSSID> (SSID='MYSSID' freq=5785 MHz)
FT: Invalid key management type (2)
wlan0: Associated with <MY BSSID>
wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=US
wlan0: WPA: Key negotiation completed with <MY BSSID> [PTK=CCMP GTK=TKIP]
wlan0: CTRL-EVENT-CONNECTED - Connection to <MY BSSID> completed [id=0 id_str=]

This loop continues...

kimocoder, 2020-03-08:
Try this someone ..

```
git clone https://github.com/aircrack-ng/rtl8812au -b v5.3.4
cd rtl8812au/os_dep/linux/
rm -rf wifi_regd.c
wget http://167.71.7.191/wifi_regd.c
cd ../../ && make -j4
```

kimocoder, 2020-03-08:
Also, I found to issues related, which is interesting..

https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/1748839

https://www.reddit.com/r/linuxquestions/comments/b1b8jo/comment/eitv4oh

https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/1761003

Doppel-D, 2020-03-08:
Thanks for your investigation! Tried your suggestion with ./dkms-install.sh....still no connection possible.

Doppel-D, 2020-03-08:
IIRC I tried it on ubuntu 18.04 LTS and it was working fine. So while a new version of my distro is around the corner I will test and probably it is already fixed. Until then I stick with an old backuped version of 5.3.4

kimocoder, 2020-03-08:
Leave it open, I'll investigate this one right away

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)

Doppel-D, 2022-07-21:
Tried with Linux amilo 4.19.0-21-686-pae #1 SMP Debian 4.19.249-2 (2022-06-30) i686 GNU/Linux which worked.

closed 2022-07-21 by Doppel-D

Doppel-D, 2022-07-21:
Only worked once ;- ) Had to revert to v5.1.5