aircrack-ng/rtl8812au #541: Ubuntu 18.04 update has stopped driver form working?

bradleysmith, 2020-01-23:
my wlan1 (an ALFA AWUS036ACH adapter) is no longer present. I believe this happened after an Ubuntu update, possibly including a kernel upgrade. I'll try and pre-empt the info needed to help me:

Kernel version is now 5.3.0-26-generic

Output from lsusb includes this line:
Bus 001 Device 007: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter

output from dmesg | grep 8812 after plugging the device in from a cold boot is:
[   75.400475] usb 1-2: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00

Output from iwconfig though is:
eth0      no wireless extensions.

wwan0     no wireless extensions.

lo        no wireless extensions.

wlan0     IEEE 802.11  ESSID:"VisitorWLAN"  
          Mode:Managed  Frequency:5.3 GHz  Access Point: D8:54:A2:5C:10:A5   
          Bit Rate=400 Mb/s   Tx-Power=22 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=57/70  Signal level=-53 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:143   Missed beacon:0

This seems to show my ethernet, a 3G card (i have a Thinkpad T470s), dunno what lo is, and the built in Wifi on wlan0. So, no wlan1.

I removed what I could find of the old drivers, git cloned latest (10am GMT 23rd Jan 2020), did a make and then a sudo make install (I had to use sudo it seems as it complained about not having access permissions without it).

Any suggestions?
Many thanks - Brad

bradleysmith, 2020-01-23:
Solved this myself. A look at the dmesg output just past the 8812 detection showed this:

75.400480] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   75.400483] usb 1-2: Product: 802.11n NIC
[   75.400486] usb 1-2: Manufacturer: Realtek
[   75.400489] usb 1-2: SerialNumber: 123456
**[   75.940698] Lockdown: systemd-udevd: Loading of unsigned module is restricted; see man kernel_lockdown.7**

Disabling secure boot on my Thinkpad has allowed the driver to load once more and wlan1 is now showing up.

I'm not sure what changed on my machine to cause this problem to occur though and would like to understand what happened but from my perspective at least I am up and running again.

kimocoder, 2020-01-23:
Newer Ubuntu kernels needs modules to be signed in order to work. Did write about this not long back, and also added a script for turning off secure boot in the 'tools' folder. G rl

Lexus89, 2020-01-30:
@kimocoder I have the same issue however when running the script in a VM i get the following error: "EFI variables are not supported on this system"

5.3.0-28-generic #30~18.04.1-Ubuntu

dmesg:
[  965.032174] 88XXau: loading out-of-tree module taints kernel.
[  965.033232] 88XXau: module verification failed: signature and/or required key missing - tainting kernel

airmon-ng error (after running check kill):
Error setting channel: command failed: Device or resource busy (-16)
Error -16 likely means your card was set back to station mode by something.

Haffi112, 2020-02-04:
> Newer Ubuntu kernels needs modules to be signed in order to work. Did write about this not long back, and also added a script for turning off secure boot in the 'tools' folder. Great you got it worki

@kimocoder, can you tell me where you wrote about modules needing to be signed? (I cannot find the instructions).

I have a similar issue. I recently upgraded Ubuntu and the module stopped working. Been trying to update without success and dmesg complains about modules needing to be signed. I'm at work now but can provide more info when I get back home if it turns out to be necessary.

Haffi112, 2020-02-04:
I tried using the script for secure boot in the tools folder but it is stuck on the command

    /sbin/depmod -a 5.3.0-28-generic

Is that normal? I've been waiting for more than 10 minutes now with no change/feedback.

kimocoder, 2020-02-04:
@Lexus89 airmon-ng start has to be run two times. I'll fix it soon.

closed 2020-03-08 by kimocoder