aircrack-ng/rtl8812au #542: Build fails on CentOS 8 - Kernel 4.18

TruckerZer0, 2020-01-23:
Hi all! I had no luck building the driver on CentOS 8.1 1911 with kernel 4.18.0-147.3.1.el8_1, the compilation ends with this error:

/home/0perator/Scaricati/rtl8812au-5.6.4.2/os_dep/linux/os_intfs.c:1516:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, void *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, void *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/home/0perator/Scaricati/rtl8812au-5.6.4.2/os_dep/linux/os_intfs.c:1516:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
In file included from /home/0perator/Scaricati/rtl8812au-5.6.4.2/include/osdep_service.h:54,
                 from /home/0perator/Scaricati/rtl8812au-5.6.4.2/include/drv_types.h:26,
                 from /home/0perator/Scaricati/rtl8812au-5.6.4.2/os_dep/linux/os_intfs.c:17:
/home/0perator/Scaricati/rtl8812au-5.6.4.2/os_dep/linux/os_intfs.c: In function ‘rtw_ndev_notifier_call’:
/home/0perator/Scaricati/rtl8812au-5.6.4.2/include/osdep_service_linux.h:518:71: warning: ‘ndev’ is used uninitialized in this function [-Wuninitialized]
 efine rtw_netdev_priv(netdev) (((struct rtw_netdev_priv_indicator *)netdev_priv(netdev))->priv)
                                                                     ^~~~~~~~~~~

/home/0perator/Scaricati/rtl8812au-5.6.4.2/os_dep/linux/os_intfs.c:1430:21: note: ‘ndev’ was declared here
  struct net_device *ndev;
                     ^~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:313: /home/0perator/Scaricati/rtl8812au-5.6.4.2/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [Makefile:1547: _module_/home/0perator/Scaricati/rtl8812au-5.6.4.2] Error 2
make[1]: uscita dalla directory "/usr/src/kernels/4.18.0-147.3.1.el8_1.x86_64"
make: *** [Makefile:2232: modules] Error 2
[0perator@JerryLee rtl8812au-5.6.4.2]$ 

Gcc version used: 8.3.1 20190507 (Red Hat 8.3.1-4)
Driver branches tested: 5.1.5/5.2.9/5.2.20/5.3.4/5.6.4.2 (All no luck and with the same error)

Thank you in advance for any help, and sorry for my poor english!

kimocoder, 2020-01-25:
I'll add some patches related to _select_queue for you tomorrow. I'll leave a note when they are pushed.

TruckerZer0, 2020-01-26:
Thanks a lot, i really appreciate it!

shauavik, 2020-02-18:
I am also encountering the same issue with rtl8812au. 
Kindly help

TruckerZer0, 2020-02-18:
Well, since nothing happened until now here i think i've found a (sort of) solution, but it has it's flaws: I recently discovered the existence of ELRepo LTS and mainline kernel repo for Enterprise Linux, and which now also supports CentOS 8.x; So what i've done is simply install latest mainline kernel and headers from this repo and voilà, driver compiled via dkms and so far everything works smooth as silk; I now this isn't a true solution, especially if you didn't want to risk loosing system stability by changing official system packages, but as i said since there isn't better solution around for the moment, and that my isn't a production machine, installing unofficial packages  isn't a problem at all for me, at least now i can use my AWUS1900...

If can interest you this is ELRepo  kernel  section website: http://elrepo.org/tiki/tiki-index.php

shauavik, 2020-02-21:
> @shauavik Well, since nothing has happened here for the moment i think i've found a (sort of) solution, but it has it's flaws: I recently discovered the existence of ELRepo LTS and mainline kernel repo for Enterprise Linux, and which now also supports CentOS 8.x; So what i've done is simply install latest mainline kernel and headers from this repo and voilà, driver compiled via dkms and so far everything works smooth as silk; I now this isn't a true solution, especially if you didn't want to risk loosing system stability by changing official system packages, but as i said since there isn't better solution around for the moment, and that my isn't a production machine, installing unofficial packages isn't a problem at all for me, at least now i can use my AWUS1900...
> 
> If can interest you this is ELRepo's kernel section website: http://elrepo.org/tiki/tiki-index.php

@TruckerZer0 Thanks a lot. Will try this and update. Appreciate your help

blackhole0064, 2020-02-29:
I too encountered, and resolved, this particular problem using CentOS 8.1 1911 and Kernel 4.18.0-147. Please the following in mind about the 'fix' I am presenting (until the official patch is pushed):

- I only plan on utilizing the interface which will be using this module for basic network connectivity, not the more advanced features which are provided by the aircrack project

- I only just implemented this 'fix'. I have confirmed basic functionality for the interface using the module. I don't know what, if any, issues will arise long term.

**The 'Fix'**
I edited './os_dep/linux/os_intfs.c' removing lines 1369, 1371, 1372 and 1373 (shown below) and then followed the standard build process using make

1367 static u16 rtw_select_queue(struct net_device *dev, struct sk_buff *skb
1368 #if LINUX_VERSION_CODE >= KERNEL_VERSION(3, 13, 0)
1369         #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 19, 0)
1370         , struct net_device *sb_dev
1371         #else
1372         , void *accel_priv
1373         #endif

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)

fs30000, 2023-01-13:
> fyi it fails to build on Rocky Linux 8.5, but there is a hack from https://quintrokk.subness.net/?p=2057
> 
> [uglyhack_for_rocky85.patch.gz](https://github.com/aircrack-ng/rtl8812au/files/7814742/uglyhack_for_rocky85.patch.gz)

Server down :/