aircrack-ng/rtl8812au #544: Fails to compile on 5.4.14

Freethefire, 2020-01-26:
I know this is a new kernel but I was hoping that it would work. I'm guessing it's not a simple fix?

`media@media:~/rtl8812au-5.6.4.2$ sudo ./dkms-install.sh 
[sudo] password for media: 
About to run dkms install steps...
Error! DKMS tree already contains: rtl8812au-5.6.4.2
You cannot add the same module/version combo more than once.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.4.14-050414-generic KSRC=/lib/modules/5.4.14-050414-generic/build...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812au: 5.6.4.2 not found
Error! Bad return status for module build on kernel: 5.4.14-050414-generic (x86_64)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.4.14-050414-generic KSRC=/lib/modules/5.4.14-050414-generic/build...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812au: 5.6.4.2 not found
Error! Bad return status for module build on kernel: 5.4.14-050414-generic (x86_64)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.
Finished running dkms install steps`

[make.log](https://github.com/aircrack-ng/rtl8812au/files/4113259/make.log)

Does anyone know which the latest supported(or working) kernel is?
Thanks!

kimocoder, 2020-01-26:
I use it on v5.4, added support for some days ago. Try compiling with 'make' instead of DKMS, let me see that log

Freethefire, 2020-01-26:
Thanks so much for reply!

[make.log](https://github.com/aircrack-ng/rtl8812au/files/4113316/make.log)
Should have attached this as well before, sorry about that.

kimocoder, 2020-01-26:
what the hell is going on with your compiler over there? That output is more a compiler related issue, rather than a driver bug. "gcc -v" output too please, and what distro is this?

Freethefire, 2020-01-26:
Thanks again!

media@media:~/rtl8812au-5.6.4.2$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.4.0-1ubuntu1~18.04.1' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1) 

I'm on Ubuntu Mate 18.04.

kimocoder, 2020-01-26:
Ok. Confirmed. This is more or less an unresolvable kernel/compiler issue that inflicted on gcc6 and gcc7, use gcc8 or gcc9 instead.

Freethefire, 2020-01-26:
Thanks a bunch! Now it's working :)
(and thanks for your patience)

closed 2020-01-26 by Freethefire

kimocoder, 2020-01-26:
No problem, that's why I'm here! Have a nice day :1st_place_medal: