aircrack-ng/rtl8812au #545: 8812au support for 5Mhz bandwidths (quarter-clocking/quarter-rate)

Shachafinho, 2020-01-26:
Hello,

I have noticed that the 8812au chip should support 5Mhz bandwidth transmissions as part of its PHY features.
In contrast, I have also noticed that the latest version doesn't support this feature in the SOFTWARE layer.
Am I doing something wrong (the support exists but I have failed to use it correctly), or is there no support yet?
If it is the latter, do you intend to support 5/10 Mhz bandwidths?

Thanks,
Shachaf

kimocoder, 2020-01-26:
That should be Realtek's work, but I know where the code is/should be placed so I may take a look at it when I got more time again

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)

closed 2021-05-12 by Shachafinho

planewave, 2022-08-16:
> pls consider closing the issue, when it's solved by now :)

could you please tell me how to enable the 5/10 MHz channel width feature? thanks