aircrack-ng/rtl8812au #547: "airmon-ng start" Causes System To Hang With No Obvious Logs Or Errors

bonedaddy, 2020-01-27:
Using a new `AC1200` in combination with this driver, and `airmon-ng start` causes my system to hang with no obvious logs or errors. Despite running `sudo airmon-ng check kill` immediately before running `airmon-ng start`, and not having any active network processes, I occasionally get a message like ` please run "rfkill unblock 2" to use this interface.` Unfortunately the moment I run `airmon-ng start` attempting to launch another terminal process to run this command fails, so I'll have to have another terminal window open (note, I'm also unable to create a new terminal tab as this process hangs), and running `sudo rkill unblock 2` does not help.

# System Specs

`/etc/lsb_release`
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=19.10
DISTRIB_CODENAME=eoan
DISTRIB_DESCRIPTION="Pop!_OS 19.10"
```

`uname -a`
```
Linux dark 5.3.0-7625-generic #27~1576774560~19.10~f432cd8-Ubuntu SMP Thu Dec 19 20:35:37 UTC  x86_64 x86_64 x86_64 GNU/Linux
```

# Logs

`airmon-ng start` logs:

```
$ sudo airmon-ng start wlx00c0caaa5340

PHY     Interface       Driver          Chipset

phy0    wlp2s0          iwlwifi         Intel Corporation Wireless 8265 / 8275 (rev 78)
phy1    wlx00c0caaa5340 88XXau          Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
        wlx00c0caaa5340 is soft blocked, please run "rfkill unblock 2" to use this interface.
rfkill error, unable to start wlx00c0caaa5340

Would you like to try and automatically resolve this? [y/n] y
Interface wlx00c0caaa5340mon is too long for linux so it will be renamed to the old style (wlan#) name.

                (mac80211 monitor mode vif enabled on [phy1]wlan0mon
````

syslog logs:

```
[   33.872307] rfkill: input handler disabled
[   42.755879] iwlwifi 0000:02:00.0: Applying debug destination EXTERNAL_DRAM
[   42.888488] iwlwifi 0000:02:00.0: Applying debug destination EXTERNAL_DRAM
[   42.953703] iwlwifi 0000:02:00.0: FW already configured (0) - re-configuring
[   42.972816] iwlwifi 0000:02:00.0: BIOS contains WGDS but no WRDS
[   43.383341] [phydm_la_set_buff_mode] Warning!
[   47.026473] wlp2s0: authenticate with ...
[   47.035802] wlp2s0: send auth to... (try 1/3)
[   47.041249] wlp2s0: authenticated
[   47.049083] wlp2s0: associate with... (try 1/3)
[   47.051994] wlp2s0: RX AssocResp from... (capab=0x1511 status=0 aid=77)
[   47.053985] wlp2s0: associated
[   47.056319] wlp2s0: Limiting TX power to 30 (30 - 0) dBm as advertised by ...
[   47.214754] IPv6: ADDRCONF(NETDEV_CHANGE): wlp2s0: link becomes ready
[   47.894139] IPv6: ADDRCONF(NETDEV_CHANGE): wlx00c0caaa5340: link becomes ready
[  123.597685] wlp2s0: deauthenticating from ... by local choice (Reason: 3=DEAUTH_LEAVING)
```

bonedaddy, 2020-01-28:
Small update, I think this might be a bug within the driver when it comes to handling very long names? I used `ifconfig` and `iplink` to rename the `wlx00c0caaa5340mon` interface to `wlan1` and was able to get around this error.

retrohacking, 2022-01-06:
would you please share che commands you used to do it? I've tried with 
ip link set intname down
ip link set intname name wlan1
ip link set wlan1 up 
and it didn't work. I have the same drivers as yours.

uSaiPrashanth, 2022-01-08:
Any updates on how to fix this? I'm having a similar error. But long names might not be a reason as I'm having wlo1 as my interface name

haxor-dot, 2022-01-20:
Experiencing same issue here. Exiting the `airmon-ng` terminal window results unusual behaviour for the whole system. Like wont show result for commands like `iwconfig` and etc.

uSaiPrashanth, 2022-01-24:
In my case, my firmware supported monitor mode (but the physical card didn't). That led to airmon-ng being stuck

paragon21, 2022-01-26:
Hello, I'm having same problem. I'm notited that run `airmon-ng start` cause uninterruptible sleep of processes.
```
root        4688  0.0  0.0   3064   940 pts/0    D+   20:48   0:00 iw phy phy2 interface add wlan0mon type monitor
root        4689  0.0  0.0  24128  5140 ?        D    20:48   0:00 /lib/systemd/systemd-udevd
```
```
4580  21:17:32 openat(AT_FDCWD, "/sys/class/ieee80211/phy1/index", O_RDONLY) = 4 <0.000061>
4580  21:17:32 read(4, "1\n", 199)      = 2 <0.000016>
4580  21:17:32 close(4)                 = 0 <0.000013>
4580  21:17:32 sendmsg(3, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=00000000}, msg_namelen=12, msg_iov=[{iov_base={{len=52, type=nl80211, flags=NLM_F_REQUEST|NLM_F_ACK, seq=1642961853, pid=1488982500}, "\x07\x00\x00\x00\x08\x00\x01\x00\x01\x00\x00\x00\x0d\x00\x04\x00\x77\x6c\x61\x6e\x30\x6d\x6f\x6e\x00\x00\x00\x00\x08\x00\x05\x00"...}, iov_len=52}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0 <unfinished ...>
4486  21:18:26 <... ppoll resumed>)     = ? ERESTARTNOHAND (To be restarted if no handler) <55.166276>
4489  21:18:26 <... read resumed>0x7ffd413f9210, 128) = ? ERESTARTSYS (To be restarted if SA_RESTART is set) <54.370147>
4486  21:18:26 --- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
4489  21:18:26 --- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
4486  21:18:26 ppoll([{fd=-1}, {fd=6, events=POLLIN}], 2, NULL, NULL, 8 <unfinished ...>
4489  21:18:26 read(3, 
```

erentar, 2022-03-07:
Same problem on Fedora 35

tester009, 2022-04-03:
The same problem on Ununtu 20.04 with TP-Link Archer T2U Plus AC600 

8812au, 5.6.4.2_35491.20191025, 5.13.0-39-generic, x86_64: installed
Bus 001 Device 002: ID 2357:0120 TP-Link 802.11ac WLAN Adapter

"airmon-ng start wlan0" hangs the system.

MatthewTingum, 2022-04-26:
Same problem `NetGear, Inc. A7000`.

Shortening the interface name didn't fix the issue.
Perhaps 2 separate issues if we can confirm the long name issue?

Is there a pragmatic way to determine if the interface don't support monitor mode?
I'd like to see more accurate error messages.

At the end of the day, this bug hangs commands like `ifconfig`.
It's maybe locking some resource and trying indefinitely?
We can split issues, but there should be a timeout / attempt limit on this.

nemanjan00, 2022-05-10:
Same problem, Alfa AWUS036ACH