aircrack-ng/rtl8812au #548: Monitor mode injection in 5.6.4.2

j-forristal, 2020-01-29:
On Nov 13, there was a commit that added the typical injection changes (particularly support for TX radiotap header parsing) ... but then right after it there was a commit to back out those changes.
https://github.com/aircrack-ng/rtl8812au/commit/506f1e3c11cc07f34a9e3acde6c358fb5bb4bda4
https://github.com/aircrack-ng/rtl8812au/commit/0c23551874afbf980a55a48e145045b223db1b5d

As of now, rtw_monitor_xmit_entry just skips past the radiotap header, using fixed/hardcoded assumptions on transmit rate and retries configuration.
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/core/rtw_xmit.c#L4437

@kimocoder Is this intentional?  Before I wind up patching the older TX radiotap support back in, I want to check if this was intentionally deprecated or there is a new direction things are heading.

Due to this missing support, my use case implementation hits two notable bugs:

- Even though rtw_monitor_xmit_entry sets retry_ctrl=FALSE, the HALs basically treat this as send 12 retries (instead of normal 6 when retry_ctrl=TRUE); basically injected packets are massively retried instead of older ability to fully turn off retries for injected packets.  My injected packets are now swamping the airwaves with retries. That actually seems like a bug in the RTL base code, but I may be missing something/intention of retry_ctrl.
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/core/rtw_xmit.c#L4474
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/hal/rtl8812a/usb/rtl8812au_xmit.c#L307

- When injecting data packets, without rate support, current code rtw_monitor_xmit_entry code (via update_monitor_frame_attrib) assumes VHT for any data TX.  This makes those data packet injections blind to any non-VHT devices.
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/core/rtw_xmit.c#L4463
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/core/rtw_mlme_ext.c#L7550

j-forristal, 2020-02-03:
Some latest commits to 5.6.4.2 branch address the issue.  All set, thanks! closing....

closed 2020-02-03 by j-forristal

mapdog-mapdog, 2021-12-29:
Hello,林，i am mapdog,i am not use driver,beacuse i hope use chip 8812cu replace 8812au,but the stock driver of 8812cu not supprt fame injecting and i want to modify the stock driver to support fame injecting
i konw when frame injecting, rtw_monitor_xmit_entry will be call, but in stock driver it can't be call, so could you tell me where to be modify  to support to call rtw_monitor_xmit_entry and to support frame injecting

by the way, we are chinese, can't you understand my english description