aircrack-ng/rtl8812au #550: AWUS036ACH Diff between built and installed module!

Abraxas333, 2020-01-31:
I want to get the above mentioned wireless adapter to work in Kali Linux 2019.4 but so far I am hitting a wall despite attempting for about two days. 

I started out following the instructions in that video: youtube.com/watch?v=yTINu4UQz4Y&t=6s
I downloaded the driver at https://github.com/aircrack-ng/rtl8812au.git, unzipped it in Downloads,
cd Downloads 
ls 
cd rtl8812au-5.6.4.2/
ls
dkms-install.sh
systemctl restart networking 
systemctl restart network-manager

In the vid it works but for me it didn't, so I tried:
apt purge realtek-rtl88xxau-dkms
cd rtl8812au
./dkms-remove.sh
make uninstall

And next:
apt update && apt dist-upgrade
apt install build-essential bc
git clone https://github.com/kimocoder/rtl8812au.git
cd rtl8812au
make
make install
modprobe 8812au

But it didn't have the desired effect and I am actually suspecting that I made it worse, because when I run:
dkms status 
I get:
realtek-rtl88xxau, 5.3.4~20200115, 5.3.0-kali2-amd64, x86_64: installed (WARNING! Diff between built and installed module!)
realtek-rtl88xxau, 5.3.4~20200115, 5.4.0-kali3-amd64, x86_64: installed (WARNING! Diff between built and installed module!)
rtl8812au, 5.3.4, 5.3.0-kali2-amd64, x86_64: installed (original_module exists) (WARNING! Diff between built and installed module!)
rtl8812au, 5.3.4, 5.4.0-kali3-amd64, x86_64: built (original_module exists)
rtl8812au, 5.6.4.2, 5.3.0-kali2-amd64, x86_64: built (original_module exists)
rtl8812au, 5.6.4.2, 5.4.0-kali3-amd64, x86_64: installed (original_module exists)

I believe this means that I installed the wrong driver and the first one was already correct even though it was not working. 
lsusb reads out:
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

I am new to this as you already figured, I know it will take a lot of work and time to make progress, but can somebody give me input what I should try next?

Guitar-Guy, 2020-01-31:
im having a similar problem trying to get my adapter to work with aircrack-ng running the off-sec ARM image for my raspi 4b

Guitar-Guy, 2020-01-31:
ive been trying to figure this out for hours and its now 5am...it keeps saying it cant locate the kernel headers but they're installed...im getting some sleep now but i'll try and paste my log tomorrow around 1pm GMT

Guitar-Guy, 2020-01-31:
./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.6.4.2/source ->
                 /usr/src/rtl8812au-5.6.4.2

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.19.93-Re4son-v8l+ cannot be found at
/lib/modules/4.19.93-Re4son-v8l+/build or /lib/modules/4.19.93-Re4son-v8l+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Error! echo
Your kernel headers for kernel 4.19.93-Re4son-v8l+ cannot be found at
/lib/modules/4.19.93-Re4son-v8l+/build or /lib/modules/4.19.93-Re4son-v8l+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Finished running dkms install steps.

crispywizard, 2020-01-31:
Does this help..

sudo apt-get update
sudo apt-get upgrade
sudo apt-get install raspberrypi-kernel-headers

Guitar-Guy, 2020-02-01:
> Does this help..
> 
> sudo apt-get update
> sudo apt-get upgrade
> sudo apt-get install raspberrypi-kernel-headers

i tried that and it says raspberrypi-kernal-headers has been replaced by kalipi-headers and when i tried to install them it says they're already installed

Error0x00000001, 2020-02-01:
1) dkms status
2) sudo apt-get  purge realtek-rtl88xxau-dkms
3) sudo git clone https://github.com/aircrack-ng/rtl8812au.git
4) sudo cd rtl8812au
5) ./dkms-remove.sh
6) sudo reboot

After Reboot

1) sudo dkms status
2) sudo apt-get install build-essential && bc && libelf-dev && linux-headers-`uname -r`
3) sudo reboot

After Reboot

1) sudo cd rtl8812au
2) sudo ./dkms-install.sh
3) sudo reboot

After Reboot
1) sudo dkms status
2) sudo iwconfig

Guitar-Guy, 2020-02-03:
>     1. sudo apt-get install dkms && build-essential && bc && libelf-dev && linux-headers-`uname -r`
> 
>     2. dkms status
> 
>     3. sudo apt-get  purge realtek-rtl88xxau-dkms
> 
>     4. sudo git clone https://github.com/aircrack-ng/rtl8812au.git
> 
>     5. sudo cd rtl8812au
> 
>     6. ./dkms-remove.sh
> 
>     7. sudo reboot
> 
> 
> After Reboot
> 
>     1. sudo dkms status
> 
>     2. sudo cd rtl8812au
> 
>     3. sudo ./dkms-install.sh
> 
>     4. sudo reboot
> 
> 
> After Reboot
> 
>     1. sudo dkms status
> 
>     2. sudo iwconfig

still not working on raspberry pi 4 using kali 1.2020  Kernel: Linux 4.19.93-Re4son-v8l+
      Architecture: arm64 !!

root@kali:~# sudo apt-get install dkms && build-essential && bc && libelf-dev && linux-headers-uname -r
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Suggested packages:
  menu
The following NEW packages will be installed:
  dkms
0 upgraded, 1 newly installed, 0 to remove and 214 not upgraded.
Need to get 76.2 kB of archives.
After this operation, 294 kB of additional disk space will be used.
Get:1 http://ftp.hands.com/kali kali-rolling/main arm64 dkms all 2.8.1-5~kali1 [76.2 kB]
Fetched 76.2 kB in 0s (183 kB/s)
Selecting previously unselected package dkms.
(Reading database ... 241098 files and directories currently installed.)
Preparing to unpack .../dkms_2.8.1-5~kali1_all.deb ...
Unpacking dkms (2.8.1-5~kali1) ...
Setting up dkms (2.8.1-5~kali1) ...
Processing triggers for man-db (2.9.0-2) ...
Processing triggers for kali-menu (2020.1.7) ...
bash: build-essential: command not found
root@kali:~# sudo apt-get install dkms build-essential bc libelf-dev linux-headers-uname -r
E: Command line option 'r' [from -r] is not understood in combination with the other options.
root@kali:~# sudo apt-get install dkms build-essential bc libelf-dev linux-headers-uname 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-uname
root@kali:~# sudo apt-get install dkms build-essential bc libelf-dev linux-headers-uname -r
E: Command line option 'r' [from -r] is not understood in combination with the other options.
root@kali:~# sudo apt-get install dkms build-essential bc libelf-dev linux-headers- uname -r
E: Command line option 'r' [from -r] is not understood in combination with the other options.
root@kali:~# sudo apt-get install dkms build-essential bc libelf-dev linux-headers uname -r
E: Command line option 'r' [from -r] is not understood in combination with the other options.
root@kali:~# sudo apt-get install dkms build-essential bc libelf-dev linux-headers uname 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'kalipi-kernel-headers' instead of 'linux-headers'
E: Unable to locate package uname
root@kali:~# dkms status
root@kali:~# sudo apt-get install dkms build-essential bc libelf-dev kalipi-kernal-headers
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package kalipi-kernal-headers
root@kali:~# sudo apt-get update
Hit:1 http://ftp.hands.com/kali kali-rolling InRelease  
Hit:2 http://http.re4son-kernel.com/re4son kali-pi InRelease
Reading package lists... Done
root@kali:~# sudo apt-get install dkms build-essential bc libelf-dev kalipi-kernal-headers
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package kalipi-kernal-headers
root@kali:~# sudo apt-get install dkms build-essential bc libelf-dev kalipi-headers
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package kalipi-headers
root@kali:~# sudo apt-get purge realtek-rtl88xxau-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'realtek-rtl88xxau-dkms' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 214 not upgraded.
root@kali:~# sudo git clone https://github.com/aircrack-ng/rtl8812au.git
Cloning into 'rtl8812au'...
remote: Enumerating objects: 118, done.
remote: Counting objects: 100% (118/118), done.
remote: Compressing objects: 100% (78/78), done.
remote: Total 9670 (delta 57), reused 78 (delta 40), pack-reused 9552
Receiving objects: 100% (9670/9670), 69.20 MiB | 2.71 MiB/s, done.
Resolving deltas: 100% (6635/6635), done.
root@kali:~# sudo cd rtl8812au
sudo: cd: command not found
root@kali:~# sudo cd rtl8812au
sudo: cd: command not found
root@kali:~# cd rtl8812au
root@kali:~/rtl8812au# ./dkms-remove.sh
About to run dkms removal steps...
Error! There are no instances of module: rtl8812au
5.6.4.2 located in the DKMS tree.
Finished running dkms removal steps.
root@kali:~/rtl8812au# apt-get install dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
dkms is already the newest version (2.8.1-5~kali1).
0 upgraded, 0 newly installed, 0 to remove and 214 not upgraded.
root@kali:~/rtl8812au# dkms status
root@kali:~/rtl8812au# sudo dkms status
root@kali:~/rtl8812au# ./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.6.4.2/source ->
                 /usr/src/rtl8812au-5.6.4.2

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.19.93-Re4son-v8l+ cannot be found at
/lib/modules/4.19.93-Re4son-v8l+/build or /lib/modules/4.19.93-Re4son-v8l+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Error! echo
Your kernel headers for kernel 4.19.93-Re4son-v8l+ cannot be found at
/lib/modules/4.19.93-Re4son-v8l+/build or /lib/modules/4.19.93-Re4son-v8l+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Finished running dkms install steps.
root@kali:~/rtl8812au#

justinvforvendetta, 2020-02-03:
you cant just && a package name without telling aptitude you need it.
`sudo apt-get install dkms && build-essential && bc && libelf-dev && linux-headers-`uname -r``

should look more like this:
`sudo apt-get install dkms build-essential bc libelf-dev && next command goes here`

justinvforvendetta, 2020-02-03:
i saw you opened a new issue, i responded there with what should be your next step.

justinvforvendetta, 2020-02-06:
@Abraxas333 can we close this?

Abraxas333, 2020-02-06:
You can close this, I solved the problem by reinstalling virtualbox.

closed 2020-02-09 by Abraxas333