aircrack-ng/rtl8812au #553: Cant install driver on kali 2020.1 image for arm64 ...cant find kalipi-headers

Guitar-Guy, 2020-02-03:
im trying to get this to work with my pi running the official arm64 image from offensive-security. i've posted in https://github.com/aircrack-ng/rtl8812au/issues/550 but the solution given was to no avail and i posted my log there...so therefore opening a new issue! im running kernel version: Linux 4.19.93-Re4son-v8l+ 

please can you enlighten me as to why this isnt working and find a fix! i need packet injection for work!
 Many thanks 
Guitar-Guy

justinvforvendetta, 2020-02-03:
from your log, your kernel headers for kernel 4.19.93-Re4son-v8l+ cannot be found at
/lib/modules/4.19.93-Re4son-v8l+/build or /lib/modules/4.19.93-Re4son-v8l+/source."

find your kernel headers, and use --kernelsourcedir to tell DKMS where they are.

justinvforvendetta, 2020-02-03:
(raspberrypi-kernel-headers should be located in /lib/modules/(kernel version)/build)

justinvforvendetta, 2020-02-06:
@Guitar-Guy any luck? can we close this yet?

kimocoder, 2020-03-08:
The kalipi-headers package has nothing to do with an issue in the driver.

closed 2020-03-08 by kimocoder