aircrack-ng/rtl8812au #557: 5.8.5 driver & VHT support

powderluv, 2020-02-10:
Any plans on porting over to the latest release 5.8.5 ? https://github.com/coolshou/rtl8814au

I have been trying to debug some strange Wifi freeze with the current 5.6.4 branch and I have the 5.8.5 building on Gentoo RPI64 but dont want to rebase your patches if you are doing it anyway. 

Also I noticed you have disabled VHT any chance to re-enable ? Your driver seems to be the most maintained compared to the 100s of forks so would rather get all my requests to you :)  Happy to send PRs if you are open to it.

fariouche, 2020-03-01:
A up to date 8814 driver! Good find!
The current support of 8814 is based on a very old driver I've ported.
I will check if this one is good enough to bother porting the other chipsets

satmandu, 2020-03-02:
VHT requires FPU support, which is missing by default on arm64 kernels. I'd love to see someone convert the VHT code to not need the float support so it works on arm64! :)

fariouche, 2020-03-02:
Quick test, sounds good enough to me to bother porting the other devices and do some more tests.
At least, no more spurious disconnections and packet loss!

bjlockie, 2020-03-03:
> VHT requires FPU support, which is missing by default on arm64 kernels. I'd love to see someone convert the VHT code to not need the float support so it works on arm64! :)

It won't work on a Raspberry Pi4?

powderluv, 2020-03-04:
I was able to build it on my Gentoo rpi4 64 bit system. No need to worry about hardfloat in arm64 aarch64 since the FPU and neon units are always there

kimocoder, 2020-03-04:
So, we actually got a new 8814au AND 8821au.. damn, this is awesome. I'll prepare a new branch @fariouche so if it gets f*** we got the v5.6.4.2 as backup :+1:

thesamesam, 2020-03-04:
(Just wanted to say, a well-maintained consolidated repo for these drivers is _so_ appreciated!)

bjlockie, 2020-03-04:
> I was able to build it on my Gentoo rpi4 64 bit system. No need to worry about hardfloat in arm64 aarch64 since the FPU and neon units are always there

Can you use it with 80211AC though?

powderluv, 2020-03-04:
No. It would register and show as .ac but authentication will always fail. I tried the latest hostapd and also ported the shitty rtl fork of hostapd but no luck so far.

fariouche, 2020-03-04:
So far, the merge is almost finished on my side.
Still have 2 duplicated functions to deal with, and hopefully I will be able to quick test the 8812au with this new base.
They have rework a lot around the vht functions.
This may be good news for the 8814 and maybe the 8812 too if this does not involve the firmware.

We'll need to Port again all the fixes and enhancement done here, as it will be impossible to test the 8812au with all the modifications

satmandu, 2020-03-04:
> > VHT requires FPU support, which is missing by default on arm64 kernels. I'd love to see someone convert the VHT code to not need the float support so it works on arm64! :)
> 
> It won't work in 80211AC on a Raspberry Pi4?

Note https://github.com/torvalds/linux/blob/8b614cb8f1dcac8ca77cf4dd85f46ef3055f8238/arch/arm64/Makefile#L49  - the -mgeneral-regs-only flag keeps any float code from working  - that's an explicit don't use floating point command to the compiler.

So unless you hack up the kernel source you're not getting fpu code working on an arm64 kernelI there is a better way, please let me know how!

(Hacking that line means that zfs compiles fail. :( )

kimocoder, 2020-03-04:
Don't worry about all patches that needs to be added after the potært, that I'll handle with easy :+1:

fariouche, 2020-03-06:
Merge done and working with 8812au.
So far, didn't experience any stalled transfer nor disconnects.
Only tested 2.4, 80Mbps download, 90 upload.
Need to check the power current, I use a high power USB 3 port.
Available on my forked repo so far, will push here soon

fariouche, 2020-03-10:
So far, I'm having the following issues:
- 8821: the AP mode does not work. Didn't tried the station mode yet
- 8812: I'm having a kernel oops, seems related to the USB interface. Maybe USB3 to USB2. I'm doing more tests (didn't see any issue when plugged to an USB 3 interface. Only when using USB2)

@kimocoder , I cannot create a pull request here as I cannot create the new branch

kimocoder, 2020-03-10:
Ok, AP moderne isn't first priority anyway, that's ok, we other may have something to work with we also. The USB is an issue then, but let's get together working on it.

Push to v5.6.4.5 remember..

@aircrack-ng can you give @fariouche some priv. in this repo? :+1:

fariouche, 2020-03-13:
Just push a pull request for the v5.8.5
I fixed the 8821 and 8811 support.
All chipsets are working, but still have an issue with AP mode when using it for more than 8h... most likely a memory leak as I cannot reproduce it with my main computer (32GB) but my router with 512MB crashes after 8h with no specific traces.
This may no affect the station mode.
I'm still investigating it, but it may take some time as it is long to reproduce and debug