aircrack-ng/rtl8812au #559: AWUS1900 not capturing any targets in Kali Linux?

LoafyCodes, 2020-02-13:
I'm trying to use the tool "airocrack-ng" with my Alfa AWUS1900 Wi-Fi adapter on Kali Linux, in VirtualBox. The adapter is recognized and I am able to put the card into monitor mode. However, the issue is that I am not receiving any output/targets when running the command "airocrack-ng wlan0", while the device is in monitor mode. I followed all instructions listed here.

[image]

I have tried on a Live USB of Kali Linux, and it worked!? I was able to receive targets like how I should be able too. But I need it to work in VirtualBox. Anyone know what the issue is? Let me know if more info/screenshots are needed.

[image]
[image]

LoafyCodes, 2020-02-14:
Provided more information and screenshots.

iMoD1998, 2020-02-22:
Had the same issue when selecting specific channel. Try using the latest version of the driver. It seemed to fix it. However, the latest driver is pretty unstable and doesnt wanna work with things like bettercap and wifite because its got problems with iwconfig.

iMoD1998, 2020-02-22:
Also if you're on the latest version, i noticed that it being in a USB3 port made it not detect as many things as it does in a USB2.

kimocoder, 2020-02-22:
I'll force it back to USB2 tomorrow, anyhow, Kali hasn't packed the latest driver, so it doesn't support the kernel

LoafyCodes, 2020-03-01:
Thank you for all the work you put in @kimocoder!
I've been using the AWUS1900 on a Live USB and it works perfectly!

kimocoder, 2020-03-08:
Just added some parches and fixes to the driver.
There's a brand new 8814AU and 8821AU firmware coming spon :+1: 

```
git clone https://github.com/aircrack-ng/rtl8812au -b v5.6.4.2
cd rtl8812au
make -j4 && make install
```

kimocoder, 2020-03-08:
And VirtualBox needs the drivers in VM software installed, VMWARE also, and I know it's been trouble all the way.. will take a look later sometime

closed 2020-03-08 by kimocoder

Swiffers, 2020-06-11:
I'm having the same issue here. Unable to find any network with the latest drivers on VirtualBox + AWUS1900 (rtl8814au).

Joemoma1337, 2020-12-25:
Good day all,
I have just gotten my Alfa AWUS1900 (RTL8814AU) and i've only gotten it to get into monitor mode, view networks, connect to networks but i am unable to capture handshakes or deauth devices (to then capture handshakes).

Has there been any updates to this?

running Kali: 5.9.0-kali5-amd64 #1 SMP Debian 5.9.15-1kali1

Merry Christmas all!

kanishkapillai, 2021-04-20:
> Good day all,
> I have just gotten my Alfa AWUS1900 (RTL8814AU) and i've only gotten it to get into monitor mode, view networks, connect to networks but i am unable to capture handshakes or deauth [...]

Have you got this to work? facing the same issue.... Or is it a problem with kali?

cyrus104, 2021-04-20:
@kanishkapillai 

There was a driver updated and the RTL8814AU was split and being maintained separately. 

https://github.com/aircrack-ng/rtl8814au