aircrack-ng/rtl8812au #560: RTL8811AU

AltyMan, 2020-02-13:
Does the Ralink RTK8811AU chipset also work with these drivers? Does it work with other drivers? Or does it not support aircrack-ng or packet injection at all?

justinvforvendetta, 2020-02-13:
as to whether it works with this driver, im not sure. but i dont see why it wouldnt support monitor/packet injection.

netskink, 2020-02-20:
I have this  usb adapter working with x86 as a wifi client and a wifi access point using hostapd/dnsmasq using this driver https://github.com/abperiasamy/rtl8812AU_8821AU_linux.

However that driver with raspberry pi did not work using similar configs/methods for use as an access point.

With this driver, it works as a wifi client but it also does not work as an access point using the same methods.  So I don't know if its a ARM implementation of the driver or something in the network stack itself on pi.

kimocoder, 2020-03-08:
The README tell you whats supported and not.

This driver has support for RTL8812AU, RTL8814AU, RTL8821AU (RTL8811AU) with frame injection, baked in a module named 88XXau.

We have a new 8814AU and 8821AU coming up too, so stay tuned.

closed 2020-03-08 by kimocoder