aircrack-ng/rtl8812au #562: RTL8812au USB 3 causing lower range.

iMoD1998, 2020-02-18:
While using the PC I can find a lot of networks about 10-12. I can also find the same amount if I use the USB 2 port on the Pi. However, when using the USB 3 port on the Pi, it can get anywhere from 4-6. 

This could be because I'm using the older version of the driver though. I'm using the one built into kali which I think is version V5.2.20.

Thought it might be some sort of interference, I tried using some ferrite chokes but it didn't seem to make any difference.

If it was fixed in a previous update I will try updating but i was having issues setting the TX power on the latest driver on PC so I've been reluctant to try.

If you need more info just let me know.

kimocoder, 2020-03-08:
v5.2.20 is left behind. Simple as that :100:

kimocoder, 2020-03-08:
USB3 should not be used, in later branches we force it down to USB2

closed 2020-03-08 by kimocoder